xoseperez/espurna #1821: "On MQTT disconnect" parameter not correct in Web Interface and resetted by Save

C64Axel, 2019-07-13:
**Bug description**
Option "On MQTT disconnect" in Menue Switch is resetted to "Don't change"

**Steps to reproduce**
Set "On MQTT disconnect" to "Turn the Switch OFF" ans press "Save"
On MQTT disconnect the Switch goes off.

Go to Menue "Switch" and "On MQTT disconnect" is set to "Don't change" 
Go to any other Menue and press Save. 
The option "Don't change" is saved to config an the Switch dosen't turn off on MQTT disconnect.

**Expected behavior**
Save must not reset option "On MQTT disconnect"

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

[339373] [MAIN] ESPURNA 1.13.5 (83195fd)
[339374] [MAIN] xose.perez@gmail.com
[339377] [MAIN] http://tinkerman.cat

[339378] [MAIN] CPU chip ID: 0x97BC1D
[339379] [MAIN] CPU frequency: 80 MHz
[339380] [MAIN] SDK version: 1.5.3(aec24ac9)
[339386] [MAIN] Core version: 2.3.0
[339387] [MAIN] Core revision: 159542381
[339392] 
[339394] [MAIN] Flash chip ID: 0x144051
[339396] [MAIN] Flash speed: 40000000 Hz
[339398] [MAIN] Flash mode: DOUT
[339401] 
[339402] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[339414] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[339417] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[339426] [MAIN] Firmware size       :   474032 bytes /  116 sectors (   1 to  116)
[339435] [MAIN] Max OTA size        :   548864 bytes /  134 sectors ( 117 to  250)
[339439] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[339448] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[339455] 
[339457] [MAIN] EEPROM sectors: 251, 250
[339458] [MAIN] EEPROM current: 250
[339460] 
[339461] [MAIN] EEPROM:  4096 bytes initially |  1499 bytes used (36%) |  2597 bytes free (63%)
[339472] [MAIN] Heap  : 37088 bytes initially | 20040 bytes used (54%) | 17048 bytes free (45%)
[339478] [MAIN] Stack :  4096 bytes initially |  1280 bytes used (31%) |  2816 bytes free (68%)
[339488] 
[339490] [MAIN] Boot version: 31
[339491] [MAIN] Boot mode: 1
[339492] [MAIN] Last reset reason: Reboot from web interface
[339500] 
[339502] [MAIN] Board: ITEAD_SONOFF_4CH
[339503] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[339519] [MAIN] WebUI image: SMALL
[339521] 
[339522] [MAIN] Firmware MD5: b4fc0f716fa699f624d042285ae4d333
[339529] [MAIN] Power: 3138 mV
[339531] [MAIN] Power saving delay value: 1 ms
[339533] 

**Tools used**
* Win10
* Chrome 75.0.3770.100

mcspr, 2019-07-30:
Does using the debug terminal and setting it manually work?
```
set relayOnDisc0 1
```
0 is the first switch, 1 is the OFF mode

I cannot reproduce with current `dev` builds

C64Axel, 2019-07-31:
Try this Steps please:

Go to Switches
Turn "On MQTT disconnect" to an option other than "Don't change"
Press Save and OK
Reload the Page
Go to Switches
"On MQTT disconnect" is set to "Don't change"

Saving now set's the option to "Don't change".

mcspr, 2019-07-31:
Yep, it is fixed in dev: 494427402a282f846a8337ce847739d1650cd035

> Development branch .bins - https://github.com/mcspr/espurna-nightly-builder/releases

Only link is in the current release notes, here:
https://github.com/xoseperez/espurna/releases/1.13.5
Might not be as discoverable as I initially thought.

closed 2019-10-06 by stale[bot]