xoseperez/espurna #1825: espurna behind apache reverse proxy

lblabr, 2019-07-14:
i'm running my espurna devices behind apache reverse proxy / haproxy setup.

could somebody give me an advice how the apache reverse proxy configuration has to be ?
or could migrate the example from 

https://github.com/xoseperez/espurna/issues/937#issuecomment-511189959

to an apache config ?

at the moment i have:

ProxyPass / http://10.100.10.62:80/
ProxyPassReverse / http://10.100.10.62:80/

ProxyPreserveHost On

ProxyRequests Off
RewriteEngine on
RewriteCond %{HTTP:UPGRADE} ^WebSocket$ [NC]
RewriteCond %{HTTP:CONNECTION} Upgrade$ [NC]
RewriteRule .* ws://10.100.10.1:6868%{REQUEST_URI} [P]
ServerName pump-brunnen.xxx.xx

thanks in advance

m-kozlowski, 2019-07-14:
this works for me
```
ProxyPass "/ws"  "ws://esp-wp-06/ws"
ProxyPass / http://esp-wp-06/
```
You might also have to set `webDomain` variable.

lblabr, 2019-07-16:
i had some problems with 1.13.5

[https://github.com/xoseperez/espurna/issues/1614](url)

can this be fixed with some setting in apache configuration ?

m-kozlowski, 2019-07-17:
Wait... You want to workaround lwip MTU related bug by putting reverse proxy between espurna and the client?
1: If it's really that bug, I doubt it would help
2: wouldn't it be easier? https://github.com/xoseperez/espurna/issues/1610#issuecomment-493725279

lblabr, 2019-07-20:
just went back to my old 1.13.1 and it worked. i didn't tested with ab build like #1610. 
i'm working more or less with arduino ide and only platform io ist referenced.... 
at the moment i didn't spent time to figure how i have to setup my framework just went back to an release download, it worked in my case...

sry for that

lblabr, 2019-07-20:
my general setup is apache reverse proxy <- openvpn -> haproxy -> servers/clients/IoT 

so it could somehow possibly to have quick workoround in that setup, not to set up an reverse proxy to fix the mtu issue

mcspr, 2019-07-20:
> i'm working more or less with arduino ide and only platform io ist referenced....

https://github.com/xoseperez/espurna/issues/1610#issuecomment-473794750 paths are for PIO, but you can do the same with IDE installation. You just need to find where `{ARDUINO_DATA_DIRECTORY}/packages/esp8266/hardware/esp8266/{CORE_VERSION}/` is located (where data is either `~/.arduino15`, `%LOCALAPPDATA\Arduino15`, `~/Documents/ArduinoData/` (?) or `~/Library/Arduino15/`

Cores 2.5.0 >= do use lwip2 by default already (still a selectable option), but you will see a pretty significant firmware size increase

mcspr, 2019-08-09:
BTW, since the problem with packet size, check out the #1843 
Instead of one large payload, which breaks even with most basic config (~1.8KB > 1460MSS), this reverts to pre 1.13.5 behaviour and sends several small packets instead. It still may break on some specific modules if they reach the 1400 byte limit, but should work for in the most common cases.

I would say it is in draft status, but it works for me on a temperature sensor + energy monitor board config without any problems so far.

closed 2019-10-09 by mcspr