xoseperez/espurna #1831: As soon as I try to connect the webbrowser I got the following Exception (29)

gernst48, 2019-07-21:
As soon as I try to connect the webbrowser I got the following Exception (29):

-----------------------------------------------------------------------------

hardware.h configuration:

    // Info
    #define MANUFACTURER        "NODEMCU"
    #define DEVICE              "LOLIN_EG_POWER"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1

    // Hidden button will enter AP mode if dblclick and reset the device when long-long-clicked
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // Light
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // A bit of I2C
    #define I2C_SDA_PIN         4
    #define I2C_SCL_PIN         5

    // BMX280 Environment sensor
    #define BMX280_SUPPORT		1

    // Energy definitions
    #define SENSOR_ENERGY_UNITS    ENERGY_KWH   // Energy units (ENERGY_JOULES | ENERGY_KWH)
    #define SENSOR_POWER_UNITS     POWER_WATTS  // Power units (POWER_WATTS | POWER_KILOWATTS)

    // ADS1x15 Energy monitor
    #define EMON_ADS1X15_SUPPORT   1
    #define EMON_MAX_SAMPLES       1000  // Max number of samples to get
    #define EMON_MAX_TIME          250   // Max time in ms to sample
    #define EMON_FILTER_SPEED      512   // Mobile average filter speed
    #define EMON_REFERENCE_VOLTAGE 3.3   // Reference voltage of the ADC
    #define EMON_MAINS_VOLTAGE     230   // Mains voltage
    #define EMON_CURRENT_RATIO     30    // Current ratio in the clamp (30A/1V)
    #define EMON_REPORT_CURRENT    1     // Report current
    #define EMON_REPORT_POWER      1     // Report power
    #define EMON_REPORT_ENERGY     1     // Report energy

-----------------------------------------------------------------------------

Serial terminal output:

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000449]

---8<-------

[000449] [MAIN] ESPURNA 1.13.6-dev
[000449] [MAIN] xose.perez@gmail.com
[000449] [MAIN] http://tinkerman.cat

[000450] [MAIN] CPU chip ID: 0x5B8CCF
[000453] [MAIN] CPU frequency: 80 MHz
[000456] [MAIN] SDK version: 1.5.3(aec24ac9)
[000460] [MAIN] Core version: 2.3.0
[... 369 lines cut ...]
[225878] [MAIN] EEPROM size         :    16384 bytes /    4 sectors (1016 to 1019)
[225885] [MAIN] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)
[225892]
[225894] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[225898] [MAIN] EEPROM current: 1016
[225901]
[225902] [MAIN] EEPROM:  4096 bytes initially |   531 bytes used (12%) |  3565 bytes free (87%)
[225912] [MAIN] Heap  : 32840 bytes initially | 16728 bytes used (50%) | 16112 bytes free (49%)
[225919] [MAIN] Stack :  4096 bytes initially |  1456 bytes used (35%) |  2640 bytes free (64%)
[225927]
[225928] [MAIN] Boot version: 31
[225931] [MAIN] Boot mode: 1
[225933] [MAIN] Last reset reason: Exception
[225937] [MAIN] Last reset info: Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4020eec2 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[225951]
[225952] [MAIN] Board: NODEMCU_LOLIN_EG_POWER
[225956] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT I2C LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[225972] [MAIN] Sensors: BMX280 EMON_ADX1X15
[225976] [MAIN] WebUI image: SENSOR
[225979]
[225980] [MAIN] Firmware MD5: cefc7773ebc73bf1e45bee6ebccae691
[225986] [MAIN] Power: 2788 mV
[225988] [MAIN] Power saving delay value: 10 ms

-----------------------------------------------------------------------------

gernst48, 2019-07-21:
Sorry I was to fast.
I want to give more information:
The board and software is still working (I get all requested MQTT information).
Only connecting the webinterface does the crash.

gernst48, 2019-07-25:
I like to debug my firmware on my own, because I detected some different behaviour when
i play around with the EMON_ADS1X15_SUPPORT configuration flags EMON_???.
Is there any debugger I could use with the Visual Studio Code ?
Or at least a ELF file viewer ?
Or how do you debug your firmware ?

tonilopezmr, 2019-07-25:
Like this proposal: https://github.com/xoseperez/espurna/issues/1823#issuecomment-511154440 you can decode your stack trace.
For debug, add logs and see the logs using USB, and then try to understand printing your own messages etc...

gernst48, 2019-08-03:
Depending on the parameter I use for the EMON_ADX1X15 there are differne behaviour.
With the default parameter the firmware starts as expected.
when I reload the web page the exception 29 happens.
Loading the web page again every thing is working again, and so on ...
Analyzing the trace and looking around with google gives me the feeling
there is somthing wrong with the heap and the stack.
I believe the heap grows up to the stack and then the exception 29 happens.
So, do you have any idea how I could check the heap fragmention ?
Or to implement an exception handler to catch and test this case ?

gernst48, 2019-08-03:
When I use my own parameters for the EMON_ADX1X15 which displays more web data in the
web status page, I got for every web page load the exception 29.
I could not display any web page anymore.
Only the serial debug interface  is still working.
Again, in my opinion there seems to be a big heap problem.

mcspr, 2019-08-04:
There are a couple of issues, but I still don't quite understand why this is happening specifically because of reload (ref: #1824)
- 28/29 most of the time is allocation failure + trying to access something that is expected to be in the allocated area. `auto obj = new Object();`, `obj->method()` -> exception, because obj is nullptr. One point is ws WebSocketIncommingBuffer creation / deletion. Another, is 3K+ JsonObject string serialization + ~1K JsonDynamicBuffer as a storage that is reallocing in size periodically
And technically, using never Cores it will cause c++ bad_alloc instead, crashing anyways.
- for heap specifically there is espurna/libs/HeapStats.h ::getHeapStats, there are `usable` and `frag_pct` struct members. Note that you will need to build with Arduino Core 2.5.2 for it to work at all. There are more heap though, so one would need to do 2 things - add `disable_extra4k_at_link_time();` somewhere in setup and `malloc(N-bytes)` to have the same heap as the old 2.3.0 does.

At first I thought this was something wrong with ADX sensor specifically, but my guess it is a more generic issue.

gernst48, 2019-08-04:
The reload error is quit clear for me.
Because it seems either the send buffer is not returned or the heap gets fragmemted and
ther is not enough memory for the next send buffer allocation.
After the exceptio 29 and the reboot everything is clean and working again.
Until the next buffer allocation ....
Maybe we should try to implement a static send buffer for test and watching what happens.

mcspr, 2019-08-04:
Right. And the main point is that sensor firmware + energy sensor, json buffer is like 6K (all those keys) + json string 3K. Static buffer is no problem as some brief tests show, but it needs a middle ground solution between sending each callback (which can be either 1 key-value or giant monster like sensor one) separately and sending all at once (which is what it is right now and kind of broken).
And likely moving *really* static things before everything else, since we would know the size beforehand.

And I'll correct myself about the realloc part - the algorithm used by basic json buffer is to start with 256 bytes and add a double-size block each time something does not fit. Manually specifying the size still does this though, unless the buffer is Static / is in the stack.

gernst48, 2019-08-08:
Sorry, does not work for me.

Maybe there is something wrong in the configuration.
I use the following in platformio.ini for my board, but you have change a lot in this file.
The base for me is the following:
[env:nodemcu-lolin]
board = ${common.board_4m}
build_flags = ${common.build_flags_4m1m} -DNODEMCU_LOLIN -DDEBUG_FAUXMO=Serial -DNOWSAUTH
My board uses the following configuration:
[env:nodemcu-lolin-eg-power]
board = ${common.board_4m}
build_flags = ${common.build_flags_4m1m} -DNODEMCU_LOLIN_EG_POWER -DDEBUG_FAUXMO=Serial -DNOWSAUTH -DSENSOR_SUPPORT=1

But I missing a lot from the old platformio.ini file e.g.:
[env:nodemcu-lolin-eg-power]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_4m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_4m1m} -DNODEMCU_LOLIN_EG_POWER -DDEBUG_FAUXMO=Serial -DNOWSAUTH -DSENSOR_SUPPORT=1
upload_speed = ${common.upload_speed_fast}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

I believe my configuration is wrong.
Please let me know what I have to configure for my board.
And by the way, is the Firmware version still 1.13.6-dev ?

Here is the status output of my board:
Manufacturer NODEMCU
Device LOLIN_EG_POWER
Chip ID 5B8CCF
Wifi MAC CC:50:E3:5B:8C:CF
SDK version 1.5.3(aec24ac9)
Core version 2.3.0
Firmware name ESPURNA
Firmware version 1.13.6-dev
Firmware revision
Firmware build date 2019-08-08 23:23:09
Firmware size 502576 bytes
Free space 2641920 bytes
Network TEKENER BSSID F8:1A:67:A5:CB:FC Channel 7 RSSI -56 IP 192.168.2.41 (telnet)
Free heap 11736 bytes
Load average 1%
VCC 2788mV
MQTT Status CONNECTED
NTP Status SYNC'D Current time 2019-08-08 23:54:33
Uptime 0d 00h 14m 56s
Last update 23 seconds ago

Hope you could enlight my dark brain.

mcspr, 2019-08-09:
[env:nodemcu-lolin-eg-power] config is fine, latest update to .ini file removed a lot of duplicated strings per env. Instead, they are inherited from the [env] block at the top
Version only changes between releases btw. Revision string is supposed to point to the specific commit but is not set for local builds right now.

#1843 does not work?
Have you tried decoding the stack trace that happens? `>>>stack>>> ... <<<stack<<<` lines contain raw addresses specific to the binary you have and we can only guess where the crash happens exactly. If you are having problems with that, just post the trace text and the .elf file (located at .pio/build/nodemcu-lolin-eg-power/firmware.elf)

I am running the config you provided in the first post, with BMX280 & ADS1X15 giving out dummy values and WebUI loads fine with Firefox on Windows machine.
Maybe some local environment is the cause? Can you try disabling Alexa (as I see some network activity there)? `ALEXA_ENABLED` & `ALEXA_SUPPORT` set to 0, or `alexaEnabled` at runtime?
I do see that 13824 bytes of free heap though, you got slightly lower value.
Have you tried connecting to the device directly, via SoftAP mode?

gernst48, 2019-08-09:
Please find attached all by traces and opinions:
[Power-Stack-Default-Decode-4.txt](https://github.com/xoseperez/espurna/files/3487237/Power-Stack-Default-Decode-4.txt)

mcspr, 2019-08-09:
Thank you!

```
21 0x4020db21: ArduinoJson::Internals::JsonPrintable<ArduinoJson::JsonObject>::measureLength() const at W:/GitRepos/espurna-dev/code/espurna/ws.ino:528
22 0x40216fed: wsSend(unsigned int, ArduinoJson::JsonObject&) at W:/GitRepos/espurna-dev/code/espurna/ws.ino:528
23 0x4021707b: _wsStart(unsigned int) at W:/GitRepos/espurna-dev/code/espurna/ws.ino:528
```
`measureLength()` part is consistent with the other issue - problem happens when json serializer tries to do something on supposedly broken json buffer structure. However, this crash is expected as this is still with the current `dev` version and not the linked PR (easily seen, as there is no longer any _wsStart() function in the new code)

The proposed fix is in the `web/fixed-size-json-v5` branch, which you can test via
- `git clone -b web/fixed-size-json-v5 https://github.com/mcspr/espurna.git espurna-test`
and build that
- add `https://github.com/mcspr/espurna` as a git remote in the existing repo and checkout the PR branch
- `git fetch origin pull/1843/head:pr-1843` & `git checkout pr-1843`
(ref: https://help.github.com/en/articles/checking-out-pull-requests-locally)

gernst48, 2019-08-09:
Sorry my error.
As always, if you used the right one everything is working as expected.
I could confirm the test branch is working for all my configurations.

closed 2019-08-12 by mcspr