xoseperez/espurna #1834: How wemos d1 mini pro flashing with arduino ide?

klotzma, 2019-07-22:
Hello everybody,
I would like to flash my wemos d1 mini Pro and connect a PZEM004t V3.

So far I have added the PZEM zip under Arduino IDE and now only has to configure the rest with Atom.

Under arduino.h I have to select the hardware now.
Which one do I choose for the Wemos D1 mini pro?

I've already tried it with Wemos D1 mini, but then comes at the build always a mistake: Hardware not supported!

mcspr, 2019-07-22:
"Hardware not supported!" will show if you have not selected any hardware define, like `NODEMCU_LOLIN` or `ITEAD_SONOFF_POW`. There is a dummy hw `NODEMCU_BASIC` that defines bare minimum for things to work: `MANUFACTURER` and `DEVICE`. Please reference comments in the header files.

ref platformio.ini:
https://github.com/xoseperez/espurna/blob/fbf2b71749307688f7258ad8235bc0f5d068e926/code/platformio.ini#L289
you can add use
```cpp
#define NODEMCU_BASIC
#define LED1_PIN 2
#define LED1_PIN_INVERSE 1
#define DEBUG_SERIAL_SUPPORT 0
#define PZEM004T_SUPPORT 1
```
`DISABLE_POSTMORTEM_STACKDUMP` does nothing in Arduino IDE environment.

klotzma, 2019-07-23:
Do I have to change something here?
https://github.com/xoseperez/espurna/blob/fbf2b71749307688f7258ad8235bc0f5d068e926/code/platformio.ini#L2

So I have to change that?
https://github.com/xoseperez/espurna/blob/fbf2b71749307688f7258ad8235bc0f5d068e926/code/espurna/config/hardware.h#L92

that it looks like that:
```c++
#elif defined(NODEMCU_BASIC)
    // Info
    // Generic NodeMCU Board without any buttons or relays connected.
    #define MANUFACTURER        "NODEMCU"
    #define DEVICE              "BASIC"
    #define LED1_PIN 2
    #define LED1_PIN_INVERSE 1
    #define DEBUG_SERIAL_SUPPORT 0
    #define PZEM004T_SUPPORT 1
```

Do I have to flash it with this ZIP?:
https://github.com/xoseperez/espurna/tree/fbf2b71749307688f7258ad8235bc0f5d068e926

What do you mean by that?
> `DISABLE_POSTMORTEM_STACKDUMP` does nothing in Arduino IDE environment.

mcspr, 2019-07-23:
Right. Now, if you add `#define NODEMCU_BASIC` line in the arduino.h, it will use those defines.
And yes, please do use `dev` tree to build.

platformio.ini is not used in any way, that was just for reference of how pre-built binary `nodemcu-pzem004t.bin` is made. `DISABLE_POSTMORTEM_STACKDUMP` will not have any effect, you can safely ignore it for your builds.

klotzma, 2019-07-23:
Which board do I choose then, so I can build the bin? 
[image]

mcspr, 2019-07-25:
You can select either Generic or Wemos D1. Just make sure that the flash size is 4MB if you want to do OTA upgrades

klotzma, 2019-07-31:
ok.
I get this error:

`In file included from sketch\config/all.h:36:0,

                 from C:\Users\Benchawan\Downloads\espurna-dev\espurna-dev\code\espurna\espurna.ino:24:

sketch\config/prototypes.h:2:25: fatal error: ArduinoJson.h: No such file or directory

 #include <ArduinoJson.h>

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board LOLIN(WEMOS) D1 R2 & mini.`

Can you help me ?

klotzma, 2019-07-31:
I now have a bunch of libraries installed and now comes the error:

`In file included from C:\Users\Benchawan\Downloads\espurna-dev\espurna-dev\code\espurna\ntp.ino:15:0:

sketch\libs/NtpClientWrap.h: In constructor 'NTPClientWrap::NTPClientWrap()':

sketch\libs/NtpClientWrap.h:15:13: error: cannot convert 'WiFiUDP*' to 'AsyncUDP*' in assignment

         udp = new WiFiUDP();

             ^

exit status 1
Fehler beim Kompilieren für das Board LOLIN(WEMOS) D1 R2 & mini.`

mcspr, 2019-08-01:
Please see https://github.com/xoseperez/espurna/wiki/ArduinoIDE#installing-libraries-manually
NtpClientLib is a different version from the one in the library manager

klotzma, 2019-08-03:
I got it now and loaded espurna on the Wemos d1 mini.
Thank you for your help!!!

closed 2019-08-03 by klotzma