xoseperez/espurna #1835: ota update Flash Write Failed

gernst48, 2019-07-22:
Trying to use ota firmware update in a serial terminal leads in the following error:

ota http://192.168.2.4/data/nodemcu-lolin-eg-power-firmware.bin
[274720] +OK
[274725] [OTA] Downloading /data/nodemcu-lolin-eg-power-firmware.bin
ERROR[1]: Flash Write Failed
[316147] [OTA] Progress: 1073667996 bytes
ERROR[1]: Flash Write Failed
[316147] [OTA] Disconnected

tonilopezmr, 2019-07-22:
it seems `1073667996 bytes` are too much!

mcspr, 2019-07-22:
@gernst48 What server are you using at the http://192.168.2.4?
1073667996 bytes feels like a bug with http parsing on our side though

mcspr, 2019-08-12:
@Niek you mentioned in the #1716 fixes for OTA headers, was that the problem?

I will try it out some time this week, and there's another possibility header parsing is broken on our side
- https://bitbucket.org/xoseperez/fauxmoesp/issues/46/cannot-use-setbinarystate-with-pywemo , headers arrive separate from the body
- it needs something like #1806 to remember parsing state (across multiple callbacks) and more safe-checks
- headers again, but too much, so we never get to the body in the same callback

I will add #1751 to clearly compare OTA implementations. And it also includes a little fix to immediately close the connection

Niek, 2019-08-13:
@mcspr the current OTA code expects 2 * \r\n in the first data chunk and assumes that's where the response headers stop and the data starts. Some web servers might send some weird headers, or very long ones, or maybe the MTU is small, so this pattern can end up in the second chunk (not sure how ESPAsyncTCP's ``onData()`` chunking works internally - but I ran across this in a few tests), so in https://github.com/xoseperez/espurna/pull/1716/files#diff-668d7fcc36707d45e425775d15bc9b5bR112-R120 I changed it to read char-by-char and it should be more fault-tolerant.

#1751 + ``OTA_CLIENT_HTTPUPDATE`` should also fix this since it handles the header stuff already.

mcspr, 2019-08-20:
True, using `OTA_CLIENT_HTTPUPDATE` should handle things properly (however, idk about if there is anything broken with the Core 2.3.0 builds)
What I tried on the weekend is to migrate HTTP parser logic from the thingspeak to some abstract "HTTP client" - https://github.com/xoseperez/espurna/compare/xoseperez:dev...mcspr:util/http
API handling is unclear, but I will get there eventually... OTA is not yet migrated, only testing with thingspeak for now.

*But!* at the same time, I started thinking about the overall idea of combining http parsing for the server and client (since we stuck with hardcoded private methods *inside* Server classes). It is pretty annoying thing to have to do. First google search for the ESP async client, which I mentioned some time ago in the InfluxDB issue regarding SyncClient - https://github.com/boblemaire/asyncHTTPrequest/blob/debce2c2c6757a8327dc10a4751dee0ab95ea7b6/src/asyncHTTPrequest.cpp#L364
We deal with the same code that needs to parse headers, which is suboptimal from the code size (and extensibility standpoint)

What probably needs to be done is to pull https://github.com/nodejs/http-parser and replace all the parsing code inside of ESPAsyncWebServer with it. The usage is super straight forward, continuously calling parse method on the (const char* data, size_t len), which is what we receive from the ESPAsyncTCP callbacks.
Client implementation would look like what I am already doing the referenced branch.

mcspr, 2019-12-20:
*🐌 *

PR should handle the packet parsing properly. Both when we receive headers and data in a separate packets (python does this, maybe some other software, please tell!) or if the headers take all the space in the packet and we need to do another onData callback before doing anything with the data.

PR #2067 (2019-12-20): OTA: verify data stream and properly handle errors
- move generic methods to ota.h
- harden webui check to reject the updater without magic byte. [Compressed OTA upgrades PR was just merged](https://github.com/esp8266/Arduino/pull/6820/files), so also check for gzip magic
- add checks similar to [UpdaterClass::writeStream](https://github.com/esp8266/Arduino/blob/56127386362157f5dc3591ef06eb4f8f928be601/cores/esp8266/Updater.cpp#L461-L515)
- send back text explanation of an Updater error
- adjust ota debug messages formatting, test throttling of otaProgress
- update asyncota error checks to han [... cut]
closed 2019-12-20 by mcspr