xoseperez/espurna #1838: TX T1UK Support

DJBenson, 2019-08-02:
Has anybody attempted to flash (or even know if it's possible to do so) the new TX T1 switches? I've opened up the board and can see the headers but the board layout is significantly different from the existing T1's.

I will attach board images shortly.

ColinShorts, 2019-08-02:
It should work fine with the T1 binaries. I picked a T3 up last week, which is supposed to be just a different colour (black versus white) than the T2.

That said, I'm only using mine as a remote, so I haven't tested that the correct relay switches on/off, but they do switch on and off.

My only grumble is that they glow way too brightly to be used in a bedroom, and it's a bit shiny in my living room as the TV reflects on the surface.

It's a much nicer looking unit than the old T1 though!

Ground the right hand side of R19 (in the top left section, it's the top right resistor), flash as normal.

DJBenson, 2019-08-02:
Thanks @ColinShorts, I'll give it a go tonight when the kids are asleep.

ColinShorts, 2019-08-02:
No worries, I actually found the info on the tasmota GitHub:

https://github.com/arendst/Sonoff-Tasmota/issues/5918

ColinShorts, 2019-08-04:
Hey, not sure how you got on, but I just noticed that I'm actually using the "yjzk switch_3ch" binary and not the t1. I've set the wifi status led to always on and it's actually off, which is the only difference between the two configs as far as I'm aware. The T1 config /should/ work correctly.

Tofandel, 2019-08-06:
The T1/2/3 have an RF chip (bottom right of the first screen) it's the same as the sonoff bridge

Could you try adding these lines to hardware.h for the T1 and compiling?

See if the RF chip works if you have some 433Mhz remotes

```
    #define RF_SUPPORT          1

    // Only used when RFB_DIRECT=1
    #define RFB_RX_PIN          4
    #define RFB_TX_PIN          5

    // When using un-modified harware, ESPurna communicates with the secondary
    // MCU EFM8BB1 via UART at 19200 bps so we need to change the speed of
    // the port and remove UART noise on serial line
    #if not RFB_DIRECT
    #define SERIAL_BAUDRATE         19200
    #define DEBUG_SERIAL_SUPPORT    0
```

Tofandel, 2019-08-06:
For the flashing it's not too hard and doesn't require soldering (at least if you have some stable pins)

You attach the serial pins normally and add a wire from ground to GPIO 0 on the back hold it by hand plug the usb and you can release it, if the wifi led is not flashing you are in flash mode

So you can flash from there (I use platformio)

[image]

DJBenson, 2019-08-06:
I actually managed to flash espurna a couple of days ago but didn't get round to updating this post. The T1 binaries are indeed compatible but I found a more fundamental issue which I should have perhaps spotted prior to ordering, our house does not have a neutral wire to the switch so this is completely useless 😄Thankfully they are cheap enough for me to not bother so much about that small fact.

ColinShorts, 2019-08-06:
Well that sucks... I'm lucky enough to be in a new build, where the only place I didn't have a neutral was on the intermediate switches either end of the stairs or rooms with two entrances. That was easy enough to overcome with a bit of blue sheathing over one of the wires. It was the only way I was going to be able to put up an external light at the backdoor without making big holes in the ceiling/walls. :)

closed 2019-10-12 by stale[bot]