xoseperez/espurna #1845: Alexa issue with espurna on sonoff dual R2 v1.13.5

Jjmphiphi, 2019-08-09:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

Jjmphiphi, 2019-08-09:
Hi
Alexa doesn't recognize espurna 1.13.5 sonoff itead dual R2. Only version 1.13.2 works.
core version is 2.3.0
ALExa running on Echo Dot

Jjmphiphi, 2019-08-19:
Sorry since my comment was not clearly readable!
This is the copy of the relevant info

Alexa doesn't recognize espurna 1.13.5 sonoff itead dual R2.
version 1.13.2 and earlier works.
core version is 2.3.0
ALExa running on Echo Dot

ColinShorts, 2019-08-20:
I don't believe you've followed the advice provided when you raised the issue (see first post), which is probably why nobody has responded.

There have been a few issues raised recently regarding alexa problems, and there have been some fixes pushed to the development branch.

If you can't build your own binaries from the tree, you could try the nightly builds (you'd like know about this if you had spent a bit of time searching/reading).

https://github.com/mcspr/espurna-nightly-builder/releases

If you are still having issues, follow the advice in the first post above and provide the details asked for, or maybe try the gitter channel before immediately jumping back in here.

Jjmphiphi, 2019-08-24:
Hi Colin
I am terribly sorry to cause any problem in the process of managing issues.
I am not so familiar with these technologies perhaps too old for that!
I don' know how to get more infos knowing that alexa ignores totaly new
versions in its process of recognition. Anyway i will try latest relases
and let you know
Rgds

ColinShorts, 2019-08-24:
No need to apologise! You're never too old so long as you're trying!

Hopefully a later build will fix the issues you've had with Alexa and you can come back to let us know it's fixed 😀

scargill, 2019-09-01:
I am new to ESPURNA - VERY new... I  set up a Sonoff TH16, called it TH16, told Espurna to use ALEXA, Alexa worked, Set up asonoff dual, called it dual, not sure what;s next for Alexa - who'sAPP seems to have found an "ESPURNA-070BEE-1" and an "ESPURNA-070BEE-2". I don;t understand.

ColinShorts, 2019-09-02:
Hey Pete, I think Alexa sees each of the relays on the dual as a separate device, so you would just rename the devices in the Alexa app (although I've not used it for a long time since getting openHAB integration working). Is the TH16 still working as normal?

It might be worth trying the nightly releases (see above), or building your own binaries from the Dev branch using Arduino or platformio. There have been a few changes of late that might help with Alexa.

Platformio run in the code directory well build a new binary for you:
$ pio run -e itead-sonoff-dual

You can then upload the compiled binary (or Max's nightly) via the admin page. There is also an ota tool for managing espurna devices (it's in the wiki) that takes care of building/uploading images of you are feeling adventurous - once it's running, it's really easy to keep devices updated.

scargill, 2019-09-02:
I haave used the nightly built 1.13.6 and indeed Alexa is working fine - forgive me as I'm new to this forum - maybe there are comments elsewhere but I've also noticed that the web admin page doesnot always reffresh on change of status (ie on/off) or if I correct a timeout on temperatuer - refreshing the screen fixes it. On/Off status changes with a real button SOMETIMES show up on the status page but not always. Aldo, Sonoff DUAL, I'm still not getting the normal double-press action for the second relay. Ok, I'll shut up now and direct issues to the right places,

ColinShorts, 2019-09-02:
Status updates will be done using ws, which could be affected by a number of things. There are threads on VPN versus MTU, and ARP cache issues that might be worth reading on.
On the last point, I don't think double click is defined for many devices, but that is a logical use for a single button multiple output device - not something I'd probably need/use sure to the location of the dual r2's (behind walls, beds, etc). Sounds like a reasonable feature request though?

scargill, 2019-09-02:
The only reason I asked about double click is that on the Sonoff Dual official firmware, you can control both relays with one button pressed twice (as against carving up the case ro get to the connector for the two internal pins) :-)

ColinShorts, 2019-09-02:
@mcspr that was fast 😀

mcspr, 2019-09-02:
Not a lot of places to look, so it was kind of an obvious problem :)
Closing, since the Alexa issue is resolved.

> you can control both relays with one button pressed twice

Currently, a single button manages a single switch and every "action" in configuration is created assuming that. I haven't used the original firmware on anything but Basic since about a year ago, so no point of reference here, sorry. But, what it sounds like... count clicks and use that as a switch number to toggle? It would be helpful to track this in a separate issue.

closed 2019-09-02 by mcspr

Jjmphiphi, 2019-09-03:
Thank you very much