xoseperez/espurna #1846: TCP/IP over serial, WiFi as a service mode

dmit2k, 2019-08-11:
Hello,

I’m looking for a solution based on generic ESP8266 board as an external controller for various sensors and relays, which should be directly attached to an embedded linux board over serial / usb. This linux board has no wireless connectivity, but I would still want to use an MQTT client on the external ESP node.

So here are my two questions:

Is there any way to provide the ESP Espurna board with TCP/IP over serial, like PPP over serial client or something like this? The built-in WiFi can still be used as a service mode for initial configuration or “if everything goes bad”.

if not, then is it possible to control / get status of the ESPurna board attached devices from the serial console commands to use in scripts from a linux board?

mcspr, 2019-08-11:
> if not, then is it possible to control / get status of the ESPurna board attached devices from the serial console commands to use in scripts from a linux board?

There is a terminal interface - https://github.com/xoseperez/espurna/wiki/Terminal , but it is closely tied up with debug printing and is mostly intended to be used with user present. For example, most of the commands do have error / ok message at the end, but were not specifically made for scripting purpose.

> Is there any way to provide the ESP Espurna board with TCP/IP over serial, like PPP over serial client or something like this? The built-in WiFi can still be used as a service mode for initial configuration or “if everything goes bad”.

On the program side of this, I'd imagine yhis would require to change debug + terminal to use some structured data protocol instead of just using new-line delimited strings.
What is the specific purpose of the linux board? Automation?

Maybe OT: Since you do mention raw protocol communication... This reminds me of the original ESP8266 AT firmware -> https://www.espressif.com/en/support/download/at
Which obviously does not have sensor drivers, but will make esp a sort-of modem device.

dmit2k, 2019-08-11:
Hi Max,

First of all I'm not talking about Smart Home usage patterns. I'm trying to build a standalone autonomous system with redundancy, environment monitoring etc, based on ready-made firmwares like Tasmota, Espruino, ESPEasy, or whatever else. ESP controllers are just perfect for controlling sensors and "black boxing" telemetry by offloading this task form 'big brothers' like routers, linux boxes, etc - they start within couple of seconds and have minimal power consumption. The problem is, that WiFi is not much reliable / secure and not always available. Even more, it seems a bit overkill to connect two devices within one metal box vi WiFi :)))) 

The idea is to use ESP board for collecting telemetry data (temperature, voltage / current characteristics, other sensors) and connect with main device via serial interface to either:

- report sensors data to main device;
- turn on/off GPIO attached external power relays;
- provide TCP/IP connectivity to the ESP board so it can work with its own MQTT client without using WiFi. 

> On the program side of this, I'd imagine yhis would require to change debug + terminal to use some structured data protocol instead of just using new-line delimited strings.
> What is the specific [...]

Yes, some kind of ))) Generally, it would be perfect if ESP board could connect not only by WiFi but also using serial 'modem like' PPP interface to the big brother )

swilson86, 2019-08-12:
You might want to look at some software named esp-link by JeeLabs. You flash the ESP8266 with this software. Then the ESP8266 can log into your network and you in turn can access the microcontroller which is physically wired to the ESP8266 chip. It acts like a serial interface so you can do over the air updates to your microcontroller too. It also has MQTT capability built in. Not sure whether it will work with a Linux board. I use Arduino instead which uses a sister add on product from JeeLabs named El-Client. If there is the equivalent of El-Client for Linux this could be your solution. 

The author of the software was trying to work with Espurna a few months ago to create a joint project between esp-link and Espurna. Have not seen any of that code yet.

dmit2k, 2019-08-12:
Thanks for the hint, but unfortunately it does exactly opposite of what I need: I need to provide the ESP board with some kind of WIRED internet ))

mcspr, 2019-08-12:
Ah. I got it backwards! But would it not be easier to manage everything on just one side?

So, what if the terminal session does not output anything unless requested and we fully control it via the commands. Some program interacts with the serial; it is periodically running commands to retrieve sensor data and save / send them somewhere else, manages relay states etc. Meaning, it does implement mqtt functionality too, managing the all connectivity instead of making ESP join a fake network.
(which can probably be achieved by writing a specific code that would manage net interfaces api for the lwip lib that esp uses for it's tcp/ip stack...)

dmit2k, 2019-08-12:
> So, what if the terminal session does not output anything unless requested and we fully control it via the commands. Some program interacts with the serial; it is periodically running commands to [...]
This is exactly what I’m playing with at the moment, but using Tasmota ESP FW. It has bidirectional serial console with it’s own command set to manage and configure sensors / peripherals. Unfortunately it also has some limitations so that’s why I’m also looking for alternatives.

Is it possible to use current ESPurna FW the same way without modifications? Then I would be happy to play with and compare...
> (which can probably be achieved by writing a specific code that would manage net interfaces api for the lwip lib that esp uses for it's tcp/ip stack...)
> 
Sometimes offloading sensor processing (including MQTT client)  in full to a standalone microcontroller is a preferred way.  That would be more consistent and stable (and better for black boxing purposes).

I just thought that ESP8266 boards while being so popular already have some workarounds to use wired connection instead of WiFi. I understand that this is pretty specific usage pattern and is opposite to what people expect from SmartHome devices, but still….. )))

As I imagine that could be an alternative to ESP8266WiFi.h which simulates native programming interface but actually working with Serial ports (which could be either default or assigned pins). Then it would be possible to recompile and use almost any firmware like ESPurna, Tasmota, whatever.

mcspr, 2019-08-12:
> Is it possible to use current ESPurna FW the same way without modifications? Then I would be happy to play with and compare...

Like I said, debugging and terminal are intertwined, that is not something implemented (yet). I don't remember the exact issue that this was mentioned before, but the general idea was to make it possible to filter out unneeded log messages by tag system. If in such system terminal output is marked as it's own thing, we can remove all of the other logging from spamming the serial.
(that is, if we are still using terminal commands to do things)

`relay <IDX> <STATUS>` for switches
`gpio <IDX> <STATUS>` for IO pins
`magnitudes` lists available sensors
one missing thing is to get sensor values, like `magnitude <IDX>`

> As I imagine that could be an alternative to ESP8266WiFi.h which simulates native programming interface but actually working with Serial ports (which could be either default or assigned pins). 

There are implementations for SPI-connected Ethernet boards (w5100, w5500):
https://github.com/esp8266/Arduino/tree/master/libraries/Ethernet which is old-school Arduino Ethernet API. The way abstractions work, there is a base Client class that anyone can implement and most of the things like HTTP, MQTT clients do accept it as a substitution for the default WiFiClient (or WiFiClientSecure, which follows the same idea)
https://github.com/d-a-v/W5500lwIP is an example of how to integrate this one level lower, replacing WiFi configuration with ethernet. Networking should work seamlessly, unless something does a bunch of checks calling WiFi... to get network settings / connectivity etc. But I think it needs some further changes to the Arduino Core internals, ref. top 2 issues in that repo & the linked yield()/delay() PR for the Core.

Technically, there is some support for the PPP-over-serial in the LWIP itself: https://lwip.fandom.com/wiki/PPP. But, it is not enabled by default and you would need to change lwip settings and rebuild it yourself. Wiring would probably look something similar to a w5500lwip.

And something that does not apply to the current Arduino Core and NONOS SDK it uses, there was some interest for PPPoS in esp-idf:
https://esp32.com/viewtopic.php?t=2913#p13664
https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/
But I don't know if this need any changes for the esp8266, since it is not included here:
https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/protocols

closed 2019-10-18 by stale[bot]