xoseperez/espurna #1848: Usage of arguments in terminalRegisterCommand() - PLUGIN1

gonzzovela, 2019-08-11:
Hi,

I'm trying to use the option to set a parameter in the plugin template and it seems that in the way it is currently coded it does not work.

When you query in the terminal the value of the parameter, it always sets the parameter to 0 instead of showing the status.

This is the code I use from the template Plugin.ino: (I had to change settingsRegisterCommand() by terminalRegisterCommand() as reported in a previous post).

      void _pluginInitCommands() {
          //* Register Terminal commad to turn on/off the plugin
          //* use this as template to create additional plugin terminal commands
          terminalRegisterCommand(F("PLUGIN1"), [](Embedis* e) {
                  if (e->argc == 0) {
                      DEBUG_MSG_P(PSTR("Pluin1 Status: %s\n"), _plugin1_enabled ? "ON" : "OFF");
                      DEBUG_MSG_P(PSTR("Send 0/1 to enable/disable\n"));
                      DEBUG_MSG_P(PSTR("+OK\n"));
                      return;
                  }
                  _plugin1_enabled = (String(e->argv[1]).toInt() == 1);
                  setSetting("PLG1_EN", _plugin1_enabled);
                  DEBUG_MSG_P(PSTR("Pluin1 Set Status: %s\n"), _plugin1_enabled ? "ON" : "OFF");
                  DEBUG_MSG_P(PSTR("+OK\n"));
              });
      }

If in the Terminal I write PLUGIN1 without any argument, I get the following answer:
[716167] Plugin1 Set Status: OFF
[716167] +OK
so instead of querying the value of the parameter _plugin1_enabled, it sets the value of PLG1_EN to 0 as if it receives a parameter (0) with the command.

Thanks,
Javier

mcspr, 2019-08-12:
```c
if (e->argc == 0) 
```
Here's the problem. Embedis (ref. arg class name) library uses the same logic for arguments as C / C++ do for it's main() function, so argc is 1 in this case, because argv[0] is present (...and is equal to `plugin1`? i think it is lowercase version here? should probably check that)
And when using e->argv[1] later you are going out of argv bounds, since it is only has 1 element.

Note that while the plugin code is still there in wiki, it is true that it was not updated recently and I wonder if settings -> terminal change was the only one. Maybe it needs to settle in the repo instead to more easily manage things like fixes and allow to read it right in the github interface?
Otherwise, it is probably advised to read any module's setup code instead

gonzzovela, 2019-08-18:
> ```c
> ```c
> if (e->argc == 0) 
> ```
> 
> 
> Here's the problem. Embedis (ref. arg class name) library uses the same logic for arguments as C / C++ do for it's main() function. So, argc is 1 in this case, because argv[0] is present (...and is equal to plugin1? i think it is lowercase version here? should probably check that)
> And later, when using e->argv[1], you are going out of argv bounds, since it only has 1 element.
> Note that while the plugin code is still there in wiki, it is true that it was not updated recently and I wonder if settings -> terminal change was the only one required to make it work. Maybe it needs to settle in the repo instead to more easily manage things like fixes and allow to read it right in the github interface?
> Otherwise, it is probably advised to read any module's setup code instead
> ```

Thanks for your help mcspr.

All fixed now.

I’ve changed to argc == 1 to identify when you only call the parameter without any value. In this case, it just prints the value of the parameter without changing any value.

Using then e->argv[1], I can access the passed value and set a new value for the parameter.

Just changing this value and using the right function “terminalRegisterCommand()” fixed it.

Thanks again.
Javier

closed 2019-08-18 by gonzzovela