xoseperez/espurna #1854: my sonoff on Espurna 1.13.5 lost wifi connection

yankoxz, 2019-08-14:
Hi,

first of all i'm sorry for my bad english i'm gonna try to make it clear and correct.

i have 12 sonoff touch 1 gang and 8 sonoff touch 2 gang in my home. i use EspUrna 1.13.5 on all devices and there are conected to 2 Tenda AC6 Routeur 1 part on the one and other part on the second routeur in espconfig i have always only one wifi configure. my problem is that some of the sonoff lost connection but still stay visible in the router config but when i try to tip the ip of there i cant connect and there are no more syncronize with the mqtt group and cant be use by mqtt so i tried diffrent thing :
-change ip adress fixed and dhcp 
-make the routeur restrart every day
-install sonoff touch / sonoff T1-1gang espurna firmaware on the device 
-put the 2 wifi on the devices 
And i have always the same problem it seems to be random devices disconnect there are not always the same that are disconect. i think about to downgrade the espurna version but i dont now if there are better solution.
also i need to add arround 30 sonoff  basic so before to do that on espurna i would like to resolve this issue.

Thanks for your answers and have an awesome day ;)

mcspr, 2019-08-16:
Looks like it is related to the #614 
There's an upstream issue that discussed possible solutions, and what they did end up with is to either:
- send out periodic gratious arp requests from esp
- send out periodic icmp ping requests from esp
- ping esp address from some other pc
Which boils down to doing anything network related.

Since you are using MQTT, can you try reducing keep-alive time to 15 seconds?
Otherwise, we need to implement those additional keep-alive mechanics via arp or ping

yankoxz, 2019-08-18:
thanks, i set all the sonoff keep-alive time to 15 seconds. i'll see if it works i have to waite now.

yankoxz, 2019-08-23:
hi the keep-alive time doesn't change anything i have one of the sonoff double thatdisconect the fist day and other one by on the next days. the device doesn't respond on mqtt , don't reconect and on the router apears to be connected and when i ping i have no response but it send paquet without looses.

mcspr, 2019-08-30:
Can you re-check with current `dev` build?
Or with .bin from https://github.com/mcspr/espurna-nightly-builder/releases? If everything is OK, today's release should be up in an hour.

closed 2019-11-05 by stale[bot]

yankoxz, 2020-02-06:
i've finaly found the problem after a long time to set the reboot of everything every day. it was not problem from the software it was the router that i use to connect the devices : tenda AC6 that disconnect the sonoff so i'm now using a new routeur TP-link archer C6 adnd it is stable but only for 32 devices so i switch the other on the old router.
After a long research on the web i've had the information that the brand don't specify that but there are a lot of router that limit the numer of devices so i'm searching a router that can connect more than hundred devices and the choice is difficult beacause of the price (asus router with asus wrt merlin) finally i'm going to try the xiaomi router gigabyt that specs say up to 128 devices. i'm waiting for it and hope i'll can connect everything on 1 or 2 xiaomi router !.