xoseperez/espurna #1855: Firmware for Tuya curtain switch

yankoxz, 2019-08-14:
Request for ESPurna :

Hi, first of all sorry for my realy bad english but i realy need help ! there it is :
i'm trying to make a custom version of ESPurna for a device that i bought :
  
[image]

With TYWE3S Module : https://docs.tuya.com/en/hardware/WiFi-module/wifi-e3s-module.html
I have tried to make a custom version of ESPurna with ArduinoIDE, I follow the wiki tutorial install all libraries and just modifiying the GPIO controlled in hardware.h with different test of button_mode  and relay but even with the datasheet of TYWE3S I never found a GPIO that work and respond I flash the module I connect to it change the password connected it to the wifi but if I push the button it doesn’t appear on ESPurna webpage and the reverse if I push button on the web interface I have no action on the device. Nothing happens I have tried every GPIO number , nothing. I have read this issue : https://github.com/xoseperez/espurna/issues/1729 but i don't found what i need.
If someone could help me it will be awesome thanks !

mcspr, 2019-08-16:
ref #1729 discussion, it could be that the way switch works is through the secondary MCU that actually has GPIO control. Only thing we can do is to send it some specific commands, but we can only guess for now which ones.

There is a https://github.com/xoseperez/espurna/compare/dev...mcspr:experimental/tuya branch that was successfully used with the dimmer hardware, I assume it should also work for this.
It is not yet in the main tree though, pending some other changes

HeavyM3, 2019-08-18:
I use this switch : Jinvoo WiFi Curtain  - this  looks very similar to your device.

**My settings:** 

//

    #elif defined(JINVOO_CURTAIN)
         #define MANUFACTURER        "Jinvoo"
         #define DEVICE              "Curtain"

     // Buttons

         #define BUTTON1_PIN         12
         #define BUTTON2_PIN         3
         #define BUTTON3_PIN         5

         #define RELAY_SYNC          RELAY_SYNC_NONE_OR_ONE
         #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
         #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
         #define BUTTON1_CLICK       BUTTON_MODE_NONE
         #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
         #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
         #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE

         #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
         #define BUTTON2_PRESS       BUTTON_MODE_ON
         #define BUTTON2_CLICK       BUTTON_MODE_NONE
         #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
         #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
         #define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE

         #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
         #define BUTTON3_PRESS       BUTTON_MODE_TOGGLE
         #define BUTTON3_CLICK       BUTTON_MODE_NONE
         #define BUTTON3_DBLCLICK    BUTTON_MODE_NONE
         #define BUTTON3_LNGCLICK    BUTTON_MODE_NONE
         #define BUTTON3_LNGLNGCLICK BUTTON_MODE_NONE

         #define BUTTON1_RELAY       1
         #define BUTTON3_RELAY       2

    // Relays
         #define RELAY1_PIN          13
         #define RELAY2_PIN          15

         #define RELAY1_TYPE         RELAY_TYPE_NORMAL
         #define RELAY2_TYPE         RELAY_TYPE_NORMAL
         #define LED1_PIN            0
         #define LED1_PIN_INVERSE    0
         #define LED1_MODE           LED_MODE_FINDME_WIFI

**Switch  0 & 2** : 
Boot mode  alsways OFF
                        Pulse mode normaly OFF
                        Pulse time 50 seconds  (depends how long it takes to open or close)

**Switch 1**:   
Boot mode  alsways ON
                 Pulse mode: Dont't Pulse

Works prefectly for me with blinds and window shades

yankoxz, 2019-08-18:
thanks a lot i'm gonna try this settings on my device.

yankoxz, 2019-08-18:
hey i've tried this it works ! i don't now what i have done before but i erease all the project download it and just add your code HeavyM3 and it work verry well i just have to find for the middle button pause your button 2 how to stop the enable relay  beacause :
#define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
     #define BUTTON2_PRESS       BUTTON_MODE_ON
     #define BUTTON2_CLICK       BUTTON_MODE_NONE
     #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
     #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
     #define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE
don't work as off all relay, i don't now if it's just the pin to change.
but i don't see in this code how it gonna have interaction with the relay or command the switch. if you have an idea ^^.

whatever thanks a lot for your help HeavyM3 and mcspr.

closed 2019-10-24 by stale[bot]

luis-sanmiguel, 2020-02-05:
For future reference:

BUTTON1 -> GPIO12
BUTTON2	-> RXD0/GPIO3
BUTTON3 -> GPIO5

NOT USED -> GPIO4

LED1-3 TOGGLE -> TXD0/GPIO1
LED1 -> GPIO16
LED2 -> GPIO14
LED3	GPIO2
LED6 -> GPIO0

RELAY1 -> GPIO15
RELAY2 -> GPIO13

Brands Maxcio, Etersky, ...