xoseperez/espurna #1856: Frequent periodic restarting (crash)

MAWoodMain, 2019-08-14:
**Sonoff light switches crash**
I have several switches running this firmware and I notice that they restart every 30-60 mins with fatal exception 29.
The firmware I'm using is the one from the releases page for this  hardware.

I have extracted the key info over telnet from one of them.

Here is the device info
```
[126916] [MAIN] ESPURNA 1.13.5 (83195fd)
[126917] [MAIN] xose.perez@gmail.com
[126918] [MAIN] http://tinkerman.cat

[126919] [MAIN] CPU chip ID: 0xB4032C
[126920] [MAIN] CPU frequency: 80 MHz
[126924] [MAIN] SDK version: 1.5.3(aec24ac9)
[126927] [MAIN] Core version: 2.3.0
[126931] [MAIN] Core revision: 159542381
[126934]
[126935] [MAIN] Flash chip ID: 0x144051
[126941] [MAIN] Flash speed: 40000000 Hz
[126943] [MAIN] Flash mode: DOUT
[126944]
[126946] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[126956] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[126960] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[126968] [MAIN] Firmware size       :   473696 bytes /  116 sectors (   1 to  116)
[126976] [MAIN] Max OTA size        :   548864 bytes /  134 sectors ( 117 to  250)
[126983] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[126990] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[126997]
[126999] [MAIN] EEPROM sectors: 251, 250
[127000] [MAIN] EEPROM current: 250
[127004]
[127006] [MAIN] EEPROM:  4096 bytes initially |  1229 bytes used (30%) |  2867 bytes free (69%)
[127014] [MAIN] Heap  : 37072 bytes initially | 24376 bytes used (65%) | 12696 bytes free (34%)
[127022] [MAIN] Stack :  4096 bytes initially |  1856 bytes used (45%) |  2240 bytes free (54%)
[127031]
[127032] [MAIN] Boot version: 7
[127034] [MAIN] Boot mode: 1
[127035] [MAIN] Last reset reason: Exception
[127041] [MAIN] Last reset info: Fatal exception:29 flag:2 (EXCEPTION) epc1:0xb48c4010 epc2:0x00000000 epc3:0x00000000 excvaddr:0x0000014c depc:0x00000000
[127054]
[127057] [MAIN] Board: ITEAD_SONOFF_T1_2CH
[127058] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB
[127074] [MAIN] WebUI image: SMALL
[127076]
[127077] [MAIN] Firmware MD5: c51c9042e067d330a8cdc30ffc9c6bc5
[127083] [MAIN] Power: 3228 mV
[127085] [MAIN] Power saving delay value: 1 ms
```

And here's the crash output

```
[719004] [DEBUG] Latest crash was at 14511873 ms after boot
[719008] [DEBUG] Reason of restart: 2
[719009] [DEBUG] Exception cause: 29
[719010] [DEBUG] epc1=0xb48c4010 epc2=0x00000000 epc3=0x00000000
[719012] [DEBUG] excvaddr=0x0000014c depc=0x00000000
[719018] [DEBUG] sp=0x3ffffea0 end=0x3fffffb0
[719023] [DEBUG] >>>stack>>>
[DEBUG] 3ffffea0: c7190000 f4ec60f7 1d583ffe b48c4010
[DEBUG] 3ffffeb0: 00003ffe 00000000 00220000 00000000
[DEBUG] 3ffffec0: c7190000 219660f7 01004010 ffff0000
[DEBUG] 3ffffed0: b48c7fff b48c3ffe 00013ffe 00010000
[DEBUG] 3ffffee0: 00230000 dcc00000 00003fff efd00000
[DEBUG] 3ffffef0: c7193ffe 220060f7 050c0000 c2784000
[DEBUG] 3fffff00: 1f303fff c2004010 00223fff b4740000
[DEBUG] 3fffff10: ffd43ffe 0000ffff 00300000 8b360000
[DEBUG] 3fffff20: 00304026 00100000 ffff0000 8d24ffff
[DEBUG] 3fffff30: b4d44026 00143ffe 00000000 3fff5800
[DEBUG] 3fffff40: 3ffed41a 00000000 3fff6104 4023dbbd
[DEBUG] 3fffff50: 0202000a 00000058 00000000 00000035
[DEBUG] 3fffff60: 00000002 00000018 4022d3c3 3ffee598
[DEBUG] 3fffff70: 3ffed3f4 3fffdcc0 3ffeac50 3ffeac50
[DEBUG] 3fffff80: 4022d336 3ffee598 00000000 3fff5964
[DEBUG] 3fffff90: 3fffdc80 00000000 3fff6104 4023582b
[DEBUG] 3fffffa0: 40000f49 3fffdab0 3fffdab0 40000f49
[DEBUG] <<<stack<<<
```

mcspr, 2019-08-16:
Can you check if it still does this with the current `dev` branch? (or https://github.com/mcspr/espurna-nightly-builder/releases , if you just use .bin releases. builds there should start in about 7 hours)
Also see #1428

I am not really sure 29 exc trace means anything, addresses there do not make any sense :/ Based on some history of 1.13.5 issues, I would assume that is not related at all and only means that WebUI  crashed at some point
Hardware exc is a tricky one, no stack trace

MAWoodMain, 2019-08-17:
Ok, I'll try the latest nightly build tomorrow and post any updates.

Thanks,
Matt

MAWoodMain, 2019-08-19:
Unfortunately it is still doing it
```
[054736] +OK
[060007] [MAIN] System OK
[081415] [WEBSOCKET] Requested action: dbgcmd
[081419] [DEBUG] Latest crash was at 1788803 ms after boot
[081420] [DEBUG] Reason of restart: 2
[081420] [DEBUG] Exception cause: 0
[081420] [DEBUG] epc1=0xac354010 epc2=0x00000000 epc3=0x00000000
[081426] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[081430] sp=0x3ffffea0 end=0x3fffffb0 saved=0x0080

[081435] [DEBUG] >>>stack>>>
[DEBUG] 3ffffea0: 00000000 00200000 0eb20000 ac354010 
[DEBUG] 3ffffeb0: 41973ffe e6984010 00003ffe 1bc58000 
[DEBUG] 3ffffec0: e6984010 dbc03ffe acb83fff 7fff3ffe 
[DEBUG] 3ffffed0: 1fc60000 f54c6a84 1d583ffe b4bc4010 
[DEBUG] 3ffffee0: 00003ffe 00000000 00300000 00000000 
[DEBUG] 3ffffef0: 1fc60000 21966a84 01004010 00050000 
[DEBUG] 3fffff00: 00000000 00200000 0eb20000 ac354010 
[DEBUG] 3fffff10: 41973ffe e5f84010 ed803ffe 000500fe 
[DEBUG] <<<stack<<<
[081578] +OK
```

And info
```

[204470] [MAIN] ESPURNA 1.13.6-dev (43a44cf)
[204470] [MAIN] xose.perez@gmail.com
[204470] [MAIN] http://tinkerman.cat

[204472] [MAIN] CPU chip ID: 0x48A4CA
[204476] [MAIN] CPU frequency: 80 MHz
[204479] [MAIN] SDK version: 1.5.3(aec24ac9)
[204483] [MAIN] Core version: 2.3.0
[204488] [MAIN] Core revision: 9826c6d
[204489] [MAIN] Build time: 1565938614
[204494] 
[204494] [MAIN] Flash chip ID: 0x144051
[204498] [MAIN] Flash speed: 40000000 Hz
[204501] [MAIN] Flash mode: DOUT
[204503] 
[204505] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[204517] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[204519] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[204527] [MAIN] Firmware size       :   481936 bytes /  118 sectors (   1 to  118)
[204533] [MAIN] Max OTA size        :   540672 bytes /  132 sectors ( 119 to  250)
[204541] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[204548] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[204555] 
[204556] [MAIN] EEPROM sectors: 251, 250
[204562] [MAIN] EEPROM current: 251
[204563] 
[204563] [MAIN] EEPROM:  4096 bytes initially |  1005 bytes used (24%) |  3091 bytes free (75%)
[204572] [MAIN] Heap  : 35864 bytes initially | 17088 bytes used (47%) | 18776 bytes free (52%)
[204580] [MAIN] Stack :  4096 bytes initially |  1320 bytes used (32%) |  2776 bytes free (67%)
[204589] 
[204589] [MAIN] Boot version: 7
[204592] [MAIN] Boot mode: 1
[204597] [MAIN] Last reset reason: Hardware Watchdog
[204600] [MAIN] Last reset info: Fatal exception:4 flag:1 (WDT) epc1:0x40000f68 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[204613] 
[204613] [MAIN] Board: ITEAD_SONOFF_T1_1CH
[204617] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[204632] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[204635] [MAIN] WebUI image: SMALL
[204638] 
[204642] [MAIN] Firmware MD5: 271d0eba921008abf506711a6a24c628
[204645] [MAIN] Power: 3135 mV
[204647] [MAIN] Power saving delay value: 10 ms
[204653] 

---8<-------

[204654] +OK

```

mcspr, 2019-08-20:
Exception address is still weird.
Since we can't get any stack trace... perhaps capturing network traffic going to and from the ESP can shed some light on this? To catch something that happens at exactly the same time ESP crashes.
What hardware do you use for the wireless AP? Maybe some other device sends some unexpected data when (and if?) they are in the same LAN?