xoseperez/espurna #1859: Wifi Scan not showing access points in web UI

wildwiz, 2019-08-15:
**Version**
Stock firmware espurna-1.13.5-itead-s20.bin works properly,
Head of dev, currently at c5827b22, has the issue.

**Bug description**
WIFI scan result never showing, the wait spinning icon goes on forever.

**Steps to reproduce**
From Espurna's web UI, go to the WIFI tab and request 'Scan now'.
WIFI scan result never showing, the wait spinning icon goes on forever (waited several minutes)

**Expected behavior**
list of BSSID/AP entries showing in a couple of seconds

**Device information**
Tested on Sonoff S20

**Additional Information**
The websocket was properly up.
A telnet connection on the device shows the request reached the device and was recognized as a 'scan' but nothing more happens
A command WIFI.SCAN via telnet triggers the actual scan and produces the expected results on the command line.

mcspr, 2019-08-15:
Action handler is not triggering, since it is never registered 
https://github.com/xoseperez/espurna/blob/c5827b228d06c7c69d163920a90ef4c7ef46aa33/code/espurna/wifi.ino#L647-L651

i.e.
```patch
diff --git a/code/espurna/wifi.ino b/code/espurna/wifi.ino
index 4d50f306..75ac7c8e 100644
--- a/code/espurna/wifi.ino
+++ b/code/espurna/wifi.ino
@@ -647,6 +647,7 @@ void wifiSetup() {
     #if WEB_SUPPORT
         wsRegister()
             .onConnected(_wifiWebSocketOnConnected)
+            .onAction(_wifiWebSocketOnAction)
             .onKeyCheck(_wifiWebSocketOnKeyCheck);
     #endif
```
But I'd probably hold on a bit and properly wrap it inside of json, since I still can't scan in webui with my unicode SSID present. JSON parser properly escapes some of the symbols, which simple string does not.

wildwiz, 2019-08-16:
Hi, thank you for the quick response, not in hurry and not really an issue for me, I just wanted the regression (if we can call it that way) to be tracked !

mcspr, 2019-08-16:
Thanks! And make sure to update to the latest latest HEAD, https://github.com/xoseperez/espurna/commit/b100d5f60385a3a2af288a6958b31bd8edcbd414 fixes one more bug in the queue processing that will leak 24...72 bytes at a time whenever ws messages are sent from the esp to the client

ruchuf, 2019-08-16:
I can confirm the same issue with itead-sonnof-t1-2ch  and itead-sonnof-t1-1ch.

Furthermore, when I add a network manually and click on SAVE, accepting reboot, the device is neither connecting to the given network nor coming back with a webserver on the previous IP address (access point mode).
After power cycling the device is back as access point and no network configuration is stored.

closed 2019-09-20 by mcspr

mcspr, 2019-09-20:
@ruchuf can you please re-check if network configuration works now? there were a couple of fixes regarding webui settings save and softap mode, I hope network saving should work fine now

PR #1913 (2019-09-20): Fix WiFi scan results in Web
fixes #1859 
Schedule wifi scan results through wsPost, remove manual check in wifi loop function
Rebuilt webui to support array value for "span" elems
Cut a bit of size by inlining wsPost