xoseperez/espurna #1864: HSV set is not HSV set

CrazyIvan359, 2019-08-17:
**Bug description**
HSV commands do not apply the H, S, and V in the command.

**Steps to reproduce**
Send an HSV command to a device with Color enabled.

**Expected behavior**
HSV set should be the same as the command.

**Device information**
* Arduino Core `2.3.0 rev 159542381`
* ESPurna version `1.13.5`

**Solution**
The "Value" of the command is being used as the brightness, then the calculated RGB channel values are scaled by it. This should not be done, as the conversion from HSV to RGB takes the V component into consideration already.

mcspr, 2019-08-20:
Can you give an example of expected values?

Following the brightness function, it does some other transformations when "useWhite" mode is on. However, basic brightness method will simply multiply each channel by the brightness, resulting in the same values as this conversion shows:
https://www.wolframalpha.com/input/?i=0,100,57+hsv+to+rgb
https://github.com/xoseperez/espurna/blob/43a44cf96447dc2e04f0ef936573950df359889e/code/espurna/light.ino#L1212
https://github.com/xoseperez/espurna/blob/43a44cf96447dc2e04f0ef936573950df359889e/code/espurna/light.ino#L117

Extracted light.ino algorithm:
https://gist.github.com/mcspr/4a0a7001c9b52c0dd09ca1af13b287c5
https://github.com/xoseperez/espurna/blob/43a44cf96447dc2e04f0ef936573950df359889e/code/espurna/light.ino#L274-L276
```
$ ./hsv 0,100,57
brightness=0, rgb=0,0,0
input=0,100,57
_setRGBInputValue(255,0,0)
brightness=145, rgb=255,0,0
```
*Note about Arduino.h => std::...*
`constrain` => `std::clamp`
`round` => `std::lround`
`floor` => `std::floor`

CrazyIvan359, 2019-08-20:
I'm running the latest release, not dev. I fixed the issue by preventing the brightness from being calculated when an HSB command is received.

I think if you were to use the maximum Brightness value (100% not `MAX_BRIGHTNESS`) when calculating the RGB values, and then set actual brightness normally, the `_light_brightness_func` should adjust the values to what they should be. I ran out of time to keep testing different solutions though.

I do see that another issue I found has been fixed now, I saw that channels 4 and 5 would be set by brightness if colour was enabled, regardless of white settings.

mcspr, 2019-09-04:
Ping. Was #1874 sufficient?

CrazyIvan359, 2019-09-04:
Sorry Max, I did not have time to load it on any of my devices yet. I will let you know the results when I do, hopefully tonight if not then tomorrow.

CrazyIvan359, 2019-09-05:
@mcspr I don't think so...

I sent `0,100,50` as the command on `.../hsv/set` and got this back over the following 4 minutes on `.../hsv`
```
0,100,12700
0,0,0
0,100,1500
18,67,1500 (repeats 5 times)
32,100,1500
18,67,1500 (repeats 5 times)
0,0,0
0,100,1500
18,67,1500 (repeats 4 times)
0,0,0
34,100,4100
20,66,4100 (repeats ~200 times at ~300ms intervals)
19,65,6300 (repeats 10 times)
0,0,0
33,100,6300
19,65,6300 (repeats 13 times)
0,0,0
0,100,6300
19,65,6300 (repeats 15 times)
0,0,0
33,100,6300
19,65,6300 (repeats)
```

It just continues to recalculate and gitter. No further input was given via UI or MQTT.

mcspr, 2019-09-06:
Wait a sec. Display value is still wrong tho
https://github.com/xoseperez/espurna/blob/43a44cf96447dc2e04f0ef936573950df359889e/code/espurna/light.ino#L394
The first message, before edit, v is 12700

Let's ignore c++ for now, and look at this purely by what's happening with math part. If r,g,b above are 0...255, should'nt they be divided by 255 (aka VALUE_MAX)?

CrazyIvan359, 2019-09-06:
Right... so my lighting automation was putting values on `channel/{0-2}/set` which is why I didn't catch it at first. When that happens, where is the brightness being calculated from? Max out of r, g, b?

Btw I switched off color mode to get around this issue until you or Xose had some time to look into .it, that's why commands were going to channels directly.

mcspr, 2019-09-06:
Yes, since channel value already has brightness applied, and we want to use percentages instead of raw numbers
https://en.wikipedia.org/wiki/HSL_and_HSV#From_RGB
https://www.rapidtables.com/convert/color/rgb-to-hsv.html for some interactivity
I'll test a bit more and add the patch.

cc @xoseperez anyway
And this might be a good time to apply some tests to the calculations (e.g., that `value == fromHSV(toHSV(value))`), but I am not yet sure how to properly integrate it here. Another candidate was #1729 module, but there I just built a host executable with bunch of test asserts for internal structures and mocked serial input / output.

mcspr, 2019-09-06:
And also thank you for the full logs! Because at first I thought it was formatting issue again. Not quite...

CrazyIvan359, 2019-09-06:
> Yes, since channel value already has brightness applied, and we want to use percentages instead of raw numbers

It is possible what I was seeing for V was the channel set command * 100. I can retest for more detailed results tomorrow if you need.

Also let me know if you have a branch you want to test before merging. I should be more available for testing this weekend.

mcspr, 2019-09-06:
Using `(target ? _light_channel[...].target : _light_channel[...].inputValue) / Light::VALUE_MAX` does indeed show proper values. The only question is whether inputValue is useful.

Running magichome-led-controller on wemos board, I do see one significant difference:
[image]
[image]
Since we get the hsv params for the web via `_fromHSV` method.

CrazyIvan359, 2019-09-06:
Brightness doesn't follow V?

mcspr, 2019-09-06:
Yes. Basically, we should not use values without brightness, `_light_channel[...].target / Light::VALUE_MAX` is the only proper conversion
I'm also seeing some rounding weirdness and sending values twice, need to check that out too.

CrazyIvan359, 2019-09-06:
Wouldn't the original arrangement work as long as `fromHSV` sets `inputValue` using `H, S, 100` and then brightness from `V`? The issue was that the values were being scaled to `V` in `fromHSV` and then again in the brightness function.

mcspr, 2019-09-06:
It would, but it was incorrect in other direction :) If target value is multiplied by brightness, we got a different hsv there. Why I focused on web is because colorpicker returns the same value both times, we push back a hsv which it parses properly

CrazyIvan359, 2019-09-06:
In the other direction? Wouldn't that be `toHSV`?

mcspr, 2019-09-07:
It would. Loading webui, it receives hsv value via toHSV. Then, we change it via the wheelcolorpicker to something else, and it sends it back. Receiving side applies it via fromHSV and immediatly reports back via toHSV again, causing colorpicker to update. That's where the rounding issue comes from, since it slightly moves one of the sliders.

CrazyIvan359, 2019-09-07:
Oh I see, WebUI gets its values back from `toHSV` before brightness is applied. Could it be changed to update based on the channels changing and not the command coming in?

PR #1902 (2019-09-10): Lights: properly divide rgb values for hsv conversion
resolves #1864
- fix directly using rgb values instead of using percentages (https://github.com/xoseperez/espurna/issues/1864#issuecomment-528661786) @CrazyIvan359 
- correct naming for kelvin and mired consts
closed 2019-09-10 by mcspr