xoseperez/espurna #1879: Configure Espurna in Shelly1 as Step-Relay

zamby, 2019-08-29:
Hi, I would configure Espurna as relay pulse mode, but I don't get a good result editing in the switch panel.
The schema is follow example:
https://hassiohelp.eu/wp-content/uploads/2018/12/Shelly-20.jpg
Thanks for any help.

wildwiz, 2019-09-08:
Here are the changes to apply to espurna to have it
IMHO that should be the default behavior of the firmware on these devices.
these are the settings I use, this patch is based on commit 43a44cf9 and sets the proper behavior for Shelly1 Shelly2 and Luani (which I submitted myself as default pushbutton but someone later changed to switch mode breaking my original pull request) 
```
input are pushbuttons, not switches

diff --git a/code/espurna/config/hardware.h b/code/espurna/config/hardware.h
index 2bf6f76..c1d6107 100644
--- a/code/espurna/config/hardware.h
+++ b/code/espurna/config/hardware.h
@@ -2638,19 +2638,15 @@
     // Buttons
     #define BUTTON1_PIN             12
     #define BUTTON1_RELAY           1
-    #define BUTTON1_MODE            BUTTON_SWITCH | BUTTON_DEFAULT_HIGH //Hardware Pullup
-
-    #define BUTTON1_PRESS           BUTTON_MODE_NONE
-    #define BUTTON1_CLICK           BUTTON_MODE_TOGGLE
-    #define BUTTON1_DBLCLICK        BUTTON_MODE_NONE
-    #define BUTTON1_LNGCLICK        BUTTON_MODE_NONE
-    #define BUTTON1_LNGLNGCLICK     BUTTON_MODE_NONE
+    #define BUTTON1_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH //Hardware Pullup
+    #define BUTTON1_PRESS           BUTTON_MODE_TOGGLE
+    #define BUTTON1_CLICK           BUTTON_MODE_NONE
 
     #define BUTTON2_PIN             13
     #define BUTTON2_RELAY           2
-    #define BUTTON2_MODE            BUTTON_SWITCH | BUTTON_DEFAULT_HIGH //Hardware Pullup
-
-    #define BUTTON2_CLICK          BUTTON_MODE_TOGGLE
+    #define BUTTON2_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH //Hardware Pullup
+    #define BUTTON2_PRESS           BUTTON_MODE_TOGGLE
+    #define BUTTON2_CLICK           BUTTON_MODE_NONE
 
     // Relays
     #define RELAY1_PIN              4
@@ -3121,8 +3117,10 @@
 
     // Buttons
     #define BUTTON1_PIN         5
-    #define BUTTON1_MODE        BUTTON_SWITCH
+    #define BUTTON1_MODE        BUTTON_PUSHBUTTON
     #define BUTTON1_RELAY       1
+    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
+    #define BUTTON1_CLICK       BUTTON_MODE_NONE
 
     // Relays
     #define RELAY1_PIN          4
@@ -3137,10 +3135,14 @@
     // Buttons
     #define BUTTON1_PIN         12
     #define BUTTON2_PIN         14
-    #define BUTTON1_MODE        BUTTON_SWITCH
-    #define BUTTON2_MODE        BUTTON_SWITCH
+    #define BUTTON1_MODE        BUTTON_PUSHBUTTON
+    #define BUTTON2_MODE        BUTTON_PUSHBUTTON
     #define BUTTON1_RELAY       1
     #define BUTTON2_RELAY       2
+    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
+    #define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
+    #define BUTTON1_CLICK       BUTTON_MODE_NONE
+    #define BUTTON2_CLICK       BUTTON_MODE_NONE
 
     // Relays
     #define RELAY1_PIN          4
```

mcspr, 2019-09-09:
> proper behavior for Shelly1 Shelly2 and Luani (which I submitted myself as default pushbutton but someone later changed to switch mode breaking my original pull request)

just a note, you can use git(hub) [blame](https://help.github.com/en/articles/tracking-changes-in-a-file) feature if you have any questions about the origins of the code:
https://github.com/xoseperez/espurna/pull/1068
https://github.com/xoseperez/espurna/pull/828

Which I assume can be rectified by a cleaned up https://github.com/xoseperez/espurna/pull/1093, instead of changing these back and forth because of conflicting use-cases

closed 2019-11-15 by stale[bot]