xoseperez/espurna #1886: Support hardware Sonoff Mini

Chrisuli, 2019-09-02:
Hi
Sonoff mini will be support soon or not ?
Thanks

pms77, 2019-09-13:
HI,

It's easy (kind of) to put espurna to work.

First you need to flash it OTA. Follow this link and replace the tasmota firmware by the Espurna sonoff basic. (https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY)

Then you can use de espurna dev and just open the arduino.h file and uncomment the line 
//#define ITEAD_SONOFF_BASIC
Then open the hardware.h file and look for the line
#elif defined(ITEAD_SONOFF_BASIC)

Now change the BUTTON2 configuration to:
#define BUTTON2_PIN         4
#define BUTTON2_MODE        BUTTON_SWITCH //| BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON2_RELAY       1

Now in arduino IDE compile for Generic ESP8285 Module and then upload the file OTA.

I've tried and it seems to work (the switch (aka button ON/OFF) can command the relay.

xoseperez, 2019-09-30:
Added support in #50eda3371bad52b246e30e4ccbf542de0c52ee9f

closed 2019-09-30 by xoseperez

cconde, 2019-10-03:
Just stating that I have downloaded a 'dev' firmware and it is working properly on my device.
Thanks!

Bochets34, 2019-11-05:
Good evening,
would you mind please sharing the .bin file for SonOff mini ?
I'm a bit lazy to compile it ...
thanks,
Olivier

mcspr, 2019-11-06:
@Bochets34 please see [Notices in the README](https://github.com/xoseperez/espurna/#notice), dev binaries are released in a separate repo

Bochets34, 2019-11-06:
fantastic, works just fine, many thanks !

JulioSDDR, 2019-11-06:
playing with the nightly release, i have found something weird. At the switches category, you can not select the "button" at pin 4, only the one at pin 12. If only one button is sported, maybe it should be the one at pin 4?  That is the one that is exposed once the sonoff mini is installed