xoseperez/espurna #1895: What version to use PWM ceiling light

henkiejan1, 2019-09-04:
Hi all,

I have a LSC ceiling light with original a Tuya module inside with a realtek module. I have now installed a wemos d1 mini on GPIO 12 and GPIO13. When i flashed ESP Easy on it i discovered that GPIO12 with PWM is used for the brightness with the command http://192.168.2.60/control?cmd=PWM,12,1024 and the colour temperature with GPIO13 with http://192.168.2.60/control?cmd=PWM,13,1024.

But i prefer to use Espurna and think it´s maybe also more simple to integrate with Domoticz. But after a long search on Google it´s for me not clear what variant of Espurna i need to use.

Could anyone help me please?

henkiejan1, 2019-09-06:
Does no one have any idea? I tried the magic home led controller version but didn´t work. The PWM goes from 5 to 1023 what i know.

ColinShorts, 2019-09-06:
Can you post some pics of the device? I'm not personally familiar with, not could find anything useful about "LSC ceiling lights". Do you have any more information on the original tuya module, are you sure it's realtek and not an esp module?

At a push, it sounds as though you only need 2 pwm channels to drive 2 sets of LEDs (warm and cold white). You could probably do that with an existing device, but is be more inclined to remove any additional circuitry and replace it with a couple of MOSFETs, which could then be driven directly without being worried about actual values.
There might be an existing configuration that might work, but you might need to change the pins you are using to match, or change the pins in the configuration.

henkiejan1, 2019-09-06:
LSC is a housebrand of a non-food discounter in the Netherlands. The light is 400LM and is arround €16,-. There are some photo´s of the inside of the lamp on this German forum: https://www.mikrocontroller.net/topic/479796#5952819 BTW i am not the same person who post the photo´s. I have made some detailed photo of the chip.

[image]
[image]

ColinShorts, 2019-09-06:
ITEAD_SONOFF_LED might be the closest configuration to get it working, but you might need to swap pin assignation in the config before bidding.

Hth

closed 2019-11-13 by stale[bot]

Whitebytes, 2019-12-14:
@henkiejan1 Did you get it to work? I would like to control this light with Tasmota, is this possible by installing a wemos d1?

ARGO1960, 2020-03-13:
Hi, I bumped into here from google as I also looked for an way to control it from homeassistant using tasmota.
So yes you meet to replace the existing controller with some other as there is no tasmota support for this RTL chipset. I used an wemos D1 and flashed it with the basic tasmota. (tasmota.bin)
https://github.com/arendst/Tasmota/wiki/Flashing
Set it up to connect to your wifi and choose Generic device (18)

[image]

You need to connect 3.3V and ground and 2 io pins I used D8 and D7.
configure D8 as PWM1 and D7 as PWM2i.
Now you can use it.