xoseperez/espurna #1897: Update req on new ESP Wifi vulnerability: ESP8266 And ESP32 WiFi Hacked!

PieBru, 2019-09-05:
Just a reminder, waiting for Espressif updates for these [vulnerabilities](https://hackaday.com/2019/09/05/esp8266-and-esp32-wifi-hacked/).

th0ma5w, 2019-09-06:
Thank you for your hard work on this project and making sure to put this note here. I can looking for this. Thanks again!!

mcspr, 2019-09-06:
See https://github.com/esp8266/Arduino/issues/6436#issuecomment-528192690
When using `platform = ${common.arduino_core_git}`  you will build with Espressif SDK that has DoS patches included. Next Core point release (2.6.0) will include those.

I wonder if those sdk files could be included on the proper release builds, but the only reason for that right now is saving ~30KB of .bin size. Custom build that I use right now is running git version without WebUI package (~60...80KB), which puts it around the same size as older 2.3.0