xoseperez/espurna #1899: Test web interface

mcspr, 2019-09-06:
**Is your feature request related to a problem? Please describe.**

Following #1898 , it would be useful to have a tests / test mode for WebUI

**Describe the solution you'd like**

https://gitter.im/tinkerman-cat/espurna?at=5d717445ae44a8412487f6c3
One of the options is to have a dummy web-server (node.js is a likely choice) that can WebUI page can interact with.
As an alternative, web page itself should have a special mode for running from `file:///` (current check could be extended)

General test suite for .js code would also be useful.

**Describe alternatives you've considered**
\-

**Additional context**
\-

tonilopezmr, 2019-10-17:
Hi, I'm learning web development and using [Cypress](https://www.cypress.io/) to make E2E testing using screenshot testing, and if we can mock the WebSocket using jest-websocket-mock commented here https://github.com/xoseperez/espurna/pull/1933#issuecomment-543166651, I think its a good way to go to create Web UI tests very easy.

mcspr, 2019-10-20:
Cypress sells itself as a all-in-one thing, which sounds interesting.
However, if I properly understood how it works, in case of interactive stuff the only target browser is Chrome?
https://github.com/cypress-io/cypress/blob/dc251cc5d6c18da2be64ae3de269a00b75985099/packages/launcher/lib/browsers.ts#L6-L31

Re-quoting https://github.com/xoseperez/espurna/pull/1933#issuecomment-543166651, the point of mentioning local webserver was to possibly reuse some of the code in case Arduino & support libraries are mocked / redone to run on the host if that would be useful. And I think I also thought about a way to locally run the WebUI as-would-be on the device, without going about unhiding menus and panels manually.
fwiw the current webui's only entrypoint is the `processData` function (minus `connectTo...`), so local tests could already somehow work if something loads up index.html as `document` and feed data objects to it.

*...and I would say I have not done as much progress on this as I would've liked to...*