xoseperez/espurna #1900: emonCMS DemandShaper Integration

techfixpros, 2019-09-06:
Please provide integration with emonCMS' DemandShaper module. Espurna combined with Sonoff modules higher than the S20 have significantly more features than the the emonESP firmware. As a emonpi/emoncms user, I would like to be able to use DemandShaper functions to automatically load shed with outlets running espurna.

References:

> https://github.com/emoncms/demandshaper

> https://github.com/openenergymonitor/EmonESP/blob/timer/sonoffS20.md

mcspr, 2019-09-09:
Some points of interest:
https://github.com/emoncms/demandshaper/search?q=%24mqtt_client-%3Esubscribe&unscoped_q=%24mqtt_client-%3Esubscribe
https://github.com/emoncms/demandshaper/search?q=%24mqtt_client-%3Epublish&unscoped_q=%24mqtt_client-%3Epublish
https://github.com/emoncms/demandshaper/blob/e23c97ea8198e64df0c716c7126d048d01f6ea01/run.php#L397-L473
If that's how it communicates with the device, integration needs to have the same protocol. Some of code reading can be avoided by using an existing installation and just subscribing to the `emon/#` to see all of the topics going back and forth (or with verbose mode on mqtt broker, seeing what clients publish)

mcspr, 2019-09-11:
Or, looking at ESPEasy's plugin:
https://github.com/letscontrolit/ESPEasy/blob/f7e0d7cfdde65d9503ff08fd92d48681dc8da702/src/_C007.ino
It can also be periodic http requests