xoseperez/espurna #1903: Wifi ceases to function

wildwiz, 2019-09-08:
**Bug description**
WIFI ceases to function. 

The device no longer responds to ICMP pings, ARP resolution, nor shows as associated anymore from the AP perspective.

However, the WIFI LED of the device (on devices having it) continues to blink slowly at the proper rate as if it were properly connected to the configured AP, does not blink as self AP mode.

The correlation between last DHCP renewals and connection loss doesn't seem relevant (several hours most cases)

The device local functions continue to operate, i.e. it responds to button presses, switches relay, operates timers if configured.

Double click to switch to AP mode produces no effect (i.e. the device does publish the AP beacon, didn't check the let blinking tho)
However, long click to reset the device restores normal operations (for how long it lasts)

Uptime from reset to WIFI loss in the range from several hours to few days.

**Steps to reproduce**
Happens erratically

**Device information**
* Sonoff bridge
* Sonoff S20 (4 of them, all behave the same)
* Shelly 1 (4 of them, some seem more affected than others)

** FIrmware information **
Firmwares showing the issue
1.13.3 to 1.13.6-dev
Sonoff S20 reverted to 1.12.6 work fine

**Additional context**
WIFI is served by two MikroTik APs sharing the SSID and backing up each other, 
Signal for all devices is in decent range (RSSI above -80)

mcspr, 2019-09-09:
From a week ago - #1877 - maybe this could help in any way? But I'd think arping from some other device would do the same thing
I also wonder if terminal connection can shed some light on what device is trying to do while this thing happens, does this mean that IP is lost but the AP connection is still present?
Or could it be that it breaks somehow with selecting a proper AP? (since they use the same SSID)

#614 / https://github.com/esp8266/Arduino/issues/2330 is a pretty large collection of all connectivity problems. 2 solutions stand out though, garp and icmp pings from the device. Garp is enabled via the PR above. ICMP is here:
https://github.com/mcspr/espurna/blob/ping-sensor/code/espurna/sensors/PingSensor.h, since I have not yet merged it
https://github.com/d-a-v/EspGoodies/blob/master/src/PingAlive.h and https://github.com/d-a-v/EspGoodies/tree/master/examples/pingAlive as a poc that came from the esp8266/arduino issue

mcspr, 2019-09-09:
Also see https://github.com/esp8266/Arduino/pull/6484, as just mentioned in the upstream issue.
Preferably, with our garp workaround disabled by setting `set wifiGarpIntvl 0` in the settings

remygiraud, 2019-09-16:
Hi,
I am having the same kind of behaviour on all my sonoff devices. I don't understand the suggestion with "set wifiGarpIntvl 0"... What should I do with that? 
Thanks,

mcspr, 2019-09-16:
`wifiGarpIntvl` is a setting name, `0` disables it. It is enabled by default in dev build.

If you change Arduino Core version here:
https://github.com/xoseperez/espurna/blob/cf3bd49bdecb137d206755df0e0712bc39a7f53f/code/platformio.ini#L32
to ${common.arduino_core_git}, I suggest to run the `set wifiGarpIntvl 0` in the terminal (and then `reload` the settings or reboot the device), because the issue is presumably fixed and the workaround is no longer required.

wildwiz, 2019-09-16:
An update,
I have attached a raspberry to the sonoff rf bridge, in order to capture all the serial output and periodically cast info and wifi commands.

Once the device got "disconnected", according to the wifi info the device still believed to be connected. Unfortunately attempting to wifi.reset caused a crash of the device, but it could have been me casting the wifi command while it was still setting things up again.

I am waiting for next disconnection to further analyze other aspects like arp tables, response to manual arp on the other side and so on, once I have enough for an analysis I will post it here.

wildwiz, 2019-09-16:
However, what make me most curious is that the s20 are working perfectly on 1.12.6 and were disconnecting often after just few hours with 1.13.3 and 1.13.6-dev. I bet that if I had 1.12.6 on shellies they would stay connected too (but they have been integrated later so we cannot test that)

mcspr, 2019-09-16:
Thank you!

The way I understood the problem solved by Core GIT and GARP workaround, is that it would stay on the wifi, but IP address will no longer be assigned. Decoded stack dump would probably show something about IP manipulation or anything related to displaying things.

As to why, crucial part of 1.13.3 was upgrade to JustWifi v2: https://github.com/xoseperez/justwifi
I would presume that it may be related to some reordering of connect / disconnect / mode-change calls, causing some weird state in some situations.

remygiraud, 2019-09-16:
Hello,
I am running espurna 1.13.1 on all of my sonoff dual r2 and I am having these "random" disconnections. They are more frequent since I move my wifi network from a single Netgear R7000 AP to a mesh of 3 Nove Tenda MW6. Coverage is better, but, more disconnection on espurna...

wildwiz, 2019-09-25:
Hi, I am back after trying with firmware based on more recent deb4657 as suggested.
The data I am providing is captured on a sonoff rf bridge, I have sonoff s20 that perform badly as well, but being connected to AC I have avoided extracting serial console from those.

The build from the new commit point deb4657 didn't bring any improvement nor in the bridge nor in the S20s. However, I have been able to gather a full log from reset to after connection loss (next comment is the log, shortened to keep the relevant items).

I have also assessed the following behaviors after connection loss:
- Manually adding a static arp to the dns/dhcp/mqtt/ntp server for the device, the device still remain unresponsive. i.e. the problem is not just a missing arp route, there is something worse in the device status.
- wifi.scan command casted from serial works and respond timely and properly.
- wifi.reset command consistently crashes after a couple of seconds.
- wifi commands shows the IP address and BSSID connected after connection loss
- After DHCP expiration, IP address will finally go to 0.0.0.0 in wifi response

wildwiz, 2019-09-25:
Device log from reset to connection loss to attempt to reset wifi
The log has the following sections:
- from reset to first info/wifi periodic outputs and log of NTP talking to the lan
- last working NTP/MQTT operations and corresponding info/wifi periodic outputs
- first indication of connection failure (NTP not reachable anymore and corresponding output of wifi command and output of info/wifi commands just after that
- later wifi command showing the expired IP
- attempts to wifi.scan / wifi.reset to revive the connection
- stack trace of the crash happening on wifi.reset
```
[071764] [MAIN] ESPURNA 1.13.6-dev (8fe495e)
[071764] [MAIN] xose.perez@gmail.com
[071764] [MAIN] http://tinkerman.cat
[071765] [MAIN] CPU chip ID: 0xXXXXXX
[071768] [MAIN] CPU frequency: 80 MHz
[071772] [MAIN] SDK version: 1.5.3(aec24ac9)
[071776] [MAIN] Core version: 2.3.0
[071779] [MAIN] Core revision: 9826c6d
[071783] [MAIN] Build time: 1569166160
[071787] [MAIN] Flash chip ID: 0x144051
[071790] [MAIN] Flash speed: 40000000 Hz
[071793] [MAIN] Flash mode: DOUT
[071797] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[071805] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[071812] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[071819] [MAIN] Firmware size       :   511552 bytes /  125 sectors (   1 to  125)
[071826] [MAIN] Max OTA size        :   512000 bytes /  125 sectors ( 126 to  250)
[071833] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[071840] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[071849] [MAIN] EEPROM sectors: 251, 250
[071852] [MAIN] EEPROM current: 250
[071856] [MAIN] EEPROM:  4096 bytes initially |  2272 bytes used (55%) |  1824 bytes free (44%)
[071865] [MAIN] Heap  : 34144 bytes initially | 17016 bytes used (49%) | 17128 bytes free (50%)
[071873] [MAIN] Stack :  4096 bytes initially |  1396 bytes used (34%) |  2700 bytes free (65%)
[071882] [MAIN] Boot version: 6
[071886] [MAIN] Boot mode: 1
[071887] [MAIN] Last reset reason: Power on
[071891] [MAIN] Last reset info: flag: 0
[071915] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[071919] [MAIN] Sensors: DHTXX 
[071922] [MAIN] WebUI image: FULL
[071926] [MAIN] Firmware MD5: 80727d8aa5f96629dd4dcf00e00e143e
[071932] [MAIN] Power: 3226 mV
[071934] [MAIN] Power saving delay value: 10 ms
[071940] +OK
[071953] [WIFI] ------------------------------------- MODE STA
[071953] [WIFI] SSID  xxxxxxxx
[071953] [WIFI] IP    10.0.0.219
[071953] [WIFI] MAC   60:01:94:xx:xx:xx
[071956] [WIFI] GW    10.0.0.254
[071959] [WIFI] DNS   10.0.0.251
[071962] [WIFI] MASK  255.255.255.0
[071965] [WIFI] HOST  http://br433.local
[071969] [WIFI] BSSID B8:69:F4:xx:xx:xx
[071972] [WIFI] CH    11
[071975] [WIFI] RSSI  -60
[071977] [WIFI] ----------------------------------------------
[071982] +OK
[977354] [NTP] UTC Time  : 2019-09-22 14:35:57
[977354] [NTP] Local Time: 2019-09-22 16:35:57
...
[070810] [MAIN] ESPURNA 1.13.6-dev (8fe495e)
[070810] [MAIN] xose.perez@gmail.com
[070810] [MAIN] http://tinkerman.cat
[070812] [MAIN] CPU chip ID: 0xXXXXXX
[070815] [MAIN] CPU frequency: 80 MHz
[070818] [MAIN] SDK version: 1.5.3(aec24ac9)
[070822] [MAIN] Core version: 2.3.0
[070825] [MAIN] Core revision: 9826c6d
[070829] [MAIN] Build time: 1569166160
[070833] [MAIN] Flash chip ID: 0x144051
[... 163 lines cut ...]
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
~ld

0x4024a9df: ieee80211_send_probereq at ??:?
0x4025624e: phy_after_init_enrx at ??:?
0x4024c657: scan_remove_probe_ssid at ??:?
0x40252f20: chip_v6_set_chanfreq at ??:?
0x401004d8: malloc at .platformio/packages/framework-arduinoespressif8266@1.20300.1/cores/esp8266/umm_malloc/umm_malloc.c:1664
0x4024c30c: scan_pm_channel_op_cb at ??:?
0x4024c2e0: scan_pm_channel_op_cb at ??:?
0x4024e0ad: chm_start_op at ??:?
0x4024c2e0: scan_pm_channel_op_cb at ??:?
0x40243079: sys_timeout at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c:296
0x4024e074: chm_start_op at ??:?
0x4024c2d6: scan_pm_channel_op_cb at ??:?
0x402432c1: sys_check_timeouts at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c:420
0x4024c27d: scan_pm_channel_op_cb at ??:?
0x4024c734: scan_clear_channles at ??:?
0x40201750: delay_end at .platformio/packages/framework-arduinoespressif8266@1.20300.1/cores/esp8266/core_esp8266_wiring.c:40
0x40238719: ets_timer_handler_isr at ??:?
0x4023875e: ets_timer_handler_isr at ??:?
0x4022fdcf: loop_task(ETSEventTag*) at .platformio/packages/framework-arduinoespressif8266@1.20300.1/cores/esp8266/core_esp8266_main.cpp:130
```

mcspr, 2019-09-25:
Have you tried running with `platform = ${common.arduino_core_git}`?
Is this setup with mesh APs, like message above? And I also remembered that with `-DDEBUG_ESP_WIFI` it would enable some additional logging from the Core.
`Connecting to SSID: xxxxxxxx` message comes from ESPurna handling JustWifi callbacks. Stack trace does not show anything though. Perhaps manual tracing could help there. simply putting a bunch of Serial.printf(...) s [after this point](https://github.com/xoseperez/justwifi/blob/8bb2c16f5b91f56de191cb8ff96a816fb672cfd2/src/JustWifi.cpp#L251)

Crash log looks weird, because it stays entirely in the SDK part. Recent Core supposedly fixed some interactions with SDK related to the interrupt handling, which caused a lot of crashes / random resets / seemingly random issues related to WiFi state management. I would try that first, but then we should probably find some faster way to crash instead of waiting for an hour (*...aluminium foil?*)

mcspr, 2019-09-27:
And a small note regarding networks where AP have the same SSID.
As I am not around any esp boards atm, I tried mocking out WiFi layer to run JustWifi locally and test state machine. What is interesting is that it seems to select the worse one out of two available.
https://github.com/xoseperez/justwifi/blob/8bb2c16f5b91f56de191cb8ff96a816fb672cfd2/src/JustWifi.cpp#L172
```
$ ./jwtest
jw msg=MESSAGE_SCANNING data=NULL
jw void _populate(2)
jw msg=MESSAGE_FOUND_NETWORK data=--> BSSID: 11:12:13:14:15:20 CH:  6 RSSI:  60 SEC: WPA2 SSID: TEST
jw msg=MESSAGE_FOUND_NETWORK data=--> BSSID: 11:12:13:14:15:10 CH:  6 RSSI:  50 SEC: WPA2 SSID: TEST
jw void JustWifi::_sortByRSSI()
jw msg=MESSAGE_CONNECTING data=BSSID: 11:12:13:14:15:20 CH: 06, RSSI:  60, SEC: WPA2, SSID: TEST
```

mcspr, 2019-10-01:
Regarding Tenda MW6 problem @remygiraud 
https://github.com/esp8266/Arduino/issues/4735
https://github.com/arendst/Sonoff-Tasmota/issues/4915
It was confirmed that using newer Core version solves the problem. Binary releases are still using 2.3.0 for compatibility / size reasons, but you can build the firmware yourself and specify platform = ${common.arduino_core_2_5_2} (as also explained above)

@wildwiz please try out the ESPAsyncTCP library change I suggested (https://github.com/mcspr/espurna/commit/de8c8e703b52be8a9a9b24d6a4aad0418918361e#diff-25ef070b6156fdfa35dd363916f0f98aL109) if you have a lot of devices on the same LAN
which Mikrotik model do you use?

wildwiz, 2019-10-05:
Access points are Mikrotik HAP AC running RouterOS v6.45.2 - I will follow up for the other points