xoseperez/espurna #1904: Alexa unreliability - Sonoff Basic - Alexa DOT series 3 (as example)

scargill, 2019-09-08:
Even with the nightly build of 4 nights ago, for Sonoff Basic - I'm getting unreliable Alexa use. I have set the hostname to sandy, enabled Alex integration and set the alexa name to sany. It worked for a couple of day then today, Alexa could not find sandy. It DID find "ESPURNA xxxxx" once, I tried rnaming that in Alexa to sandy, didn't work, tried deleting the item out of the alexa app and I'm back to square 1. Alexa will not find sandy.

**Bug description**
Alexa reliability

**Steps to reproduce**
Set up a Sonoff Basic with the latest nightly build... 
espurna-1.13.6-dev.nightly20190904+git24788474-itead-sonoff-basic.bin
Tell Slexa to "find devices" - when it succeeds, tell Alexa to turn sandy on or off

**Expected behavior**
I would expect Espurna to regognise sandy and turn it on and off.

**Screenshots**

**Device information**

[025262] [WEBSOCKET] #1 connected, ip: 192.168.1.31, url: /ws
[028350] [WEBSOCKET] Requested action: relay
[028351] [RELAY] #0 scheduled ON in 0 ms
[028359] [RELAY] #0 set to ON
[028361] [MQTT] Sending sandy/relay/0 => 1 (PID 1)
[029362] [RELAY] Setting relay mask: 1
[029725] [WEBSOCKET] Requested action: relay
[029726] [RELAY] #0 scheduled OFF in 0 ms
[029734] [RELAY] #0 set to OFF
[029736] [MQTT] Sending sandy/relay/0 => 0 (PID 1)
[030738] [RELAY] Setting relay mask: 0
[060003] [MAIN] System OK
[115274] [WEBSOCKET] Requested action: relay
[115275] [RELAY] #0 scheduled ON in 0 ms
[115284] [RELAY] #0 set to ON
[115285] [MQTT] Sending sandy/relay/0 => 1 (PID 1)
[116288] [RELAY] Setting relay mask: 1
[116538] [WEBSOCKET] Requested action: relay
[116538] [RELAY] #0 scheduled OFF in 0 ms
[116551] [RELAY] #0 set to OFF
[116553] [MQTT] Sending sandy/relay/0 => 0 (PID 1)
[117554] [RELAY] Setting relay mask: 0
[320584] [MQTT] Sending sandy/app => ESPURNA (PID 1)
[320585] [MQTT] Sending sandy/version => 1.13.6-dev (PID 1)
[320586] [MQTT] Sending sandy/board => ITEAD_SONOFF_BASIC (PID 1)
[320590] [MQTT] Sending sandy/host => sandy (PID 1)
[320594] [MQTT] Sending sandy/ssid => wififorus (PID 1)
[320599] [MQTT] Sending sandy/ip => 192.168.1.115 (PID 1)
[320604] [MQTT] Sending sandy/mac => 84:0D:8E:56:BB:95 (PID 1)
[320609] [MQTT] Sending sandy/rssi => -50 (PID 1)
[320614] [MQTT] Sending sandy/uptime => 320 (PID 1)
[320621] [MQTT] Sending sandy/datetime => 2019-09-08 17:11:28 (PID 1)
[320626] [MQTT] Sending sandy/freeheap => 21344 (PID 1)
[320629] [MQTT] Sending sandy/relay/0 => 0 (PID 1)
[320635] [MQTT] Sending sandy/vcc => 3126 (PID 1)
[320639] [MQTT] Sending sandy/status => 1 (PID 1)
[320643] [MQTT] Sending sandy/loadavg => 1 (PID 1)
[620587] [MQTT] Sending sandy/app => ESPURNA (PID 1)
[620588] [MQTT] Sending sandy/version => 1.13.6-dev (PID 1)
[620589] [MQTT] Sending sandy/board => ITEAD_SONOFF_BASIC (PID 1)
[620592] [MQTT] Sending sandy/host => sandy (PID 1)
[620597] [MQTT] Sending sandy/ssid => wififorus (PID 1)
[620601] [MQTT] Sending sandy/ip => 192.168.1.115 (PID 1)
[620606] [MQTT] Sending sandy/mac => 84:0D:8E:56:BB:95 (PID 1)
[620612] [MQTT] Sending sandy/rssi => -51 (PID 1)
[620616] [MQTT] Sending sandy/uptime => 620 (PID 1)
[620624] [MQTT] Sending sandy/datetime => 2019-09-08 17:16:28 (PID 1)
[620629] [MQTT] Sending sandy/freeheap => 21344 (PID 1)
[620632] [MQTT] Sending sandy/relay/0 => 0 (PID 1)
[620638] [MQTT] Sending sandy/vcc => 3126 (PID 1)
[620642] [MQTT] Sending sandy/status => 1 (PID 1)
[620647] [MQTT] Sending sandy/loadavg => 1 (PID 1)

I am looking at this on Windows 10 in a Chrome browser. I use MQTT on a Raspberry Pi 4. I can turn the Sonoff Basic on and off via  its button and monitor that in Espurna ewb interface, or turn on and off via the web interface, or by MQTT from the RPI.

wildwiz, 2019-09-08:
is it possible that this is due to changes in echo dot firmwares ?
https://github.com/originallyus/node-red-contrib-alexa-local/issues/100
in my case all the LAN autodetection and commands ceased to work a couple of days ago and the same moment some feedback sounds changed as well, suggesting a firmware update.
For about a day I was also unable to forget devices in alexa, they would continue to be rediscovered at next scan with nothing on the lan with the offending maes

scargill, 2019-09-08:
That sounds about right - question is - what to do about it...

scargill, 2019-09-09:
Whatever it is- here's some hopefully useful feedback - it isn't affecting Node-Red-Contrib-Amazon-Echo because after struggling with Espurna no longer talking to My Echo R3 (can't find Sandy), I JUST used Node-Red-Contrib-Amazon-Echo - added a device called Sandy and fed the output as usual through a function node to MQTT - as my device needs 1 and 0 as against on and off. It seems to be working perfectly - yet only minutes earlier, Echo could not find my little Espurna equipped "sandy" board (the same board). I made no changes to the board or Espurna.

scargill, 2019-09-09:
Right I'm having a rethink about this. Node-Red finds and talks to device "sandy" no problem (after I created it in Node-Red-Amazon-Alexa). Meanwhile back in the very same Sonoff Basic device running the Espurna nightly build of September 4..... if I go to the web page "http://sandy" - it logs in no problem. Status click ON or OFF - device turns on and off.... no problem... Chrome refresh no problem. GENERAL - leave hostname at sandy.... turn odd Alexa integration. SAVE - no problem. REFRESH - it has GONE - WHY? close browser window - open - "NOT SECURE - SANDY but browser window is empty. Tab says ESPURNA 0.0.0 - empty browser window.  Physically turn off Sandy - and back on,, Seconds later it is back up and running -  under GENERAL - Alexa integration is indeed disabled. Something wrong THERE. Power should no need cycling.

scargill, 2019-09-10:
I'll rephrase that - N.R.C.A.E. is working but finding devices twice...  this is definitely looking like an Amazon issue.

mcspr, 2019-09-10:
Ok, is there an issue with integration or not? I am not really following what is wrong here.

Right now device name is set on boot, so I'd assume you need to reboot before searching for the new name. Also note of the `alexaName` override, you can set this key in settings to avoid changing the device's `hostname`.
You can enable / disable Alexa via settings too, by changing `alexaEnabled` to 1 / 0

There was an earlier issue with device unique ids (or it is how I understood it):
https://github.com/xoseperez/espurna/issues/1738
Which was fixed all by itself, but other Alexa integrations had noticed that Hue protocol does some weird things with MAC<->DeviceID associations:
https://github.com/Aircoookie/Espalexa/commit/9d57e2cbc706c503e71acc651ed47e6e09496614#diff-60d0a10c47ca0f0800ce9f6773faec4aR166
Since we are too using Hue via https://bitbucket.org/xoseperez/fauxmoesp and it does not do that, perhaps it may be the cause of some of the discovery issues

This is all just guesses though, nothing definitive.
If someone is willing to investigate further, that would be great:)

scargill, 2019-09-10:
mcspr - help an ESPURNA newby here.. you refer to "settings" - which settings ? In the Espurna 1.13.6-Dev web interface I can see status, general, domoticz, hass, led, mqtt, ntp, schedul, switches, thingspeak, wifi, admin and debug - but no "settings". Alexa enabled and slexa name are set in "general" and yes you are told you need a reboot. In one instant I also had to cycle power.  Unless of course you are referring to that entire menu as "settings".

mcspr, 2019-09-10:
Yes, those are settings, but we can also use terminal commands to do the same thing:
https://github.com/xoseperez/espurna/wiki/Terminal#settings

tyriis, 2019-09-12:
FYI: i am not eure about the Alexa integration I think, ACK is used to integrate the devices to Alexa.

It seems that this integration was a developer preview. Amazon decided to shut down this functionality. See https://developer.amazon.com/blogs/alexa/post/6d7c726f-eff3-4d0c-95d2-63ced3951263/introducing-the-alexa-connect-kit-connect-devices-to-alexa-without-managing-a-cloud-writing-an-alexa-skill-or-developing-complex-networking-and-security-firmware-apply-today-for-the-preview comment on the top.

scargill, 2019-09-22:
I have now grabbed espurna nightly build from Sept 21 2019 so I am up to date.  Alexa has found one of my Sonoff devices and also now works just fine with node-red-contrib-amazon-echo.. However, I just took my NWELY SET UP Espurna, gave it an alexa name of susan and enabled alexa on the Sonoff basic.... and it is not being discovered, Is there still an active issue with Espurna and Alexa?

scargill, 2019-09-22:
> I'll rephrase that - N.R.C.A.E. is working but finding devices twice... this is definitely looking like an Amazon issue.

That WAS in fact a having 2 raspberry pis talking to the same devices issue - ignore the finding devices twice issue. But wee my new comment below - 22/9/2019, yesterdays Espurna build for sonoff basic... not finding the device.

mcspr, 2019-09-23:
No changes concerning Alexa here...

scargill, 2019-09-23:
So, new device "margaret" this morning. Espurna Sonoff Basic nightly build 21/09/2019 (on a new sonoff- Alexa could not find device "margaret". If I disable device "margaret" in the Sonoff - and add device "margaret" to node-red-contrib-alexa-echo with MQTT out to the real device, all of that works - but actually adding "margaret" elexa compatibiliyt to the Sonoff does not. It USED to work.

mcspr, 2019-09-23:
Like I said, nothing changed in Alexa integration here. I can't test this, so if someone else is also having this problem my advice would be to check out if unique ids work (see message above) and maybe even compare the traffic between fauxmoESP and node-red-contrib-alexa-echo

*Have you tried male names? :)*

scargill, 2019-09-23:
Well, here's a thing. I have a Sonoff Basic R3 called "office wall"- and my Alexa is DOT R3.  Alexa talks to that board directly - but the original sonoffs - I have to use node-red-contrib-amazon-echo as they won't talk directly to Alexa (cannot be discovered) - yet these same boards WERE talking directly to Alexa using Tasmota (I'm moving over to Espurna one board at a time). These boards are on-off only so I don;t understand what IDs have to do with anything, sorry.

fcoaaron, 2020-01-21:
Hi! I'm new using Alexa, I've an Echo Dot Gen3 and was trying to power on/off my Sonoff Pow with latest firmware (1.14.1). The wiki info for Alexa is a bit weak, but in fauxmoesp bitbucket I found this:

"(1) When using gen3 devices TCP port must be 80 always. (2) Arduino Core for ESP8266 requires LwIP set to "v1.4 Higher Bandwidth"."

Indeed, I found that HTTP port in the Admin settings MUST be in port 80 in order to work with Alexa. I found also this:

"fauxmoESP 3.1.X: When using with gen3 devices TCP port must be 80. You can define it with the setPort method."

Then, it might be possible to set different port for fauxmoesp and for HTTP? As I have several Sonoff devices, if I set all to port 80 for working with Alexa, I won't be able to control each one with REST API... unless I find some way to redirect ports.