xoseperez/espurna #1908: 4CH relay module ESP

z4rn0x, 2019-09-11:
Hello guys,
I'd like to use a 4 channels relay for my garage door, and I'm wondering if this type of equipment could work with the ESPURNA firmware: 
[https://www.aliexpress.com/item/32963978306.html?spm=a2g0o.productlist.0.0.550bb4506in1nB&algo_pvid=fb10b88e-e18c-4e7f-8168-f4f25e5472f7&algo_expid=fb10b88e-e18c-4e7f-8168-f4f25e5472f7-3&btsid=9839488a-e5db-40b0-8a60-a5affeee2e00&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_52](url)
I haven't ordered it yet...

ColinShorts, 2019-09-11:
It looks like it's using an [external chip](https://community.home-assistant.io/t/4-channel-relay-esp8266-wifi-board-working-with-esphome/120594) to drive the relays. I suspect that the RELAY_PROVIDER_STM with 4 dummy relays **_might_** just work.

z4rn0x, 2019-09-11:
Sounds good... I ordered one, hopefully I'll be able to make it through...

closed 2019-11-18 by stale[bot]