xoseperez/espurna #1911: Ef

Carmelo66, 2019-09-15:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

Carmelo66, 2019-09-15:
I have a problem with my Sonoff bridge Rf with espurna 1.13.5 and connected to the gate of my house, when it is powered and connected to the network it makes me open my gate.

mcspr, 2019-11-11:
*Sorry for the delay, I was indisposed when this was posted!*

See if you have some relays configured to be on on boot via the debug panel and using `get relayBoot0` command, replacing 0 with 1...7 for each relay. Otoh I think this is already fixed since the 1.13.5, WebUI was not properly displaying the boot modes. Please try out the development build instead.
https://github.com/mcspr/espurna-nightly-builder/releases

closed 2019-11-11 by mcspr