xoseperez/espurna #1912: Teckin SP22 new layout

rikthefrog, 2019-09-15:
Hi,

I have a Teckin SP22 with a new board layout. It is the EU version and bought by Amazon the 12th of September.
[image]

I'm really curios to find the pin layout. Is there any way I can check/test it?
[image]

The back of the TYWE2S board looks like this 
[image]

This is maybe a beter layout picture.
[image]

Does anyone an idea or can give me tips?

mcspr, 2019-09-15:
Have you removed the shield from the Tuya module? The crab on the chip means it is Realtek:
https://www.realtek.com/en/products/communications-network-ics/item/rtl8710bn
And Tuya does indeed manufacture boards based on RTL8710BN too, in addition to the ESP8266, supplying manufactures with an app SDK
https://fccid.io/2ANDL-WR4/User-Manual/User-Manual-4015611 (also named WR2 and WR3)

But, we can't run espurna on Realtek hardware if that's the case :/

closed 2019-11-11 by mcspr

Blackest-Nighthawk, 2022-11-21:
Look at this:
https://github.com/ct-Open-Source/tuya-convert/issues/269

Maybe it helps.

chetank288, 2023-05-07:
bps101 carkit bord