xoseperez/espurna #1919: Solution : Support for Generic Smart Plugs

varungujjar, 2019-09-22:
[image]
[image]

Just beed on this since 3 days now and finally figured out the pins.. hope this helps someone.. :) successfully flashed firmware with following config 

Also this works with the SONOFF_SV Config  in hardware.h file

#define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_SV"

    // Buttons
    #define BUTTON1_PIN         1
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          14
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1

wildwiz, 2019-09-22:
Looks like it is a Tuya's TYW-E2S ESP8285 WI-FI module, same as I found time ago in the Zhilde power strip about an year ago https://github.com/xoseperez/espurna/wiki/Hardware-Zhilde-ZLD-44EU-W On that page there are some links and some more details on how to program the TYW-E2S

varungujjar, 2019-09-23:
No it's a esp8266 confirmed and successfully.flashed sonoff SV firmware and working fine

varunquartic, 2019-09-23:
> Looks like it is a Tuya's TYW-E2S ESP8285 WI-FI module, same as I found time ago in the Zhilde power strip about an year ago https://github.com/xoseperez/espurna/wiki/Hardware-Zhilde-ZLD-44EU-W On that page there are some links and some more details on how to program the TYW-E2S

yes the module is the same.. may be we can document it..  for this plug as well.?

ColinShorts, 2019-09-23:
The esp8285 is an esp8266 with flash memory built in.

varungujjar, 2019-09-23:
> The esp8285 is an esp8266 with flash memory built in.

got it just checked the post  that xose posted :)

varungujjar, 2019-09-23:
> Looks like it is a Tuya's TYW-E2S ESP8285 WI-FI module, same as I found time ago in the Zhilde power strip about an year ago https://github.com/xoseperez/espurna/wiki/Hardware-Zhilde-ZLD-44EU-W On that page there are some links and some more details on how to program the TYW-E2S

maybe we can document this.. ? and add it to the hardware page ?

mcspr, 2019-09-23:
> maybe we can document this.. ? and add it to the hardware page ?

Please do. You can specify the seller or some board markings. I do see something near the ESP board, photo needs another angle :)