xoseperez/espurna #1922: E1.31 (Streaming ACN) Protocol Support

CalcProgrammer1, 2019-09-23:
I've been using Espurna on some RGB light bulbs (AiLight) and was planning to use these bulbs as part of my Halloween/Christmas light setup this year.  The majority of my lights are WS28xx pixels controlled with ESPixelStick firmware on DIY ESP-01 breakout boards.  This firmware uses E1.31 to receive color data from PC sequencing software.

I'm wondering if Espurna could support this protocol as well.  ESPixelStick uses this E1.31 library:

https://github.com/forkineye/ESPAsyncE131

My thought on this would be a configuration page where you select which universe to listen to and then which channel for each output.  Channel values are 0-255.  For relays, could just be zero = off, nonzero = on.

CalcProgrammer1, 2019-09-25:
I'm taking a shot at adding support myself.  The ESPAsyncE131 library looks very simple to use.  I have it receiving packets from the Vixen sequencing software in unicast mode but all it's doing so far is printing a debug message when it receives a packet.  I tried in multicast mode (which is how I'm driving my ESPixelSticks) and it did not seem to receive successfully so I need to do more investigating.

mcspr, 2019-09-25:
Also note that ESPixelStick builds from git version of Arduino Core:
https://github.com/esp8266/Arduino
https://github.com/forkineye/ESPixelStick

Which may cause some subtle differences
https://github.com/forkineye/ESPixelStick/blob/178a2e90fe567cf5af986499931308b000b745d6/.travis.yml#L31

CalcProgrammer1, 2019-09-25:
I had to update Arduino Core to make it build if I remember correctly.  I had build issues, wiped, and rewrote parts of my code so I'll have to double check that it's still using the newer version.  It also puts the firmware.bin over the OTA upload size.

I was able to get it setting the four light channels on an E1.31 packet last night so next I'm going to try to make it configurable in the web UI.  Maybe something like this:

[image]

mcspr, 2019-09-25:
Sort of like that. idk if lights module specifically needs to know about e131 presence, right now that requirement is only for data output. Can we just parse the packet and handle all of it right there?

https://github.com/xoseperez/espurna/compare/dev...mcspr:experimental/tuya (ref [light.ino](https://github.com/mcspr/espurna/blob/experimental/tuya/code/espurna/light.ino) and [tuya.ino](https://github.com/mcspr/espurna/blob/experimental/tuya/code/espurna/tuya.ino) was a first-try to implement externally controlled channels for double-MCU Tuya devices. I have not merged it yet, but this might be the time to port some thing from there or re-do them properly. For one, properly distinguish sources of changed values (web, protocol etc.)

CalcProgrammer1, 2019-09-25:
In my test code I have it all driven from e131.ino but I was trying to see how the other modules handled light control and did not see references to the light control functions outside of lights.ino.  If it's acceptable to call the light control functions from e131.ino then that would make my life easier.

CalcProgrammer1, 2019-09-26:
I managed to get multicast mode working.  It looks like you can set unicast mode in the setup function but not multicast mode.  You have to wait for WL_CONNECTED.  I modified my code to set an initialized flag to false in the setup function and then begin the e131 multicast mode when the WiFi is connected, clearing the flag.  It is working, but I'm noticing it is rather slow to update.  I've set CPU frequency to 160MHz and tried increasing the buffer size.

CalcProgrammer1, 2019-09-27:
I've added options to configure the E1.31 protocol to the web interface.  I basically implemented everything from my mockup for lights 0-4.  I'm not sure if there's a better way to populate the lights, just hard coded 5 channels for now.  This should be mostly usable now that you can configure multiple devices on different channels.

https://github.com/CalcProgrammer1/espurna/commit/f4bae69597783d66b929885215ec02172fc701f5

CalcProgrammer1, 2020-01-09:
I got the functionality working, but the code is a mess and probably should automatically detect channel count rather than just hard code it.  I needed this functionality working for Halloween, as I was using it to drive lights in cauldron decorations connected to an E1.31 light show.  After Halloween I've been busy with other projects and haven't had time to clean my code up to a merge-worthy state.  If anyone else wants to clean up my code and get it merged feel free to do so, otherwise I'll try to come back to it when I have downtime on my other project (an RGB control app and music visualizer for PC peripherals, also supporting E1.31).

CalcProgrammer1, 2020-09-15:
I'm going to submit a PR for this soon, hopefully this week.  I just started fresh and pulled my old E1.31 stuff into the latest dev branch.  I have it building but had to really cut down my image to OTA, so I'll do a two-step OTA soon with the full image.  The image I have now is working on my AI Thinker bulb receiving E1.31 smoothly from my OpenRGB app.

mcspr, 2020-09-15:
*OT about "I have it building but had to really cut down my image to OTA"
See https://arduino-esp8266.readthedocs.io/en/2.7.4_a/ota_updates/readme.html#compression*

CalcProgrammer1, 2020-09-16:
New branch: https://github.com/CalcProgrammer1/espurna/tree/feature_e131

It's working, including the Web UI

[image]

and I've cleaned up the code a lot since the old branch.

The only thing left is that I don't know if I have all my compile flags in the right place so I want to do more testing before opening a PR.  I have flags for enabling E1.31 and reliance on web support and light.