xoseperez/espurna #1931: Restore backup to a newly flashed device

cconde, 2019-10-03:
**Is your feature request related to a problem? Please describe.**
After messing with firmware I tend to have to restore a backup. 
In order to restore a backup I have first to reset the default password and then go to "Admin" to restore the backup.

**Describe the solution you'd like**
It would be superb to have a "Set password & restore config" button at that very first page.

**Describe alternatives you've considered**
I don't know if it's possible, it would be totally awesome if the restore would also restore the original password.

**Additional context**
I keep in mind this may only be useful to few people.

Thanks for this great work!

mcspr, 2019-10-11:
On a sort-of related note, the starting page could use WiFi setup as well.

You can also skip WebUI completely, see:
https://github.com/xoseperez/espurna/wiki/Configuration#direct-http-settings-api
This will overwrite configuration using the current password

Tofandel, 2019-10-17:
Was thinking about improving the first step as well for the UI, it's a tad annoying

Isn't the password stored in the config backup?
Cause then you can just make it

Backup or (Password and Wifi)

And then connect to the device via local network instead of the usual
Change password => reconnect => setWifi => reconnect again

Tofandel, 2019-10-19:
Also it's possible to skip the change password step by setting the build flag WEB_FORCE_PASS_CHANGE to 0

sfilinto, 2020-01-27:
i reset my devices sometimes so in order to bring them to a desired state very quickly i have made 2 .JSON files which have the desired configuration items. i run the below command to configure the device `curl -F "file=@1.json" -u admin:fibonacci http://192.168.4.1/config` and it take a couple of seconds.
1st JSON file changes the default password & adds my wifi ssid & password.
login with a browser and reboot the device.
2nd JSON file has all the desired configuration and hostname.
again i login and reboot the device.

entire procedure take under 5 minutes to bring a device to a desired state & also avoids spelling mistakes.
HTH.

Tofandel, 2020-01-27:
@sfilinto if you want there is a trick to do that in one step even faster, use `jq` to merge your two json files send the output and tadam done