xoseperez/espurna #1934: Xiaomi Desklamp Rotary Encoder Bugs

fluffymadness, 2019-10-07:
There are some weird issues with my xiaomi desklamp
I upgraded to the latest nightly that is linked via github
and the rotary encoder behavior is completely broken
when i try to dim down in dims down a bit and is then full brightness again
same with color change
when i turn the wheel to the right it changes between yellow and white gradually
worked fine on older builds
newest tasmota with the xiaomi desklamp module has the same issue for me

(espurna-1.13.6-dev.nightly20191001+gitbd385f9f-xiaomi-smart-desk-lamp) broken
https://github.com/xoseperez/espurna/issues/1621#issuecomment-471177403 -> works fine

mcspr, 2019-10-07:
Thanks for the report! I do see the issue here:
https://github.com/xoseperez/espurna/blob/bd385f9fbc71bff007f8b9fe5825777b3bf03dce/code/espurna/light.ino#L939

When step is negative and brightness already approaches 0, lightBrightness function receives negative value. Due to signed / unsigned conversion, resulting brightness is converted from -1 to some value near 4294967295 and in the end into the 100% that you are seeing
I can't see any issues with the encoder handling, but will re-check in the morning and commit the fix.

fluffymadness, 2019-10-08:
thanks for the quick reply. Another thing I've noticed, is it maybe possible to expose color temperature via mqtt ? The pregenerated homeassistant template from espurna only gives me brightness control.

```
light:
  - name: ESPURNA_83D22DF2
    state_topic: 
    command_topic: 
    payload_on: 1
    payload_off: 0
    availability_topic: 
    payload_available: 
    payload_not_available: 
    brightness_state_topic: 
    brightness_command_topic: 
```

mcspr, 2019-10-09:
See if #1938 fixes this. I have a simple encoder setup to mimic xiaomi lamp and it seems to work in both button modes.

For color_temp topic to appear you need to enable both `useWhite` and `useCCT` in the settings

PR #1938 (2019-10-09): Lights: fix broken unsigned arithmetic + refactoring
- replace unsigned return values with the signed ones to avoid undefined behaviour when mixing signed and unsigned in one expression or accidentally casting negative values to unsigned (like lightBrightness(...), lightChannel(id, ...), causing value to become 100% when the intent was to set it to 0%)
- update encoder configure method to only configure encoder objects once, on boot. should be cleaned-up anyway for runtime pin config, either requiring reboot re-using existing objects when nothing had changed.
fixes #1934

some related changes
 [... cut]
closed 2019-10-09 by mcspr