xoseperez/espurna #1935: Where to find info for "KMC 4 Outlet" device support?

ulygit, 2019-10-08:
> Other supported boards (beta): KMC 4 Outlet...

The ESPurna README includes this reference, but I can't find any information about the status of support for the KMC 4-outlet socket. Is this supported in any way?

mcspr, 2019-10-16:
https://github.com/xoseperez/espurna/pull/847 merged this by accident (?)
perhaps @gn0st1c was in the process of adding it and remembers something about it

gn0st1c, 2019-10-27:
i just noticed this mention.
back then (may 2018), i tracked the pins but never flashed the device so i didn't want to merge something i did not test.

here are the pin defs if you'd like to try;
```
// -----------------------------------------------------------------------------
// KMC 4 Outlet /w Energy Monitoring
// https://www.amazon.com/gp/product/B0786B1LGM/
// -----------------------------------------------------------------------------

#elif defined(KMC4)

	// Info
	#define MANUFACTURER		"KMC"
	#define DEVICE				"KMC4"

	// Buttons
	#define BUTTON1_PIN			16
	#define BUTTON1_MODE		BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

	// Relays
	#define RELAY1_PIN			15
	#define RELAY1_TYPE			RELAY_TYPE_NORMAL
	#define RELAY2_PIN			13
	#define RELAY2_TYPE			RELAY_TYPE_NORMAL
	#define RELAY3_PIN			14
	#define RELAY3_TYPE			RELAY_TYPE_NORMAL

	// HLW8012
	#ifndef HLW8012_SUPPORT
	#define HLW8012_SUPPORT		1
	#endif

	#define HLW8012_SEL_PIN		12
	#define HLW8012_CF1_PIN		5
	#define HLW8012_CF_PIN		4

```

closed 2020-01-02 by stale[bot]