xoseperez/espurna #1941: Mi Desklamp Mired Scaling incorrect

fluffymadness, 2019-10-10:
I notices something when playing around with homeassistant and the fluxer component. The xiaomi desklamp has a  color range from 2700k to 6500k
When using Fluxer to adjust color temperature to 2700k it sends 370mired to the lamp. However the lamp on the led is still coldish white.
To have correct 2700k warm light I have to move the mired slider all to the right on espurnas webinterface to 500mireds which however equals 2000k.
Are there some settings where the mired range of the light could be adjusted?

mcspr, 2019-10-11:
Those are the default values:
https://github.com/xoseperez/espurna/search?q=LIGHT_COLDWHITE_MIRED&unscoped_q=LIGHT_COLDWHITE_MIRED
https://github.com/xoseperez/espurna/blob/0a11a74f94653cab6f350e27b9c2929847d28985/code/html/index.html#L1977
https://github.com/xoseperez/espurna/blob/1949bc8e3bb237c62e2f939753242a33277f0586/code/espurna/config/general.h#L1213-L1229

Do you mean that Desklamp defaults should have those set to 2700 and 6500 instead of 2000 and 6553?

fluffymadness, 2019-10-11:
Yes. I think so. According to the product specs https://www.xiaomitoday.com/xiaomi-mijia-mjtd01yl-led-desk-lamp-review/
the lamp has a color range from 2700k-6500k  => 370mired to 153mired

 <input type="range" min="153" max="500" class="slider pure-u-lg-1-4" id="mireds"> 
 		                          =6500k     =2000k
 <input type="range" min="153" max="370" class="slider pure-u-lg-1-4" id="mireds"> 
                                          =6500k     =2700k

Right now when I want to have 2700k color temperature with the lamp I have to move the slider to 500mired in the webinterface
which would equal 2000k color temperature in reality.

So the problem is when the color temperature is set externally (e.g 370 mireds come in from homeassistant flux or circadian addon)
the lamp does not set 2700k.
It feels more like ~3500k-4000k when looking at the color temperature of the light.
So the mired scale doesn't translate to the color temperature of the lamp atm.

mcspr, 2019-10-12:
Ok, so the configuration needs to change.
- mired <-> kelvin defines should be interlinked (probably via mired being the main one and having `KELVIN_COLDWHITE = (10 * ((1000000 / MIREDS_COLDWHITE) / 10))`
- webui needs a small adjustment to receive coldwhite and warmwhite limits to modify min=... and max=... attrs on the input=range
- and, finally, Desklamp should have two new mired defines setting the range to 153...370

Runtime setting could be useful too, `lightMiredsCold`, `lightMiredsWarm`?

fluffymadness, 2019-10-12:
Sounds great.
For the runtime settings i think there could be a use case for led strip controllers that have dual white channels like the h801 where its basically the users choice which color temperature the strips have that are connected. e.g one 3000k for the warmwhite and a 5000k for the cold white channel.

PR #1945 (2019-10-16): Lights: configure mired warm / cold at runtime
- move mired cold / warm limits from constexprs into local vars
- add runtime settings `lightColdMired`, `lightWarmMired` , update webui input
- move kelvin calculation to the configure method
- update Xiaomi Desklamp limits per https://github.com/xoseperez/espurna/issues/1941#issuecomment-541207093

fix #1941
closed 2019-10-16 by mcspr