xoseperez/espurna #1958: Sonoff dual r2 + DALLAS in GPIO9 Failed getting lock

reydecopas, 2019-10-23:
Hi,
I am trying to add a dallas sensor to a sonoff dual r2, after compiling with -DDALLAS_PIN=9 -DDALLAS_SUPPORT=1 -DSENSOR_SUPPORT=1
and modifying hardware.h to 
//#define BUTTON2_PIN         9       // Button 1 on header                
#define BUTTON2_PIN         GPIO_NONE       // Button 1 on header

I am getting in the console:
[294200] [SENSOR] Initializing Dallas @ GPIO9
[294201] [GPIO] Failed getting lock for GPIO9
[294201] [SENSOR]  -> ERROR 7

What am I doing wrong?

mcspr, 2019-10-24:
Firmware tracks Dallas GPIO usage via some helper methods here:
https://github.com/xoseperez/espurna/blob/e17ee6168e0879878d88c04197ae40025e0c0f89/code/espurna/gpio.ino#L17-L28
But it does not even get to the locking part, because GPIO number is less than 12:
https://github.com/xoseperez/espurna/blob/e17ee6168e0879878d88c04197ae40025e0c0f89/code/espurna/gpio.ino#L12

Adjusting 12 to 9 will fix the specific issue, because 8285 does not have GPIO9 connected to the SPI

reydecopas, 2019-10-24:
perfect!!! thank you!! it works!!!

PR #1964 (2019-11-06): gpio: detect esp8285 chip to allow the use of GPIO9 and GPIO10
fix #1958
 
- Read efuse registers to detect ESP8285. I can't find any documentation on that, esptool.py is my only source. (part of it is used for MAC, other bytes are not)
- Redo GPIO locking via std::bitset. Slightly nicer interface than basic shifts and easier to manage for this kind of thing specifically.
closed 2019-11-06 by mcspr