xoseperez/espurna #1959: Loaded wrong file to blitzwolf shp6

NickDomoticz, 2019-10-23:


NickDomoticz, 2019-10-23:
Hi Guys, i flashed a blitzwolf shp6 with OTA tuya, with a Generic 8ch device using the tuya-convert tutorial, but now i wanted to flash it with the correct firmware but i got the following error:There was an error trying to upload the new image, please try again (ERROR 8). i found out that i flashed a 4mb file onto a 1mb flash, but i want to know how to get it to the correct firmware (1mb). is it possible OTA? thanks in advance. sorry i'm a bit new to this

mcspr, 2019-10-24:
Per https://gitter.im/tinkerman-cat/espurna?at=5db0d8a9e1c5e91508c238a2

If user uploads 4MB .bin to 1MB / 2MB board, it will hardlock the device on that specific firmware without any way of recovery.
tuya-convert *does* warn about this: https://github.com/ct-Open-Source/tuya-convert/blob/d3aa848fc228863370a4038e9942ecaa3c5c4d6d/start_flash.sh#L119
> echo "BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!"

- We can add a specific method to do the same check as Updater does and in case of mismatch, overwrite the configuration byte with the correct value. Same rules apply though? As is, method should copy the whole sector into some temporary storage (OTA free space / RAM), apply changes, erase the sector and write the data back.
- tuya-convert could be a bit more diligent and check this byte for us and compare / adjust to the same size as the real size.
- Wiki could use an additional note which .bin files to use

---
Error is defined as:
[`#define UPDATE_ERROR_FLASH_CONFIG       (8)`](https://github.com/esp8266/Arduino/blob/1e17dddd895883806c9bfd3dd7b7042aa813d94f/cores/esp8266/Updater.h#L17)
And happens right after updater starts up:
https://github.com/esp8266/Arduino/blob/1e17dddd895883806c9bfd3dd7b7042aa813d94f/cores/esp8266/Updater.cpp#L99-L102
The required condition is:
[`getFlashChipRealSize() >= getFlashChipSize()`](https://github.com/esp8266/Arduino/blob/1e17dddd895883806c9bfd3dd7b7042aa813d94f/cores/esp8266/Esp.cpp#L443)
Where ChipRealSize is determined directly from the flash chip via spi_flash_get_id() call and the ChipSize is taken from the 3rd byte of the flash:
https://github.com/esp8266/Arduino/blob/1e17dddd895883806c9bfd3dd7b7042aa813d94f/cores/esp8266/Esp.cpp#L299

In the case above, generic-ch8 has `0x40` for 4MB and blitzwolf-bwshpx has `0x20` for 1MB
For comparison, allterco-shelly1.bin has `0x30` for 2MB

mcspr, 2019-11-26:
Closing via https://github.com/ct-Open-Source/tuya-convert/pull/408
Intermediate fw v1.5 will modify flash mode so that any .bin will be able to OTA later, regardless of flash size set when building it

closed 2019-11-26 by mcspr