xoseperez/espurna #1963: I don't recognize hw - sonoff rf bridge 433 - hardware hack

lpt2007, 2019-10-26:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

I can't find my hardware on this site:
[Hardware Itead Sonoff RF Bridge Direct Hack](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack)

Because of that I don't know how to hack this device.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

[image]
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

ColinShorts, 2019-10-26:
I suspect it's a very early model, seems to be shown elsewhere: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433

I'm not sure about whether there are any differences between this and the older of the two pictured ones on the wiki, at least in terms of the pins that you would attach resistors to. Assuming you use the recommended values, it should be relatively safe to try.

lpt2007, 2019-10-27:
Hi, thanks for answer.

I already have tasmota firmware on my device. Can I flash from tasmota? Or I must do hardware hack fisrt?

ColinShorts, 2019-10-27:
I'm not 100% certain about moving from tasmota to espurna via the ota method, but to be safe it might be better to erase the flash before flashing espurna.

You could do the hack before or after the flash, assuming that you do one after the other - it's not like you'll build it back up between times, right?

wildwiz, 2019-10-30:
Hi, I am the author of the hack.
Your board should be an R1V1.0
Follow the instructions for R1V2.0 and you will be fine.
You will need to cut the track one on the components sides from R13 toward EFM8BB1),
but ou don't need to cut tracks on the back of the PCB because on this PCB the USB is clearly not connected to the pins we use.

lpt2007, 2019-11-12:
Sorry for delay, here are the pictures:
[image]
[image]

closed 2020-01-19 by stale[bot]