xoseperez/espurna #1967: PLUGINs: Help wanted | Arduino IDE | includes files

lblabr, 2019-10-31:
tools used:
Arduino IDE, version 1.8.9
espurna: dev 

i like arduino ide and the espurna framework on my esp8266's. sometime i need a special solution, logik, support for sensors and actors.

in the past i tried the plugin concept, with more or less success....
i started new and try to implement an plugin for my mcp4725 DA output converter.

I basis i used the [3rd party plugin](https://github.com/xoseperez/espurna/wiki/3rd-Party-Plugins)

arduino ide compiles all stuff in ./src so i placed the 3rd party files in that special subdirectrory.
in the plugin  i would like to use the custom.h but that seems a little bit diffucult, with also including i get a lot of "multiple definition" errors. so i copied the defintions of prototypes.h in the plugin.h files. 

is there a better and comfortable way to use the original include files ? just include all.h ?
with the procedure above i have one issue left.

WARNUNG: Unberechtigter Ordner .circleci in der Bibliothek 'Homie'
sketch\src\mcp4725b\mcp4725b.cpp.o: In function `extraSetup()':
sketch\src\mcp4725b\../../config/custom.h:67: multiple definition of `extraSetup()'
sketch\espurna.ino.cpp.o:sketch\config/custom.h:67: first defined here

collect2.exe: error: ld returned 1 exit status

relevant files:
./config/custom.h
./src/mcp4725/mcp4725.h
./src/mcp4725/mcp4725.cpp (includes #include "../../config/custom.h")

custom.h

//* Include flags include the plugins code in the image
//* The plugin can be enabled/diabled in run time (API + TERMINAL commands)
	#define INCLUDE_MCP4725B		1

//* USE_EXTRA is espurna 3rd party code integration hook
//* flag is used by espurna.ino to call extraSetup() function
//#define USE_EXTRA            INCLUDE_MCP4725A || INCLUDE_MCP4725B
	#define USE_EXTRA            	INCLUDE_MCP4725B

//* Plugin integation point
	#if USE_EXTRA

    //* Declare espurna calling function
		void plugin_mcp4725b_Setup();

		void extraSetup() {
        //* extraSetup is called by espurna.ino (depended on USE_EXTRA flag)
        //* This is a single entry point to the plugin code
        //* Call the plugin setup function
//			#if INCLUDE_MCP4725B
//				plugin_mcp4725b_Setup();
//			#endif
		}
	#endif

if i define extraSetup i get error above..... if i comment extraSetup the plugin works (by calling plugin_mcp4725b_Setup();) in espurna.ino

enclosed the repository for plugin testing ...
[plugin repo](https://github.com/lblabr/plugin/tree/dev/code/espurna/src/mcp4725b)

mcspr, 2019-11-03:
The error means that you have `extraSetup() { .... body .... }` in both espurna object file (from all of the .ino files concated together) and mcp4725b.o, thus linker can't decide which one to pick. There needs to be only one, since both files end up in a single binary.

Do you really need a separate .cpp file?
If you create mcp4725b.ino in the root directory, then it will also be included into a big .cpp file with all of the espurna code together. Then there is no problem referencing any of the functions like getSetting, setSetting etc. without depending on function declarations
There is a pending issue to convert all of the code into a separate files https://github.com/xoseperez/espurna/issues/1306, then we could include functionality like `#include <espurna/settings.h>`, `#include <espruna/custom.h>` etc.

lblabr, 2019-11-03:
so far as i understood (I'm not an expert ...)

> The error means that you have extraSetup() { .... body .... } in both espurna object file (from all of the .ino files concated together) and mcp4725b.o, thus linker can't decide which one to pick. There needs to be only one, since both files end up in a single binary.

could that be solved by (if already included ?? ifdef  _CUSTOM_H_ or something like this ? )

> Do you really need a separate .cpp file?

i would like to separete my module from the core (espurna), most likely in a nice directory structure...
if you open the ardunio ide, all ino's are opened and gets a little bit confusing. most of the files are more or less libraries for me and i won't touch....

> There is a pending issue to convert all of the code into a separate files #1306, then we could include functionality like #include <espurna/settings.h>, #include <espruna/custom.h> etc.

could you strictly separate definitions and declarations (hope the wording is correct), prototypes.h is little it confusing... and from my point of view not user friendly,  would like to include all definitions in my plugins ;)

lblabr, 2019-11-03:
i googeld around to the function of the preprocessor, so everytime i have a compile unit (cpp-file ??) and i include custom.h with the declaration of extrasetup i will have another version of extrasetup....

but if i would have an plugins.ino in the root folder of espurna (where the others ino's) are and place the extrasetup there and remove ist from the custom.h all would be fine ? in the plugins.ino i just call my setups from my pluigns, as i would do from espurna.ino ....

for update reasons i would like to leave espurna files itself untouched ... 

i wil try that way

any hints are welcome

mcspr, 2019-11-06:
Right. The gist of it is to use .ino for espurna integration code, like extraSetup(), and .cpp for generic code. Think of it as a separate library.

In theory, Arduino IDE and the likes of it add function declarations aka prototypes when converting .ino files into a single .cpp. Sometimes it needs manual intervention though, like the extern "C" bits (this may have been fixed however, I have not tried recent arduino-cli changes from the 1.8.10) and to help later migration to a separate headers.

lblabr, 2019-11-10:
i'm working with arduino ide 1.8.9, are there things to know if i move to 1.8.10 ?

lblabr, 2019-11-10:
at least i found my way for plugins with espurna and ardunio ide, it seems to work ...

in short:
- plugins.ino (calling extraSetup)
- my extensions in ./src 
- functions/class definitions in an extra prototypes_definitions.h

closed 2020-01-16 by stale[bot]