xoseperez/espurna #1968: Sonoff POW REQUEST : if vaule W is over "X" or Range vaule  SONOFF POW send automatic HTTP API instruction.

cent0, 2019-11-01:
@mcspr hello i need adjust Pow interface and add this string code. if have overcurrent need switch off other interface.
please help for searc point for insertion

int the future for develop gui interface

This is code:
_____________
 http.begin("http://192.168.123.101/api/relay/0?apikey=1234567890123456&value=0"); //HTTP
                                                                                                              // start connection and send HTTP header
                                                                                                              int httpCode = http.GET();

                                                                                                               http.end(); 
_______________

best regards

mcspr, 2019-11-23:
See https://github.com/xoseperez/espurna/issues/1988#issuecomment-552336479
And also see this example from influxdb module:
https://github.com/xoseperez/espurna/blob/9a5698b342d6357bdd42267eb30932df9ce30260/code/espurna/influxdb.ino#L134
https://github.com/xoseperez/espurna/blob/9a5698b342d6357bdd42267eb30932df9ce30260/code/espurna/influxdb.ino#L48-L50

Depending on the `BROKER_REAL_TIME` define. When set to `0` (default), it will either report each N readings (`snsReport` / `SENSOR_REPORT_EVERY`). When set to `1`, it will publish sensor data each reading (`snsRead` / `SENSOR_READ_INTERVAL`).

I would also suggest to use `schedule_function([bind,variables,here]() { ... your http code here ... });` in the broker callback.

closed 2019-11-23 by mcspr

MihataBG, 2019-12-06:
In master branch 1.14.0 BROKER_REAL_TIME is set to 1, is it correct and are you changed default behavior?

mcspr, 2019-12-06:
@MihataBG Can you explain the problem with it?
Looks unintentional, I thought that was 0 at the time of writing this (...since I switched off RULES_SUPPORT default in https://github.com/xoseperez/espurna/pull/1984)

MihataBG, 2019-12-06:
The problem is with POW R2, it crashes when BROKER_REAL_TIME is set to 1.

cent0, 2020-05-30:
> See [#1988 (comment)](https://github.com/xoseperez/espurna/issues/1988#issuecomment-552336479)
> And also see this example from influxdb module:
> https://github.com/xoseperez/espurna/blob/9a5698b342d6357bdd42267eb30932df9ce30260/code/espurna/influxdb.ino#L134
> 
> 
> https://github.com/xoseperez/espurna/blob/9a5698b342d6357bdd42267eb30932df9ce30260/code/espurna/influxdb.ino#L48-L50
> 
> How often it is called depends on the `BROKER_REAL_TIME` define. When set to `0` (default), it will report each N readings (`snsReport` / `SENSOR_REPORT_EVERY`. MQTT works like this). When set to `1`, it will publish sensor data each reading (`snsRead` / `SENSOR_READ_INTERVAL`. WebUI is updated each reading).
> 
> I would also suggest to use `schedule_function([bind,variables,here]() { ... your http code here ... });` in the broker callback.

@mcspr  @lblabr 
For solve this is possible add this line in main.cpp?
 #include newfunction.h

And new file in espurna folder.
In this file newfunction.h  

#include esp8266HTTPClient
#include sensor.h
If (MAGNITUDE_POWER_ACTIVE >100); //working this call?
{

_
//http.begin("http://192.168.123.101/api/relay/0?apikey=1234567890123456&value=0"); //HTTP
// start connection and send //HTTP header
// int httpCode = http.GET();

// http.end(); "

}

mcspr, 2020-05-31:
@cent0 see https://github.com/xoseperez/espurna/wiki/3rd-Party-Plugins