xoseperez/espurna #1981: HowTo: I2C in own module / Implementing own sensor/actor

lblabr, 2019-11-08:
i would like to implement some sonsors/actors, and i would like to follow your suggest way. could somebody describe what is the best workflow ?

i read in the existing sensors...

the flow is:
1) _begin_i2c();  i just have one address, whats the idea behind register the address ?
2) i2c_write_uint16(_address, 64, _output << 4);    write/read to/from address ? 
3)  do i need free/end something ?

thanks in advance

Lars

mcspr, 2019-11-10:
> whats the idea behind register the address ?

Do you mean locking? The goal is to notify through the debug log about address collisions, if / when they happen with a single i2c sensor or multiple

> do i need free/end something ?

?
If you mean "do something with Wire or I2CSensor object", then there is no need. i2c_read and i2c_write are just a wrappers for Wire, they already handle everything

lblabr, 2019-11-10:
okay, great

closed 2019-11-10 by lblabr