xoseperez/espurna #1983: POW r2 does not report Power parameters

jjviton, 2019-11-09:
Hi,

I´m facing a similar issue, I´m not able to read the Power paramater of my flashed POW.

Enviroment: SONOFF POW R2 // Espurna version 1,13 // sensor chip CSE7759

I´ve read in other threads about this issue, but I don´t find a clear answer.

Could you please advise me how to fix this issue?

Thanks in advance.

mcspr, 2019-11-11:
Could be related to #1823? Also see https://github.com/xoseperez/espurna/issues/696 in regard to the CSE7759. I do not have of any POW devices with CSE7759, so no idea here. I'd ask in that issue instead.

Also note that you are using the R1 POW, not the R2 with CSE7766. Some sources refer to it as POW V2, since the old HLW8012 was replaced with the CSE7759

closed 2019-11-11 by mcspr

jjviton, 2019-11-11:
Thanks for your inputs @mcspr 
I´ll flash the r1 verison of espurna in order to check if it works.
I keep you informed

jjviton, 2019-11-13:
Hi,
Issue solved.

Important to notice: even you see un the PCB "POV rev 2", it does not mean you have version 2.

This was my confusion, uploading bin POV verison 1 works!!!

thanks @mcspr

mcspr, 2019-11-13:
Do you reckon this is the same issue as https://github.com/xoseperez/espurna/issues/1038?
Maybe we could add a photo of both sides of this version to the wiki?

This knowledge somehow got lost in the issue tracker :/ https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-POW-R2 lists [CSE7759**B**](http://www.chipsea.com/en/afe/CSE7759B.html) with a serial interface, while [CSE7759 without the suffix](https://lcsc.com/product-detail/PMIC-Current-Regulation_CSE7759_C126724.html) seems to only have CF,CF1,SEL interface, same as HLW chip of the previous model. Searching through here and Tasmota archives actually pinpointed the problem with distinguishing the chip version