xoseperez/espurna #1988: PLUGIN: http get to extern uri

lblabr, 2019-11-10:
i would like to implement an url request inside a plugin. how this could be achived ?
just a hint would be great

- is there already an object wich can be used, if not wich is preferred ?
- any style guide ?

thanks in advance 

Lars

mcspr, 2019-11-11:
See https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient/examples ?
Note that default timeouts for request failure are pretty generous, I'd look into reducing them for any real use.

I had done another weird client in the #1909 using the ESPAsyncTCP, but that is WIP and I need to get back to that with proper header handling and easier API

mcspr, 2019-11-23:
Btw, also see my comment in the https://github.com/xoseperez/espurna/issues/1968
Plugin code still uses brokerSubscribe and generic callback, those were replaced with SensorBroker (callback-per-event, see broker.h).

closed 2019-11-23 by mcspr