xoseperez/espurna #1993: Switching from fauxmoESP for Espalexa, for better Alexa lightbulbs support (brightness, colour, temperature)?

grisevg, 2019-11-15:
Hey folks,

I'm considering swapping  fauxmoESP for Espalexa in my local project, to make Alexa understand that my lightbulb is indeed a lightbulb and not 5 different abstract percentage devices.

Can't find any mentions if anyone has tried to put it into espurna already. Would you folks be interested in a pull-request?

mcspr, 2019-11-17:
ATM there is an outstanding issue with fauxmoESP not being discover-able at all - https://github.com/xoseperez/espurna/issues/1904 
ref https://github.com/xoseperez/espurna/issues/1904#issuecomment-529957777, something with generated ids.

I am not using Alexa, so either fixing fauxmo discovery or making a switch to espalexa (which is known to work) would be greatly appreciated!

grisevg, 2019-11-18:
kewl beans. The generated ids kinda do work, but they're useless beyond turning it off and on (brightness is not exposed at all and who can say raw, scaled hue and saturation values of top of their heads).

I got Espalexa working and it works pretty great, it's now properly exposed as a lightbulb. 
Tho because I have a RGBW lightbulb, I exposed it as two separate deviced, one RGB and one W and I do find it more convenient than having the lightbulb switch between "Color" and "White" modes (you have to tell alexa to change the color to white or to a specific color and that switches the modes).

I guess I can expose both ways through a define flag.
I'll start cleaning up my code and preparing the pull request. I'm also considering to keep fauxmo for non-light devices.

mcspr, 2019-12-09:
*ping*

I wonder if this would work? Looking at emulated_hue component in HASS, it skips some of the properties based on device type:
https://github.com/Aircoookie/Espalexa/blob/fa40eccc91df14706ca4bf202ffcd419341b9367/src/Espalexa.h#L146-L158
https://github.com/home-assistant/home-assistant/blob/8dea7f0f98df8c93d4a628d4cba470a64b263877/homeassistant/components/emulated_hue/hue_api.py#L615-L711
(not sure if this is relevant, but protocol stuff looks similar enough. it could be a different version though)

mcspr, 2020-03-30:
Quoting https://twitter.com/xoseperez/status/1241293353729380354
> I'm stopping maintaining the fauxmoESP library (https://bitbucket.org/xoseperez/fauxmoesp). I have no time to do it and I'm no longer willing to provide Alexa compatibility options. Anyone interested in maintaining it, please contact me.

grisevg, 2020-03-31:
sorry i've been really swamped with everything going on... I'll do it, it's working well, I'll prepare my code to be shared, sorry for delays everyone :((

mcspr, 2020-03-31:
Thank you!
Glad to hear it is already working.

BTW does the existing fauxmo integration work at all? I was under the assumption that it broke completely for some specific hw version (at least via the fauxmo issue tracker)

*This was not meant as a nudge :) Just a note if someone finds this in google / search, I do appreciate that you take the time to do this.*

buserror, 2020-04-08:
Alexa integration never worked for me. I installed countless version of the firmware on my switch (for MQTT mostly) but the built-in alexa support hasn't worked, ever. Sonoff touch, S20s mostly

jsponz, 2020-04-26:
@grisevg Thanks for taking the lead. When do we expect to have a new version?