xoseperez/espurna #1994: Compile issue with pwm.c for specific RFLink Wi-Fi Gateway config

kevinkahn, 2019-11-17:
I can build all the standard configurations with no problem using the platformio core etc. on a RPi.  However, another developer has created a configuration for the robotdyn mega plus esp board to use with RFLink (https://gitlab.com/rrenato/rflink-wi-fi-gateway-board/tree/master).  I want to build this against the current development version because using the last release version of espurna I get some webui issues that I have seen discussed and probably solved since that release.  Unfortunately, when I try to build this version following his instructions except using the current development version of espurna, I reliably get a compile time error related to pwm.c as follows:
*******************************************************************************
espurna\pwm.c:115:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'pwm_intr_handler'
 pwm_intr_handler(void)
 ^
espurna\pwm.c:161:1: error: conflicting types for 'pwm_init'
 pwm_init(uint32_t period, uint32_t *duty, uint32_t pwm_channel_num,
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:34:6: note: previous declaration of 'pwm_init' was here
 void pwm_init(uint32 period, uint32 *duty,uint32 pwm_channel_num,uint32 (*pin_info_list)[3]);
      ^
In file included from espurna\pwm.c:49:0:
espurna\pwm.c: In function 'pwm_init':
espurna\pwm.c:196:33: error: 'pwm_intr_handler' undeclared (first use in this function)
  ETS_FRC_TIMER1_NMI_INTR_ATTACH(pwm_intr_handler);
                                 ^
C:\Users\Kevin\.platformio\packages\sdk-esp8266\include/ets_sys.h:56:16: note: in definition of macro 'ETS_FRC_TIMER1_NMI_INTR_ATTACH'
  NmiTimSetFunc(func)
                ^
espurna\pwm.c:196:33: note: each undeclared identifier is reported only once for each function it appears in
  ETS_FRC_TIMER1_NMI_INTR_ATTACH(pwm_intr_handler);
                                 ^
C:\Users\Kevin\.platformio\packages\sdk-esp8266\include/ets_sys.h:56:16: note: in definition of macro 'ETS_FRC_TIMER1_NMI_INTR_ATTACH'
  NmiTimSetFunc(func)
                ^
espurna\pwm.c: At top level:
espurna\pwm.c:407:1: error: conflicting types for 'pwm_set_duty'
 pwm_set_duty(uint32_t duty, uint8_t channel)
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:37:6: note: previous declaration of 'pwm_set_duty' was here
 void pwm_set_duty(uint32 duty, uint8 channel);
      ^
espurna\pwm.c:419:1: error: conflicting types for 'pwm_get_duty'
 pwm_get_duty(uint8_t channel)
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:38:8: note: previous declaration of 'pwm_get_duty' was here
 uint32 pwm_get_duty(uint8 channel);
        ^
espurna\pwm.c:427:1: error: conflicting types for 'pwm_set_period'
 pwm_set_period(uint32_t period)
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:39:6: note: previous declaration of 'pwm_set_period' was here
 void pwm_set_period(uint32 period);
      ^
espurna\pwm.c:438:1: error: conflicting types for 'pwm_get_period'
 pwm_get_period(void)
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:40:8: note: previous declaration of 'pwm_get_period' was here
 uint32 pwm_get_period(void);
        ^
espurna\pwm.c:444:1: error: conflicting types for 'get_pwm_version'
 get_pwm_version(void)
 ^
In file included from espurna\pwm.c:50:0:
espurna\libs/pwm.h:42:8: note: previous declaration of 'get_pwm_version' was here
 uint32 get_pwm_version(void);
        ^
*** [.pio\build\generic-robotdyn-atmega2560-esp8266\src\pwm.o] Error 1
========================= [FAILED] Took 9.68 seconds =========================

I have interacted with the configuration author (see https://gitlab.com/rrenato/rflink-wi-fi-gateway-board/issues/1) on the error and he can't figure it out and apparently doesn't see it when he tries to build.  I see it on the Pi environment and also in a Win10 build environment so it would appear not to be something about my platformio installation.  Have you seen anything like this and if so can you at least point me in a direction to look?

mcspr, 2019-11-17:
> C:\Users\Kevin.platformio\packages\sdk-esp8266\include/ets_sys.h:56:16: note: in definition of macro 'ETS_FRC_TIMER1_NMI_INTR_ATTACH'
NmiTimSetFunc(func)
^

You need to fix the [env:...] configuration. As you can see, other board definitions are reduced to only have `board = ...` and `build_flags =`. Please try removing every other line but those two.

When using the example from the https://gitlab.com/rrenato/rflink-wi-fi-gateway-board/tree/master#instructionsthe-following-instruction-explain-how-build-your-personal-rflink-wi-fi-gateway, `framework = ${common.framework}` effectively makes your env build with some old espressif sdk instead of the Arduino.

kevinkahn, 2019-11-18:
Thanks for such a quick reply.  I cleaned up the environment from the other project per your suggestion and got a bin generated.  On to see if it works.  Thanks again.
Kevin

closed 2020-01-24 by stale[bot]