xoseperez/espurna #1998: Support for new Gosund SP111 and SP112

knopserl, 2019-11-19:
Has anybody sucess and a tested configuration for the new Gosund SP111 and or SP112?
Therey are very attractive and already working with Tasmota but I rather wanzted to have all my plugs working with Espurna.
Blitzwolf BW SHP2 does not work but SP112 is equal to a Blitzwolf BW SHP5 and Gosund SP111???

mcspr, 2019-11-23:
I guess the existing `BLITZWOLF_BWSHPX` / `blitzwolf-shpx` profile should work?

Googling for any info on these, I found tasmota device listings with both devices present:
https://blakadder.github.io/templates/gosund_SP111.html
https://blakadder.github.io/templates/gosund_SP112.html

GPIO pins are decoded through reading this enum (starting from 0 aka GPIO_NONE / unused pin):
https://github.com/arendst/Tasmota/blob/63b001021c5166b63078f782f177b6412dc7718f/tasmota/tasmota_template.h#L26

knopserl, 2019-11-23:
Thanks I found at least a solution for the SP111. It can use BLITZWOLF_BWSHPX but the pin-assignment for CF1 14 does not work, it is 4.
The SP112 does not work, as it has a new power monitor chip with serial communication. Tasmota has a driver for it. So it would require some extention (using code from tasmota?)

mcspr, 2019-11-26:
Sorry, I missed the last comment.
Could you please post the required defines?

Tasmota config for SP112 uses CSE7766 connected as receive-only which we already support, Sonoff POWr2 was the first to use it

JoanMCD, 2019-11-28:
I'm also working at this device - as additional info I found: 
How to program
https://www.malachisoord.com/2019/11/24/flashing-custom-firmware-on-a-gosund-sp111/

knopserl, 2019-11-30:
it would be very helpful and extend the flexibiliy of espurna, if we could change pins after flashing such as tasmota is doing with the profiles.
So the SP111/Blitzwolf BW-SHP5 vendor exchanged recently the CF1 pin from 14 to 4 and the power monitor did not work. If that could be "soft-changed" I could have simple changed that after flashing. Maybe that is worth to make a seperate/new issue? What do you think?

mcspr, 2019-12-01:
I agree, it would. ATM the planned solution is to add new cfg keys `hlw8012CF1`, `hlw8012CF` and `hlw8012SEL` with pin numbers as values, trying reading those first and then falling back to the `HLW8012_..._PIN` values. There are a couple “global” issues regarding soft configuration (see the header and v2 branch), but we could just cover one problem at a time.

knopserl, 2019-12-03:
@mcspr That is already a very good start if the 3 power monitor pins could be reassigned afterwards (if the chip is the same). The SP112 (BW-SHP6 has even changed the chip, which has serial line interface (2 pins). It might be also good if the LED and relay pins could be also soft-configured after flashing.

NickDomoticz, 2020-02-08:
What is the current status of implementing this? I have the new SP111 but can't get any power readings.

mcspr, 2020-02-09:
Would you be able to check today's release from here?
https://github.com/mcspr/espurna-nightly-builder/releases

As noted in my comment, SHPX is the .bin to try.
Only difference is now you can input `set snsHlw8012Cf1GPIO 4` and reboot. CF1 should change to a proper pin.

NickDomoticz, 2020-02-09:
Thank you for your fast reply. I have changed the pin to GPIO 4, but i couldn't get any data. The problem is that i bought the 2NICE SP111 instead of the Gosund SP111, and these plugs don't have the measuring chip inside. Someone mentioned this on another website (too late for me, i bought them 2 months ago). Thanks anyway!

SilentFez, 2020-02-12:
Can confirm that using `set snsHlw8012Cf1GPIO 4` with `espurna-1.14.2-dev.nightly20200209` works as intended. I've got the new BlitzWolf SHP6 15A switches (Gosund SP111 v1.1) and it is now giving me a reading! Thanks a bunch :]

[image]

PoloB12, 2020-06-28:
> Can confirm that using `set snsHlw8012Cf1GPIO 4` with `espurna-1.14.2-dev.nightly20200209` works as intended. I've got the new BlitzWolf SHP6 15A plugs (Gosund SP111 v1.1) and it is now giving me a reading! Thanks a bunch :]
> 
> [image]

Hi, does your MQTT work ? I managed to get the plug working with that specific version and  `set snsHlw8012Cf1GPIO 4` . Power readings seem ok and works fine. I just don't get any MQTT readings transmitted.

SilentFez, 2020-06-28:
@PoloB12 Yep, it is working as it should. Are you using auto discovery or adding it manually?
If you are using auto, make sure to also click on HASS in the switch interface and activate discover.
What MQTT broker are you using? Do you get anything in the log hinting of a connection problem?

PoloB12, 2020-06-28:
I'm not sure what you mean, but I've setup my broker, port, user, password and topic as I do normally but nothing arrives at my broker. My other devices still broadcast normally. Also manual testing from other Pi's for example works fine. I must have missed something somewhere. I'm using these exact devices : 

> 

> I'm also working at this device - as additional info I found:
> How to program
> https://www.malachisoord.com/2019/11/24/flashing-custom-firmware-on-a-gosund-sp111/

SilentFez, 2020-06-28:
Is your broker connected to Home Assistant or some other interface?

PoloB12, 2020-06-28:
I'm not using home assistant. I'm only logging the data of all my sockets to Influxdb via mosquittto to Influx bridge on my VPS. It's a regular procedure for me, also do it with a few Teckin SP23's I've been using with Espurna. I'm not sure what `set snsHlw8012Cf1GPIO 4` did, but maybe something also needs to be changed for mqtt to pick up the data ?

PoloB12, 2020-06-28:
Hmm, I do see mqtt  stuff happening in de debug log :  

`[901089] [MQTT] Sending xxx/yyy/current => 0.000 (PID 8403)`
`[901090] [MQTT] Sending xxx/yyy/voltage => 276 (PID 8404)`
`[901092] [MQTT] Sending xxx/yyy/power => 0 (PID 8405)`
`[901097] [MQTT] Sending xxx/yyy/reactive => 0 (PID 8406)`
`[901104] [MQTT] Sending xxx/yyy/apparent => 0 (PID 8407)`
`[901110] [MQTT] Publish ACK for PID 8403` 
`[901115] [MQTT] Sending xxx/yyy/factor => 0 (PID 8408)`
`[901120] [MQTT] Sending xxx/yyy/energy => 0.046 (PID 8409)`
`[901132] [MQTT] Publish ACK for PID 8404`
`[901154] [MQTT] Publish ACK for PID 8405`
`[901154] [MQTT] Publish ACK for PID 8406`
`[901154] [MQTT] Publish ACK for PID 8407`
`[901279] [MQTT] Publish ACK for PID 8408`
`[901280] [MQTT] Publish ACK for PID 8409`

Need to investigate further whats's going on, but it really seems nothing is actually being sent although the log seems to show activity.

PoloB12, 2020-06-28:
It was something else, nothing related to espurna. Misconfiguration from my side on the vps. Thanks for your reply and checking mqtt !

StefanS52u1, 2020-07-06:
Today i made an OTA flash of the Gosund SP112. Flashing and configurations seems to work quite well. But after i tested the configurations. It seems the the relais is really switching, the main switch has power, but both USB-Ports are out of power. The button on the switch is also not working. 
I ordered this stuff last week from amazon germany and both devices are showing the identical behaviour. With the original software the usb ports had power and either the button works.

-> Previously i had flashed multiple Gosund Sp1 without any issues. 

Has the company relased a new hardware-version of the Gosund SP112?.