xoseperez/espurna #1999: Add 'magichome' (zj-wfmn-c-11) single channel LED dimmer.

davebuk, 2019-11-19:
I have bought a single channel 'magichome' WiFi LED dimmer. I have used the `env:magichome-zj-espm-5ch-b-13` environment as there is no IR connected and changed the number of channels to 1 on GPIO 12.

Can an environment for 'zj-wfmn-c-11' be created?
https://photos.google.com/photo/AF1QipMpfPds44SgSZbk0_YIwZCrZvcudnwmuaCpf9AF
https://photos.google.com/photo/AF1QipO6EVPysXkGkDbBCwPgUcwddPSiQ8WGqt-xfhfj
https://photos.google.com/photo/AF1QipN0lsrMRiGvFZb-PCYEFHBK18yTjtkGxTnopdE5

mcspr, 2019-11-23:
Sure, is this it?
```c
#elif defined(MAGICHOME_ZJ_WFMN_C_11)

    // Info
    #define MANUFACTURER        "MAGICHOME"
    #define DEVICE              "ZJ_WFMN_C_11"
    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // Light
    #define LIGHT_CHANNELS      1
    #define LIGHT_CH1_PIN       12      // COLD WHITE
```

davebuk, 2019-11-23:
Yes, plus `#define LIGHT_CH1_INVERSE   0`.

davebuk, 2019-12-03:
Is there any progress on implementing percentage values for single channel dimmers? Ref #1460

davebuk, 2019-12-03:
@mcspr  I have noticed that when using the above definitions, although I have an ON/OFF switch on the status page and the output is controlled correctly, there is nothing showing under the 'Switches' section of the webUI section.

I have a magic home RGB controller as well. That device **doesn't** have any `BUTTONS` definitions in its environment at all, but does have an entry in the 'Switches' section. I wanted my be able to set the boot status of the my single channel dimmer to ON but don't have that option in the webUI.

mcspr, 2019-12-03:
See `LIGHT_MAX_BRIGHTNESS` and `LIGHT_MAX_VALUE` defines, output will scale based on those. But I think there are still some hardcoded “rogue-value / 255” that may break some interactions. Gamma too, but that does not matter for 1ch device.

Have you checked with the 1.14.0?

davebuk, 2019-12-03:
Just to be clear, can I just as those to my custom.h e.g. `#define  LIGHT_MAX_BRIGHTNESS 100`etc or temporarily modify the general.h file?
I'll try building a 1.14.0 instead of the dev version tomorrow.

mcspr, 2019-12-03:
Either way, there are ifndef things around them so custom.h or PIO build flags should work.

davebuk, 2019-12-04:
Using the definition below in custom.h I still do not see any switches.

```
#elif defined(MAGICHOME_ZJ_WFMN_A_11) //_SINGLE

    // Info
    #define MANUFACTURER        "MAGICHOME"
    #define DEVICE              "ZJ_WFMN_C_11"
    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1

    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define SCHEDULER_SUPPORT       0
    #define SENSOR_SUPPORT          0
    #define THINGSPEAK_SUPPORT      0
    #define LIGHT_MAX_VALUE         100
	#define LIGHT_MAX_BRIGHTNESS 	100
	
	// Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
	
    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // Light
    #define LIGHT_CHANNELS      1
    #define LIGHT_CH1_PIN       12      // WHITE
    #define LIGHT_CH1_INVERSE   0
```

[image]

[image]

mcspr, 2019-12-04:
Yep, that's a WebUI bug because of disabled scheduler:
https://github.com/xoseperez/espurna/blob/978cf2af95a53232ceebd4b0cb3ac1e4758fbd74/code/html/custom.js#L1078-L1085

`data.sch_last` does not exist, there should be a check similar to the following:
```js
if ("sch_last" in data) {
...
}
```

davebuk, 2019-12-04:
I'll re-enable scheduler in my build :-) It was only disabled to reduce file size and keep under 500kB, but as i'm using core 2.5.2 and up in my builds, its not an issue now as they are always bigger and require 2-step OTA.

davebuk, 2019-12-08:
@mcspr Did you want me to raise a PR for this device? It's that an easier way for you guys to track these additions?

closed 2019-12-16 by mcspr

seritos, 2020-01-04:
Hi All,

I also have this Magic Home, and as far as I can see in the board, there is a channel for CW and other for WW. Shouldn't be possible to use both channels and control a White and Warm White led strip? 

Thanx!

davebuk, 2020-01-05:
Possibly. There does seem to be an unused solder pad. You could look to build a custom espurna build and add a second light channel. You just need to find the right GPIO assignment to control it. I don't know the code well enough, but there might be something in espurna that if more than one light channel is specified, it'll presume it's an RGB setup rather than a device with two outputs. I wouldn't know how it would treat one channel as warm white with the other as cold white, without specifying the normally corresponding RGB channels to go with them?

Maybe flash one of the 5 channel pre built .bin files and see which outputs are controlled by which light channel to find the other connected GPIO?

mcspr, 2020-01-05:
Check out comment here:
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/config/general.h#L1149-L1154
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/config/hardware.h#L3466-L3470
Just need to figure out correct GPIO value for 2nd channel

seritos, 2020-01-05:
Thanx for your comments.

I have connected a led strip with two channels, every channel to each pad. 
Looks like the GPIO12 is connected to both channels, as I increase the brightness and both channels increases at the same time. 
I believe that if they would be different channels, one of them would be enabled.

I have tried to follow the electrical path, and both transistors ( for ww and cw)  are connected

I don’t think I have any chance.... Don’t you think?

denes44, 2021-01-26:
> I have tried to follow the electrical path, and both transistors ( for ww and cw) are connected
> 
> I don’t think I have any chance.... Don’t you think?

I bought the same controller, and now trying to figure out the GPIO numbers.
As I can see, you can modify the ZJ-WFMN-C v1.1 board to a CCT board, but you need to move a resistor and that includes soldering. There are four resistors with 512 value, and two with 201 value, and a blank position. You need to move one of the 201 resistors to the blank position. The one which has the same direction as the 512 ones.
So basically the board can do two channels, but it's hardware modified so both channels are controlled by the same pin from the ESP.

PR #2051 (2019-12-16): Device: MAGICHOME_ZJ_WFMN_C_11
fix #1999
@davebuk https://github.com/xoseperez/espurna/blob/15accf8090e4c059dff81061eb7d9c8e1248f628/code/espurna/config/defaults.h#L650-L652
INVERSE can be omitted