xoseperez/espurna #2006: compile errors in platformio + visual studio code

knopserl, 2019-11-23:
I wanted to compile espurna with the latest versions of platformio and visual studio code.
I was able to compile espurna on my previos computer but dodnt get it to run on my new pc running Windows 10 1903 64Bit.
the espurna Wiki hints, installed all required components, platforms, extentiosn, git, gulp etc. all in latest version. I get several warning/errors and do get further. 

I have spent already several evenings and have followed the honts I have found on espurna and searching in the web. I hope somebody experienced can help me further.
There is e.g. a permission error which seem to be rather a flase error as I have all perrmission and have tried other disks directories and the directories were empty etc.
 
Here is the output

> Executing task: C:\Users\internet\.platformio\penv\Scripts\platformio.exe run --environment blitzwolf-bwshpx <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing blitzwolf-bwshpx (platform: espressif8266@1.5.0; framework: arduino; board: esp01_1m)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting espurna.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
LibraryManager: Installing ESPAsyncTCP
git version 2.24.0.windows.2
Cloning into 'd:\mcdev\vscode\espurna\code\.pio\libdeps\blitzwolf-bwshpx\_tmp_installing-bs2a9h5b-package'...
HEAD is now at 55cd520 Add posibility to handle packet buffers directly
[WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'd:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\blitzwolf-bwshpx\\_tmp_installing-bs2a9h5b-package' 
Please manually remove the file `d:\mcdev\vscode\espurna\code\.pio\libdeps\blitzwolf-bwshpx\_tmp_installing-bs2a9h5b-package`
PermissionError: [WinError 5] Zugriff verweigert: 'd:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\blitzwolf-bwshpx\\_tmp_installing-bs2a9h5b-package\\.git\\objects\\pack\\pack-c06bfa45e86338612aa04d98e684bc449d91bf10.idx':
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\internet\.platformio\platforms\espressif8266@1.5.0\builder\main.py", line 375:
    target_elf = env.BuildProgram()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "C:\Users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\internet\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 398:
    os.unlink(fullname)
======================================================================================================================= [FAILED] Took 5.01 seconds =======================================================================================================================

Matthias-GER, 2019-11-30:
Did you find a solution for your problem? I have exactly the same problem after a fresh installation. 
Thanks for any help!

mcspr, 2019-12-01:
Does it work just via CLI, platformio / pio commands, without VSCode running? Like pio call in the log above, `C:\Users\internet.platformio\penv\Scripts\platformio.exe run --environment blitzwolf-bwshpx`, or `pio -e blitzwolf-bwshpx`

My guess that has something to do with either VSCode itself doing something with git or some Antivirus solution locking the temporary files that pio creates. But in any case, I’d start with https://github.com/platformio/platformio-core/issues

Matthias-GER, 2019-12-01:
Hi @mcspr!

Many thanks for your help. As you suggested, I tried now to build directly with pio CLI. The results were the same. But your hint with the virus scanner was perfect! I deactivated the "real time protection" and it worked! Once all the missing libraries are installed, the virus scanner can be activated again and the build works anyway.

Again many thanks for your help!!! 👍

davebuk, 2019-12-08:
I have similar issues when changing between build versions. Disabling anti virus didn't always help. Running `pio lib update` at the command line, I believe gets all the required libraries before trying to build and I then don't have any errors, even with anti virus on.

oscarsan1, 2019-12-15:
Go on here with the Platformio lib_deps instaling error #2062.
Every build run only one library installed, after 21 buils all libraries are installed and build success.
I've tried disable antivirus without success. 
Also pio lib update only update already downloaded library but don't download new libraries.
Every time you change to new env_default all the libraries have to download again for the new envirorment. You can copy all the libraries from old envirorment for example C:\espurna-master 1.14.0\code\.pio\libdeps\itead-sonoff-mini to the new one and avoid to download again all the libraries.

mcspr, 2019-12-15:
Yes, the script will run `pio lib -g install all_of_libdeps`. Note the -g, this will allow us to share libraries between envs.
You can also manually do that by copying any of the successfully installed env's libdeps dir into the `%userprofile%\.plattormio\lib` or maybe `c:\.platfotmio\lib` if PIO is installed there. The resulting dir should contain libs directories like `~\.platformio\lib\ESPAsyncTCP`
Perhaps I need to turn that on by default

I cannot reproduce the issue thought.
Does it matter if VScode is closed at the moment of the lib installation? Do you use SSD or HDD drive? Is there any problem installing libs when the build directory is inside user Documents or any other place under user's profile and not `c:\espurna-master-1.14.0\`?

oscarsan1, 2019-12-15:
You're right, I have space in subprocess_libdeps(), I correct it but the same result. I don't see any of the prints of the C:\espurna-master 1.14.0\code\scripts\pio_pre.py in the terminal. Is this correct? Where are this prints printed?

I copy the libraries to %userprofile%\.plattormio\lib, with this there is no need of download libraries for each env_default. The pio creates the directory C:\espurna-master 1.14.0\code\.pio\libdeps\itead-sonoff-t1-2ch, for example, but no liabraies are put there, only use the shared global ones. No mather of subprocess_libdeps(env.GetProjectOption("lib_deps"), storage) is None or other. 
I delete now all local libraries as I don't need it. I use the shared ones. 
Anyway this don't correct the first build without any library in shared directory.
My antivirus is the windows 10 default antivirus. I try run pio outside VScode inside a Windows PowerShell with the same result, one library installed each build. I have SSD drive. 
I don't kwow how to test your last question, how can I chage this directory.

mcspr, 2019-12-15:
@oscarsan1 The function that you moved depends on env variables `ESPURNA_PIO_SHARED_LIBRARIES` and `TRAVIS` being non-empty. Right now it only activates when building release binaries.

BTW I have tried again with a different Windows env, and I see the issue when Python 3.7 is installed. Everything was ok when I tried earlier with 2.7. Can you try again with the latest 3.8? https://www.python.org/downloads/release/python-380/

Make sure that VSCode isn't running. And after installing 3.8, just check that running `py -3 --version` in the terminal shows the correct version. You would need to remove the [existing virtual environment used by the PIO](https://docs.platformio.org/en/latest/installation.html#virtual-environment) - I think it is at `C:\Users\Osr\.platformio\penv` based on the logs you provided. After removing it, run `py -3 -mvenv penv` and `.\penv\Scripts\pip.exe install platformio` to recreate it based on 3.8

oscarsan1, 2019-12-15:
I've done this but there is no change. Same error. 
But then I execute directly in PowerShell: 
C:\_proyecto\_Espurna\espurna-master 1.14.0\code> platformio run
And download ALL the libraries and build success. This is part of the log:

`Looking for PubSubClient library in registry
Found: https://platformio.org/lib/show/89/PubSubClient
LibraryManager: Installing id=89
Using cache: C:\Users\Osr\.platformio\.cache\60\bcf94560201e022015c1c79750fa0460
PubSubClient @ 2.7 has been successfully installed!
Looking for rc-switch library in registry
Found: https://platformio.org/lib/show/246/rc-switch
LibraryManager: Installing id=246
Using cache: C:\Users\Osr\.platformio\.cache\3c\4963a5e2c9626c15a4b03a63d7cfe43c
rc-switch @ 2.6.2 has been successfully installed!
LibraryManager: Installing RFM69
git version 2.24.0.windows.2
Cloning into 'C:\_proyecto\_Espurna\espurna-master 1.14.0\code\.pio\libdeps\itead-sonoff-mini\_tmp_installing-_i4_4hbk-package'...
Note: switching to 'cb68f71ae81adb5801d197d7835065d7e1f1e125'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

RFM69 @ cb68f71 has been successfully installed!
Installing dependencies
Looking for SPIFlash library in registry
Found: https://platformio.org/lib/show/125/SPIFlash
LibraryManager: Installing id=125
Using cache: C:\Users\Osr\.platformio\.cache\9e\0d799acce6b7fdf6e8b8832f916d149e
SPIFlash @ 3ddba62515 has been successfully installed!
LibraryManager: Installing rpnlib
git version 2.24.0.windows.2
Cloning into 'C:\_proyecto\_Espurna\espurna-master 1.14.0\code\.pio\libdeps\itead-sonoff-mini\_tmp_installing-lgg7q6dk-package'...
Note: switching to 'b95cd7256ff48391b72e982333fdd51311a4183b'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false
`

It seems thar there are the same errors but this time don't stop the build and going on with the next library.

mcspr, 2019-12-16:
@knopserl @Matthias-GER @davebuk could you please check whether the Python version is 3.7.x and does changing the Python version help with the issue?

@oscarsan1 Maybe that is the wrong penv? I do see two, in `c:\.platformio\penv` and in `c:\users\<username>\.platformio\penv`. Not sure which one is actually picked for VSCode and which one is used in the shell PATH - both `<path>\penv\Scripts\python.exe --version` output should be `Python 3.8.0`, not `3.7.4`.

Those warnings are harmless, git just really wants us to know about the switch syntax. It can be disabled via `git config --global advice.detachedHead false` globally, since we are not the ones controlling it anyway.

davebuk, 2019-12-16:
@mcspr `C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code>python --version` gives `Python 2.7.13`
> I copy the libraries to %userprofile%.plattormio\lib, with this there is no need of download libraries for each env_default. The pio creates the directory C:\espurna-master [...]

Similarly to @oscarsan1 , I copied all the lib folders out of one of my previously built binaries `C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\libdeps\itead-sonoff-basic` and copied to `C:\Users\<username>\.platformio\lib` and disabled the git warnings as described above.

I pulled the latest dev version in and built a new version of my `magichome-zj-wfmn-a-11` device after deleting all files from `C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\libdeps\magichome-zj-wfmn-a-11`. The binary built without any errors.

davebuk, 2019-12-16:
I then tried `pio lib update` and got:
```
Library Storage: C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\libdeps\itead-sonoff-basic
Updating ArduinoJson                     @ 6.13.0         [Up-to-date]
Updating ArduinoJson                     @ 5.13.4         [Detached]
Updating AsyncMqttClient                 @ ddbf4d1        [Up-to-date]
Updating AsyncTCP                        @ 1.1.1          [Up-to-date]
Updating Brzo I2C                        @ 1.3.3          [Up-to-date]
Updating DebounceEvent                   @ fdc9c08        [Up-to-date]
Updating EEPROM_Rotate                   @ 00b9d6c        [Up-to-date]
Updating ESP Async WebServer             @ b0c6144        [Detached]
Updating ESP8266_SSD1306                 @ 6575873        [Up-to-date]
Updating ESPAsyncTCP                     @ 7e9ed22        [Detached]
Updating ESPAsyncTCP                     @ 1.2.2          [Up-to-date]
Updating Embedis                         @ 1.2.0          [Up-to-date]
Updating EspSoftwareSerial               @ 5378868        [Up-to-date]
Updating EspSoftwareSerial               @ 6.3.0          [6.4.0]
Uninstalling EspSoftwareSerial @ 6.3.0:         [OK]
Looking for EspSoftwareSerial library in registry
Found: https://platformio.org/lib/show/168/EspSoftwareSerial
LibraryManager: Installing id=168 @ 6.4.0
Using cache: C:\Users\<username>\.platformio\.cache\63\452cb32049478370b740332dd3d9fb63
Unpacking  [####################################]  100%
Error: Traceback (most recent call last):
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 253, in lib_update
    lm.update(library, only_check=only_check)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 816, in update
    self.install(name, latest, after_update=True)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301, in install
    force=force,
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 151, in _install_from_piorepo
    requirements,
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497, in _install_from_url
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "c:\users\<username>\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583, in _install_from_tmp_dir
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\<username>\.platformio\python27\Lib\shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "C:\Users\<username>\.platformio\python27\Lib\shutil.py", line 177, in copytree
    os.makedirs(dst)
  File "c:\users\<username>\.platformio\penv\lib\os.py", line 157, in makedirs
    mkdir(name, mode)
WindowsError: [Error 5] Access is denied: 'C:\\Users\\<username>\\Documents\\PlatformIO\\ESPurna dev\\espurna\\code\\.pio\\libdeps\\itead-sonoff-basic\\EspSoftwareSerial_ID168'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
This is the issue I have faced before. VSCode will try and go through updating the libraries for each of my previously built devices and sometime fail. For this test I then deleted the files in `C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\libdeps\itead-sonoff-basic` and tried again. This time VSCode checked the library versions for each of my other devices I'd previously built. Kaspersky AV was running throughout all tests.

Should VSCode be trying to keep a copy of the libraries for each previously built environment? If not, when do the contents of `C:\Users\<username>\.platformio\lib` get updated?

mcspr, 2019-12-17:
VSCode calls `pio run` on launch, which installs libraries for env_default:
https://github.com/xoseperez/espurna/blob/e5b72a0d43892eec7e4496e73031180bd2f3acb7/code/platformio.ini#L2
After that, `pio run -e <env>` for each build task.

Global libs are never updated unless we modify our scripts to do so. Default behaviour is to check every library path for something that matches lib_deps entry, and if there is nothing found it will install one in the `.pio/libdeps/<env>/`. Either we could have some configuration file ([`extra_configs = ...`](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs)?) or keep using the system environment variable to always try to share libraries.
`.platformio/lib` is not the only option btw, it could also be inside of `code/lib(raries)` if we choose so.

@davebuk and you might want to update to either Python 2.7.17 or 3.8, I see [a bunch of issues for 2.7.13 shutil](https://raw.githubusercontent.com/python/cpython/84471935ed2f62b8c5758fd544c7d37076fe0fa5/Misc/NEWS)

davebuk, 2019-12-18:
> Make sure that VSCode isn't running. And after installing 3.8, just check that running `py -3 --version` in the terminal shows the correct version. You would need to remove the [existing virtual environment used by the PIO](https://docs.platformio.org/en/latest/installation.html#virtual-environment) - I think it is at `C:\Users\Osr\.platformio\penv` based on the logs you provided. After removing it and while in the `C:\Users\Osr\.platformio` directory, run `py -3 -mvenv penv` and `.\penv\Scripts\pip.exe install platformio` to recreate it based on 3.8

I tried the above suggestion and built a binary from a previously built environment which built fine. I then tried building an environment that I had never used before. It failed with the error `WindowsError: [Error 5] Access is denied: 'C:\\Users\\<username>\\Documents\\PlatformIO\\ESPurna dev\\espurna\\code\\.pio\\libdeps\\...` when trying to clone one of the libraries. I tried again and it failed at a different library. A section from the output shows:
```
FauxmoESP @ 0614271 has been successfully installed!
Installing dependencies
LibraryManager: Installing id=305 @ >=1.2.0
ESPAsyncTCP @ 1.2.2 is already installed
LibraryManager: Installing id=1826 @ >=1.0.1
AsyncTCP @ 1.1.1 is already installed
LibraryManager: Installing hlw8012
git version 2.20.1.windows.1
Cloning into 'C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\libdeps\itead-motor-ota\_tmp_installing-9e8f_64x-package'...
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\<username>\\Documents\\PlatformIO\\ESPurna dev\\espurna\\code\\.pio\\libdeps\\itead-motor-ota\\_tmp_installing-9e8f_64x-package\\.git\\objects\\10\\926e87f113fb026c366866d6fa466061562870':
  File "C:\Users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\<username>\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\<username>\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\<username>\.platformio\platforms\espressif8266\builder\main.py", line 203:
    target_elf = env.BuildProgram()
  File "C:\Users\<username>\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\<username>\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 792:
    rmtree(src)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 730:
    return _rmtree_unsafe(path, onerror)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 603:
    _rmtree_unsafe(fullname, onerror)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 603:
    _rmtree_unsafe(fullname, onerror)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 603:
    _rmtree_unsafe(fullname, onerror)
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 608:
    onerror(os.unlink, fullname, sys.exc_info())
  File "c:\users\<username>\appdata\local\programs\python\python38-32\lib\shutil.py", line 606:
    os.unlink(fullname)
```

I then added an anti-virus exception to the folder: `C:\Users\<username>\Documents\PlatformIO\ESPurna dev\espurna\code\.pio\` and tried again. It worked without any issues. For me, it looks my anti-virus causing the problem.

knopserl, 2020-01-04:
I still face the same problem! I have uninstalled, VSC + Platformio, deleted all directories and did a fresh install. I have also installed the latest stable Python version 3.8.1. No change.
I followed all the hints in this thread. Avast antivirus disabled (no other virus scanner running). 
pio run works and does a build of a wemos-d1-mini-relay shield. Any other selective build fails.
BTW: I can build/upload etc. in VSC + PIO but espurna fails.

CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting espurna.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266HTTPUpdateServer\library.properties file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266HTTPUpdateServer\library.properties file
Looking for Embedis library in registry
Found: https://platformio.org/lib/show/408/Embedis
LibraryManager: Installing id=408
Using cache: C:\Users\internet\.platformio\.cache\cc\338ee11d777d75e24f4008826cd592cc
Embedis @ 1.2.0 has been successfully installed!
LibraryManager: Installing espsoftwareserial
git version 2.24.0.windows.2
Cloning into 'D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-16pyir93-package'...
Note: switching to '5378868de76e1a38d34e0fc888d26e3612a5497d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-16pyir93-package\\.git\\objects\\21\\452c9bf32fc95f03c8adb16ad3382d92f0994e':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\internet\.platformio\platforms\espressif8266@1.5.0\builder\main.py", line 375:
    target_elf = env.BuildProgram()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 793:
    rmtree(src)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 731:
    return _rmtree_unsafe(path, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 609:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 607:
    os.unlink(fullname)
====================================================================================================== [FAILED] Took 5.44 seconds ======================================================================================================

mcspr, 2020-01-04:
> BTW: I can build/upload etc. in VSC + PIO but espurna fails.

But are you using `lib_deps = ...` with git urls in other project? The issue is with those as it seems.
Stack trace above points to the read-only `D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-16pyir93-package\.git\objects\21\452c9bf32fc95f03c8adb16ad3382d92f0994e` path that cannot be removed for some reason after `shutil.move(tmp_dir, pkg_dir)` is done copying the data to the `.pio/libdeps/generic-esp01s-relay-40-inv/espsoftwareserial`.

Essentially, this is MVCE for the issue. Can you also try running this?
(`CWD` var path points to ~/Documents, but you may try any directory. Note slash escape)
```python
import os
import sys
import stat
import shutil
import tempfile
import traceback
import subprocess

def stats(path):
    try:
        print(stat.filemode(os.stat(path).st_mode))
    except Exception as e:
        print(e)

def rmtree(path):
    def _onerror(func, path, __):
        try:
            st_mode = os.stat(path).st_mode
            if st_mode & stat.S_IREAD:
                os.chmod(path, st_mode | stat.S_IWRITE)
            func(path)
        except Exception:
            print('> custom rmtree failed')
            traceback.print_last()

    return shutil.rmtree(path, onerror=_onerror)

CWD = os.path.join(os.environ["USERPROFILE"], "Documents")
LIB = 'asyncmqttclient'

target_lib = os.path.join(CWD, LIB)
tmp_lib = tempfile.mkdtemp('-package', '_tmp_installing_test', CWD)

print('> cloning into {}'.format(tmp_lib))
subprocess.run(['git','clone','--depth=1', 'https://github.com/marvinroger/async-mqtt-client', tmp_lib])
print('> moving {} to {}'.format(tmp_lib, target_lib))
shutil.move(tmp_lib, target_lib)

print('> stats')
stats(target_lib)
stats(tmp_lib)

print('> basic rmtree {}'.format(target_lib))
try:
    shutil.rmtree(target_lib)
except Exception:
    print('> basic rmtree failed!')
    exc_type, exc_value, exc_traceback = sys.exc_info()
    traceback.print_tb(exc_traceback)

print('> custom rmtree {}'.format(target_lib))
rmtree(target_lib)
``` 

But I too see such error on Windows 10:

```
C:\Users\maxim>py -3 script.py
py -3 script.py
> cloning into C:\Users\maxim\Documents\_tmp_installing_testntp82948-package
Cloning into 'C:\Users\maxim\Documents\_tmp_installing_testntp82948-package'...
remote: Enumerating objects: 64, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 64 (delta 17), reused 23 (delta 1), pack-reused 0
Unpacking objects: 100% (64/64), done.
> moving C:\Users\maxim\Documents\_tmp_installing_testntp82948-package to C:\Users\maxim\Documents\asyncmqttclient
> stats
drwxrwxrwx
[WinError 2] Не удается найти указанный файл: 'C:\\Users\\maxim\\Documents\\_tmp_installing_testntp82948-package'
> basic rmtree C:\Users\maxim\Documents\asyncmqttclient
> basic rmtree failed!
  File "script.py", line 45, in <module>
    shutil.rmtree(target_lib)
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 730, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 608, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\maxim\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 606, in _rmtree_unsafe
    os.unlink(fullname)
> custom rmtree C:\Users\maxim\Documents\asyncmqttclient

```

Not sure what to make of it yet, will post to the PIO issue tracker, I do also see some weird git interaction in play. Quick search only points to this old issue https://github.com/platformio/platformio-core/issues/2844
Only default Windows Defender on the PC in question, but perhaps it isn't strictly an AV issue.

mcspr, 2020-01-04:
And for something concrete, here's modified PlatformIO:
https://github.com/platformio/platformio-core/compare/develop...mcspr:fs/custom-move
https://github.com/mcspr/platformio-core/tree/fs/custom-move
Replacing `shutil.move` with `shutil.copy(tmp_dir, target_dir)` followed by the `fs.rmtree(tmp_dir)` (that  should properly erase files that trigger WinError 5

knopserl, 2020-01-05:
@mcspr Thanks for your comprehensive reply. But I'm a bit confused what I should do now (sorry).
What do you exactly want me to try? I did not get it.

BTW: All mentioned irectory are full accessible by anybody. I veryfied and changed the permissions and files are created in all that directoties and nay user has full access to all the directory trees.

build and upload work with tasmota (which is from size/complexity/functionality somehow compareable to espurna), but I don't know if it also uses lib_deps = ...
https://github.com/arendst/Tasmota

mcspr, 2020-01-05:
@knopserl The gist of the issue is that Python's `shutil.move`, which PlatformIO library installer uses, should rename directory in-place:
From `D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-16pyir93-package`
To `D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\espsoftwareserial`
But you are hitting the `shutil.copytree()`, where it first copies everything from `_tmp_installing-16pyir93-package` to `espsoftwareserial` and then tries to remove `_tmp_installing-16pyir93-package`

And BTW as discussed above, you can move any installed contents of the `.libdeps/<env>` to the `.platformio/lib` (or even `code/lib`? maybe) so that it can reuse library installation.
 
The issue itself is a mystery though as to why exactly it happens, since you do have correct permissions and directories do not cross drives letters.
Script above is just a verification of how things break with `git/....` files on Windows, note the `moving` step from above:
```
> moving C:\Users\maxim\Documents\_tmp_installing_test306fhvsz-package to C:\Users\maxim\Documents\asyncmqttclient
```
Is what I wondered about, since it fails for `D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv` location for you

Tasmota kind of avoids that through keeping all of the libraries in repo itself:
https://github.com/arendst/Tasmota/tree/development/lib
I'll try to do something similar by installing to either `code/lib` or `code/libraries` automatically and reusing that. Our travis CI scripts already do that btw, optional user install is enabled via `ESPURNA_PIO_SHARED_LIBRARIES=y` environment variable. But there are some subtle issues + this one, so that kind of stalled since the last time I looked at it.

mcspr, 2020-01-05:
And sorry for wall of text :)

As to what to test, the script above would be interesting to try out in different locations.
Another useful thing is to understand wtf exactly is the `os.rename` issue, but you would need to modify Python files to show the stack trace:
https://github.com/python/cpython/blob/v3.8.0/Lib/shutil.py#L779-L781
```diff
diff --git a/Lib/shutil.py b/Lib/shutil.py
index 1e89256cc3..51493359df 100644
--- a/Lib/shutil.py
+++ b/Lib/shutil.py
@@ -778,7 +778,13 @@ def move(src, dst, copy_function=copy2):
             raise Error("Destination path '%s' already exists" % real_dst)
     try:
         os.rename(src, real_dst)
+        print('successfuly renamed {} to {}'.format(src, real_dst))
     except OSError:
+        import traceback
+        print('>' * 80)
+        print('failed renaming {} to {}'.format(src, real_dst))
+        traceback.print_exc()
+        print('<' * 80)
         if os.path.islink(src):
             linkto = os.readlink(src)
             os.symlink(linkto, real_dst)
@@ -787,6 +793,7 @@ def move(src, dst, copy_function=copy2):
             if _destinsrc(src, dst):
                 raise Error("Cannot move a directory '%s' into itself"
                             " '%s'." % (src, dst))
+            print('copytree fallback')
             copytree(src, real_dst, copy_function=copy_function,
                      symlinks=True)
             rmtree(src)
```
The file in question is `"C:\Program Files (x86)\Python38-32\lib\shutil.py"`

knopserl, 2020-01-06:
@mcspr Thanks again for your effort. Not sure if I can follow all your thoughts ;-)
What I did is to modifiy the shutil.py script  here: C:\Program Files (x86)\Python38-32\lib\shutil.py

Below is the output of the build task: There are German parts (I guess you can put it to translate.google.com) but it sometimes says not only permission error but in this case also that the file is locked by another process (maybe concurrent threads running in parallel?)
Hope this provides useful infos for you.
Anything else I can contribute? (you need to give me instructions for a "novice", I'm not a Python nor a PIO expert).
 

> Executing task: C:\Users\internet\.platformio\penv\Scripts\platformio.exe run --environment generic-esp01s-relay-40-inv <

Processing generic-esp01s-relay-40-inv (board: esp01_1m; platform: espressif8266@1.5.0; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting espurna.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266HTTPUpdateServer\library.properties file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\internet\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266HTTPUpdateServer\library.properties file
LibraryManager: Installing IRremoteESP8266
git version 2.24.0.windows.2
Cloning into 'D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-lj52x8nk-package'...
Note: switching to '4d57a2b37afeaf2fee1fa75bbcab260404e8f3c7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Submodule 'lib/googletest' (https://github.com/google/googletest.git) registered for path 'lib/googletest'
Cloning into 'D:/mcdev/vscode/espurna/code/.pio/libdeps/generic-esp01s-relay-40-inv/_tmp_installing-lj52x8nk-package/lib/googletest'...
Submodule path 'lib/googletest': checked out '887d569eb0bac3760cb3adf5877c8284e4d7768e'
>>>>>>>>>>>>>>>>>>Traceback (most recent call last):
>>>>  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 781, in move
>>>    os.rename(src, real_dst)
>>>PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-lj52x8nk-package' -> 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\IRremoteESP8266'
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
failed renaming D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-lj52x8nk-package to D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\IRremoteESP8266
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
copytree fallback
[WinError 32] Der Prozess kann nicht auf die Datei zugreifen, da sie von einem anderen Prozess verwendet wird: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-lj52x8nk-package\\lib\\googletest'
Please manually remove the file `D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-lj52x8nk-package\lib\googletest`
[WinError 145] Das Verzeichnis ist nicht leer: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-lj52x8nk-package\\lib' 
Please manually remove the file `D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-lj52x8nk-package\lib`
[WinError 145] Das Verzeichnis ist nicht leer: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-lj52x8nk-package' 
Please manually remove the file `D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-lj52x8nk-package`
PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-lj52x8nk-package\\.git\\modules\\lib\\googletest\\objects\\pack\\pack-cc51e268acc6d39a182162db91cd0093c9179d89.idx':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
[... cut ...]
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 609:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 607:
    os.unlink(fullname)
====================================================================================================== [FAILED] Took 11.61 seconds ======================================================================================================

mcspr, 2020-01-06:
Thanks! What is weird is that it shows WinError 5 for rename too. 2nd error from googletest dir is new though, process conflict perhaps points to the `git` still running in the directory, thus not allowing anything to change directory contents / rename it.

*(btw I edited the log you provided to surround it with \`\`\`, that way it does not apply any formatting to the text and uses monospace font)*

mcspr, 2020-01-06:
@knopserl does adding this line to the `exec_command` function inside of `proc.py` change the behaviour?
`"C:\users\internet\.platformio\penv\lib\site-packages\platformio\proc.py"`
```diff
diff --git a/platformio/proc.py b/platformio/proc.py
index 80e50201..dfd77cd4 100644
--- a/platformio/proc.py
+++ b/platformio/proc.py
@@ -135,6 +135,8 @@ def exec_command(*args, **kwargs):
         if v and isinstance(v, string_types):
             result[k] = result[k].strip()

+    p.wait()
+
     return result

```

knopserl, 2020-01-06:
@mcspr thanks for your patience.
I have modified the proc.py as you've sugested. The "locked by another process" disapear, but that does not show up every time.
I also created a new directory and moved the espurna project there, just to ensure, that it's not tthe directoty permission. I checked the permission and all user have full access. It's same if I use another admin account! I also used another build task. I've also tried the newest and older espurna version.
At least a year ago, I was able to build espurna on that PC but many things changed (new Windows 10 installation, new VSC, new PIO etc.) 

```
        if v and isinstance(v, string_types):
            result[k] = result[k].strip()
            
    p.wait()    

>>>  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 781, in move
>>>    os.rename(src, real_dst)
>>>PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-npvu43kt-package' -> 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\JustWifi'
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
failed renaming D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\_tmp_installing-npvu43kt-package to D:\mcdev\vscode\espurna\code\.pio\libdeps\generic-esp01s-relay-40-inv\JustWifi
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
copytree fallback
PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\generic-esp01s-relay-40-inv\\_tmp_installing-npvu43kt-package\\.git\\objects\\01\\3e8594930a1ae1a3885a20bff6b48680cf104a':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:

>>>  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 781, in move
>>>    os.rename(src, real_dst)
>>>PermissionError: [WinError 5] Zugriff verweigert: 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\_tmp_installing-0uxti8f8-package' -> 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\AsyncMqttClient'
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
failed renaming D:\vscode\espurna\code\.pio\libdeps\esp8266-1m-ota\_tmp_installing-0uxti8f8-package to D:\vscode\espurna\code\.pio\libdeps\esp8266-1m-ota\AsyncMqttClient
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
copytree fallback
PermissionError: [WinError 5] Zugriff verweigert: 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\_tmp_installing-0uxti8f8-package\\.git\\objects\\03\\2bb680a76fa5d1afbef50438388d1ccad38c23':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")

>>>  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 781, in move
>>>    os.rename(src, real_dst)
>>>PermissionError: [WinError 5] Zugriff verweigert: 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\_tmp_installing-m4wor570-package' -> 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\EEPROM_Rotate'
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
failed renaming D:\vscode\espurna\code\.pio\libdeps\esp8266-1m-ota\_tmp_installing-m4wor570-package to D:\vscode\espurna\code\.pio\libdeps\esp8266-1m-ota\EEPROM_Rotate
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
copytree fallback
PermissionError: [WinError 5] Zugriff verweigert: 'D:\\vscode\\espurna\\code\\.pio\\libdeps\\esp8266-1m-ota\\_tmp_installing-m4wor570-package\\.git\\objects\\00\\b9d6c9a4a765a96958ffe1aaf5d95357c2aebc':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
```

mcspr, 2020-01-07:
Don't think changing espurna versions would do much, library installation is handled by the PIO. Something might have changed during last year in how it handles it though, old versions might've done things differently. But there were some changes to the build system so you would not be able to build current version 1.14.1 with PIO <= 4.0.0, last compatible version would be the 1.13.5

> log outputs (multiple)

Are those the only errors and does the PIO process continue to download libraries then? If yes, then it works just fine, you just need to remove Python's `shutil.py` modifications from the previous comment.
I'd continue the discussion at the https://github.com/platformio/platformio-core/issues/3328, if `proc.py` change actually helps.

As previously said, I'd bet it has something to do with AV software, but not sure how it would interact with things when "Avast antivirus disabled (no other virus scanner running).". Stock Windows Defender AV does not cause any issues with my installation :/ Just running the default settings, no extra stuff and directories in question are not even added to any exceptions. It just works as-is.

knopserl, 2020-01-07:
@mcspr No this is not just a simple error, it breaks the build and link job. so I don't get an image and it does then of caourse also not upload. So basically I can't use the espurna project as it breaks the build after 5 seconds and does not finish. So unusable. That was the reason why I had to switch to tasmota because there I can build firmware, but I rather wanted to continue with espurna.

I could run my virtual machine under Minut Lunx 19.2 (Tina) and try to do espurna development in Linux (which is rather a workaround), until it gets solved.

Disabling the antivirus did not help either (I have tested it several times). I could also do a test on another PC where another antivirus is installed or deistall the antivirus completly (just run Windows Defender). But as I said, when I switch of all AV components, it has no effect.

Do you develop under Windows 10 (1909) and do you have such problems or do you use Linux or another Windows 10 version or do you know from others having this problem?

So you propose, that I continue the discussion in this PIO thread?

mcspr, 2020-01-08:
Let's review other errors in a separate issue. I assumed that the project was building ok if libraries are present, but I really don't get how that could go wrong too.

I do routinely use Windows PC to build things without any problems. Besides info based on comments in this thread at least 3 other people had similar issue, but I am not aware of any other problems with any other systems.

> So you propose, that I continue the discussion in this PIO thread?

Yes. Just make sure to undo both PIO and Python modifications done so far.

knopserl, 2020-02-02:
@mcspr I did not gave up and didi further testing.
Here are my latest findings: 
When I do a Build task without any environment/device selection it takes the wemos-d1mini-relayshield and that build works. But that default devices seem to do no cloning,
When I use any other devices/env. it fails with the access denied error.

Now I tried your previous recommendation and changed to espurna/code directory and executed:
platformio.exe run --environment blitzwolf-bwshpx
Surprisingly, that worked now too.
So it seem to be a VScode + platformio problem.
Any further ideas how to get that further investigated and solved?
I can also post this to the platformio issue.

mcspr, 2020-02-04:
PlatformIO guys would know both parts better, please refer there.

Repeating the discussion above, you can move contents of `.pio/libdeps/<env>/` to either:
- `~/.platformio/lib/`
- `<espurna-root>/code/libs/` 
- `<espurna-root>/code/libraries/`
It will use them *for every environment* until we update something in .ini changing versions.

There are some developments in the pio-core about pre-packaging library archives, perhaps that is something to look into instead of "vendoring" source code as-is.

knopserl, 2020-02-04:
@mcspr thanks. I posted the infos to vscode and platformIO (+platformIO.vscode extention) teams. The platformIO team did not further respond since 4 weeks and it worked standalone so I infomred vscode team, but the rejected and pointed me back to platformIO (speficically the extention team).

Thanks for the hint. Where do I configure the .pio/libdefs location to use/move to another root directory?

mcspr, 2020-02-05:
>  And would it solve the permission problem?

It would stop installation each time new environment is used, for now.

> Where do I configure the .pio/libdefs location to use/move to another root directory?

Nowhere. Just place contents of `.pio/libdeps/wemos-d1mini-relayshield/` into one of the directories listed above. PIO checks those first, then it tries to install locally to `.pio/libdeps/....`

knopserl, 2020-02-06:
@mcspr thanks, will try this weekend.

closed 2020-04-14 by stale[bot]

knopserl, 2020-04-26:
@mcspr I have seen this topic was closed but not solved. I opend issues at vscod and pio + pio add-on team but non of them helped. So for me, it's no longer possible to use espurna (at least not with pio). All firther updates don't helped. As said, it's reprodocable on other PC's and I'm wondering how it could work for others (under Windows 10).

mcspr, 2020-04-28:
I'd continue to pursue platformio-core issue. You have to understand that it looks like some specific setup issue, as neither of us (PIO core developers and me personally) can reproduce it with our Windows 10 setups , in VMs or on real hardware, with Home or Pro editions :(

I can't offer any other help but to add some workaround here with library vendoring i.e. I would download libraries myself and add them into the repo as .zip files, build system will use them instead of trying to fetch them and trigger the issue.

Without fixing platformio itself, you would still have the issue with any other project if they are using `lib_deps`
For example, https://github.com/1technophile/OpenMQTTGateway/blob/development/platformio.ini build with WiFiManager

mcspr, 2020-04-28:
https://github.com/platformio/platformio-core/issues/3328 is still open btw