xoseperez/espurna #2007: Smart Bulb

lblabr, 2019-11-24:
I have a smart bulb from Aoycocr. it does not lights white when all 3 RGB channels are at 255. it also seems it could also be shine brighter ... to prevent firmware update i didn't tested with smart life app...

i reorded another bulb to verfify...

[image]
b

do you have an hint, why its not shines more or less white with the first three channels at 255 ?

WollyCGN, 2020-01-17:
I've also this device. Which image did you flash? My first tests with tasmota looks the same. No real white as factory default. I opened one device and there are RGB and CW Leds inside. so I assume it's a RGBW device.
btw, I'm talking about Aoycocr Q3CM GU10 Smart Bulb

mcspr, 2020-01-18:
Channel \#3 is white LED. It is set to 0 on the screenshot above, so... there is no white LED on 👐

WollyCGN, 2020-01-21:
Thanks, but again what file to flash for this device? Maybe I'm blind :-)

mcspr, 2020-01-21:
No... There is no .bin for this device, I think @lblabr used some custom profile and built it.

Is this the bulb in question?
https://templates.blakadder.com/aoycocr_Q3CM.html
I do see `itead-sonoff-b1` that has the same connections for my92xx driver, but I am not sure about the channel map it uses:
https://github.com/xoseperez/espurna/blob/cbd7064c5c99622204c04203de7e2e818871011c/code/espurna/config/hardware.h#L806-L809

WollyCGN, 2020-01-22:
Yes, this is the correct bulb.
I just use tasmota. The SONOFF B1 will not work because of other mappings and also works with a 10 digit hexcode for color-w-cw.

closed 2020-03-29 by stale[bot]