xoseperez/espurna #2008: small firmware package

lblabr, 2019-11-24:
i switched to arduino ide 1.8.10 how do i get small firmware images ? the arduino ide takes core 2.6.1, in my opinon the files are larger as before

are there any hints to have imgae sizes < 500k

i would like to have: MQTT_SUPPORT, WEB_SUPPORT, DEBUG_WEB_SUPPORT, ALEXA_SUPPORT, SCHEDULER_SUPPORT, LED_SUPPORT, BUTTON_SUPPORT, SENSOR_SUPPORT

thanks in advance

Lars

mcspr, 2019-11-29:
Besides disabling some default modules, also see `MDNS_SERVER_SUPPORT` and `OTA_ARDUINOOTA_SUPPORT`, on latest Core it takes ~35KB by using those features. If web server is still around, OTA could be done through web or [`/upgrade`](https://github.com/xoseperez/espurna/wiki/OTA#updating-via-http)

lblabr, 2019-12-01:
thank you very much!
how do i may switch between core libraries, depending on device ?

mcspr, 2019-12-01:
Look up platformio env: stuff, all build configuration is easily changed with a single .ini file
At the top of ours there a comments regarding `platform = …` setting

lblabr, 2019-12-01:
i'm working with arduino ide, are there similar settings ?

closed 2020-02-06 by stale[bot]