xoseperez/espurna #2009: Smart bulb, restore switch state after reboot/get powered

lblabr, 2019-11-24:
on my plugs may configure, what the relays should do after restart. is that also possible for bulbs with dummy relay ?

my configuration:

    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1

    // Light
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       14      // RED
    #define LIGHT_CH2_PIN       12      // GREEN
    #define LIGHT_CH3_PIN       5      // BLUE
    #define LIGHT_CH4_PIN       15       // WHITE
    #define LIGHT_CH1_INVERSE   0
    #define LIGHT_CH2_INVERSE   0
    #define LIGHT_CH3_INVERSE   0
    #define LIGHT_CH4_INVERSE   0

reaper7, 2019-11-26:
I don't know if it also works for dummy relays 
but try to add:
```
#define RELAY_BOOT_MODE           1                                       // 0 - off, 1 - on, 2 - same
```

lblabr, 2019-11-26:
thank you for responding!

RELAY_BOOT_MODE is a global setting and not relay specific ?

lblabr, 2019-11-28:
i did worked for me...

closed 2019-11-28 by mcspr