xoseperez/espurna #2011: physical relay and dummy relay

reaper7, 2019-11-25:
I use current dev branch and after recent changes,
my configuration has stopped working correctly:

I have itead s20 and additionally I use the RX pin 
to control the led strip (via mosfet transistor).

mcspr, 2019-11-25:
Huh, looking at https://github.com/xoseperez/espurna/commit/f4b0e9fe5b19f8cc3bf030496e934e175b80c0e8 again that did not actually include real relays and used dummy_count as total relays count

https://github.com/xoseperez/espurna/commit/87e5d21322c0986b84f909d4f477fc4cd6a9bbb2 should fix this

reaper7, 2019-11-25:
with this fix real relay still do not work
but dummy relay switch to on only when 2 switches on web interface are on ( logical AND )

@mcspr - there have been many changes after working for me https://github.com/xoseperez/espurna/commit/c96f2e4e3a36629ef78187f00fae10c46a7a849f 
maybe my old configuration, that previously worked, is not logically correct with the new relay concept ? 

including my configuration for this device from hardware.h and custom.h, it's looks like this:
```
  #=== from hardware.h ===

    // Info
    #define MANUFACTURER              "ITEAD"
    #define DEVICE                    "S20"

    // Buttons
    #define BUTTON1_PIN               0
    #define BUTTON1_MODE              BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY             1

    // Relays (hardware relay)
    #define RELAY1_PIN                12
    #define RELAY1_TYPE               RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                  13
    #define LED1_PIN_INVERSE          1
    
  #=== from custom.h ===

    #define RELAY_BOOT_MODE           2                                       // 0 - off, 1 - on, 2 - same
    #define LED1_MODE                 LED_MODE_MQTT

    #define RELAY_PROVIDER            RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER            LIGHT_PROVIDER_DIMMER
    #define LIGHT_USE_COLOR           0                                       // Use 3 first channels as RGB
    #define LIGHT_USE_WHITE           0                                       // Use the 4th channel as (Warm-)White LEDs
    #define LIGHT_WHITE_FACTOR        1
    #define LIGHT_USE_TRANSITIONS     1                                       // Transitions between colors
    #define LIGHT_TRANSITION_TIME     2500                                    // Time in millis from color to color
 
    #define DUMMY_RELAY_COUNT         1
    #define LIGHT_CHANNELS            1    

    #define LIGHT_CH1_PIN             3                                       //GPIO1 RX
    #define LIGHT_CH1_INVERSE         0
```

mcspr, 2019-11-25:
Ok, my bad (again). You need the full tree which that commit is the HEAD of, see the link at the top of the #2012 
Boot sequence was broken and never triggered relay provider action

reaper7, 2019-11-25:
but I made the changes manually (relay.h + relay.ino)

you suggest me to download: https://github.com/mcspr/espurna/tree/relay/dummy-resize  ?

mcspr, 2019-11-25:
So this part manages the id selection, but I have not changed it recently in any way besides switching from DUMMY_RELAY_COUNT to _relayDummy var:
https://github.com/xoseperez/espurna/blob/a6c5c7e7c7367a56451f02f981fce7c5f943e020/code/espurna/relay.ino#L256-286

As I understand it, real relay1 still controls the global light state, thus channel values are 0 until it is on. That should probably not happen and only dummy relays should control relay state

mcspr, 2019-11-25:
...and that is why it did https://github.com/mcspr/espurna/commit/f4284044c562435473322c33acfe1f3704564bbe

reaper7, 2019-11-25:
no, real relay has nothing to do with light.

- physical Relay1(on/off) controlled by physical Button1 (on attatched above screenshot this is Switch#0) which is complete itead_s20
- and second, virtual relay (dummy) with output on RX pin, (on attatched above screenshot this is Switch#1), supporting also dimming function.

It was basically for saving esp devices, ITEAD S20 switches on multimedia devices (230V)
and a free RX pin controls the TV backlight (12V via mosfet):
[image]

reaper7, 2019-11-25:
this is it! https://github.com/mcspr/espurna/commit/f4284044c562435473322c33acfe1f3704564bbe
please do not close this issue, I will check this carefully tomorrow.

P.S. 
as I said, you know this project very well, tnx!

lblabr, 2019-11-26:
@reaper7 could you help me on my task ? 

[issue](https://github.com/xoseperez/espurna/issues/2009)

the bulb should immidiatly switch on on power on. how do i configure that ?
the switches section is empty ....

thanks in advance

Lars

reaper7, 2019-11-26:
@mcspr - fix tested on all my devices, works OK!

PR #2012 (2019-11-26): relay: compare sum of physical + dummy relays instead of just dummy
fix #2011
closed 2019-11-26 by reaper7