xoseperez/espurna #2022: Support for HBN Smart plug

davebuk, 2019-11-28:
Request for another device environment. Its an 'HBN Smart Plug' [http://www.hbn-ergo.com/product/product-2-746.html](url)

The model is BNC-50/E75T on the label. Internally it uses a TYWE2S chip. I used the SWA1 environment to build using the following settings. The LED function can then be set in the web UI for what ever function the user wants. 

[image]
[image]
[image]

```
#elif defined(LINGAN_SWA1)
// Info
    #define MANUFACTURER        "LINGAN"
    #define DEVICE              "SWA1"
    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          12 
    #define RELAY1_TYPE       RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            4  //Blue LED
    #define LED1_PIN_INVERSE    1
    #define LED2_PIN            5  //Red LED
    #define LED2_PIN_INVERSE    1
```

mcspr, 2019-11-28:
Also see
https://github.com/xoseperez/espurna/blob/55a0abf1e15cc45a6b7de98c7708face7cb9e1c4/code/espurna/config/hardware.h#L3787-L3814
No pull-up set though

davebuk, 2019-11-29:
`LITESUN LA-WF3` works fine with this device even without pull up being defined.

closed 2019-11-29 by davebuk