xoseperez/espurna #2032: How to set NTP server == Gateway ?

PieBru, 2019-11-30:
**Question**
Hi all, I'm using `custom.h` with different 3 wifi networks where the NTP server is always the default gateway. Those 3 networks (home, recreational vehicle, mom's home) have different IP subnets (192.168, 172.16, 10.10), so the IP address and hostname of the gateway dinmically depends on the wifi network connected. I can't changing this scheme easily.

Using `custom.h` I usually set the NTP server as an IP address, but in my situation it would be nice to define it as `@default_gateway` or any other similar way.

I can "C" code in the Espurna codebase, but I would appreciate to be guided toward the best (and possibly simplest) "Espurnish" way to get this result.

Any hint?
Thank you,
Piero

PieBru, 2019-11-30:
Forgot to mention: same logic for the MQTT server (I use OpenWRT with NTP server and MQTT broker).
In my case, the Home Assistant server also depends on the connected wifi, but it is a RasPI4 other than the default gateway.

mcspr, 2019-12-01:
Hey,

I think there is a case in replacing `#define NTP_SERVER “pool.ntp.org”` with an empty string that does just that – use current gateway IP (`WiFi.gatewayIP()`) as a server. But we could explicitly use some substitutions like mqtt stuff and avoid implicit stuff like that. The same way where static network config (`WIFI1_IP`, `WIFI1_GW` etc.) could just use `#define NTP_SERVER WIFI1_GW`

But I kind-of get a feeling it could also be solved via custom DNS names 😊

PieBru, 2019-12-01:
Great! I will try ths ASAP.
Thanks,
Piero

mcspr, 2019-12-01:
Ah! Wait. It does not work like that atm.
Existing NTP module and NTPClientLib do not do that, I was suggesting the possible change to have either empty string that will automatically call WiFi.gatewayIP() or use some kind of substitution strings like `{wifi.gatewayIP}` in place of server hostname (but I wonder if DNS is a proper way to handle this, delegating it to the network instead of the device)

There is also #1380 so that NTP server will be discovered via DHCP, but it will only work with 2.6.2+ Arduino Core

PieBru, 2019-12-01:
I understood your reply in the wrong way, sorry.
Thanks,
Piero

closed 2020-02-06 by stale[bot]

mcspr, 2020-02-07:
btw I closed the relevant NTP issue in the #2137 
NTP can pick up settings from DHCP server:
https://tools.ietf.org/html/rfc2132#section-8.3

For example, relevant setting for dnsmasq & openwrt is:
```
config dnsmasq
    list dhcp_option '42,<ip-address-of-the-host>'
```
Or dnsmasq itself
```
dhcp-option=option:ntp-server,<ip-one>,<ip-two>
```

And as it is apparent, I haven't added `{hostname}` substitution. For some reason it feels too much like DNS re-implementation on top of our settings :/

PieBru, 2020-02-07:
Great, thanks.
Piero